eProsima/Micro-XRCE-DDS-Agent #1: Port(UDP, UART) does not open on 1.0.0beta2

OpusK, 2018-08-01:
Hi,

I am testing with 1.0.0beta2 which I recently updated. There is a problem, and when I check it, the Agent can not open the Port(UDP, UART).

For UDP, I checked with wireshark that "destination unreachable" message(ICMP packet) is being returned.
For UART, I created and tested the main code, which does not work if the UART port does not open, but it did not work when running the Agent (version 1.0.0 beta works this way).
Also, if an attempt is made to open the same port with a serial program while the Agent is running, an error should occur. However, the serial program opens the port and communicates!

Do you know anything about this?

OpusK, 2018-08-01:
I tested with udp port number 2020, 2018.

julianbermudez, 2018-08-02:
Hi @OpusK, 
I tested the UDP server in Agent with `udp 2018` and it works for me with the current version or release/1.0.0beta2 branch. Could you confirm that it works for you?

OpusK, 2018-08-02:
Hi, @julianbermudez 

master version is working well.
But, release/1.0.0beta2 branch version is not working.

I have confirmed that the currently open port is open with loopback ip.

> udp        0      0 127.0.0.1:2018          0.0.0.0:*                           452/MicroRTPSAgent

I have been trying to connect to remote ip from now on.
In master version, I did not check which ip port is open because it worked well then. 

Anyway, is it correct to open in loopback ip?
In this condition, it is correct that remote can not reach this port.

julianbermudez, 2018-08-07:
Hi @OpusK, 
We have just launch a new release 1.0.0beta2. It should solve your issue. Please give me your feedback.

OpusK, 2018-08-08:
Hi, @julianbermudez 

Now the Agent is opening the port at 0.0.0.0!
Client and Agent are connected normally. Thank you.

closed 2018-08-08 by OpusK