eProsima/Micro-XRCE-DDS-Agent #3: Segmentation fault

OpusK, 2018-08-24:
Hi,

I am trying to use it with ROS2 from raspberryPi. 
However, after executing Agent, "Segmentation fault" error occurs after "Participant matched" log.

Of course, it works well only with Fast-RTPS and microRTPSAgent installed.
However, when I install ROS2, this is happening.

When ROS2 is installed, linked libraries of MicroRTPSAgent are connected to FastRTPS and FastCDR of ROS2. Is this related to this?
If I erase ROS2, it will work again.

Below is a debugging message from gdb.

```
pi@raspberry_kei:~ $ gdb MicroRTPSAgent
GNU gdb (Raspbian 7.12-6) 7.12.0.20161007-git
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "arm-linux-gnueabihf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from MicroRTPSAgent...(no debugging symbols found)...done.
(gdb) run serial /dev/ttyACM0
Starting program: /usr/local/bin/MicroRTPSAgent serial /dev/ttyACM0
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
[New Thread 0x768e5450 (LWP 18923)]
[New Thread 0x760e5450 (LWP 18924)]
[New Thread 0x758e5450 (LWP 18925)]
[New Thread 0x750e5450 (LWP 18926)]
Serial agent initialization... OK
[New Thread 0x744ff450 (LWP 18927)]
[New Thread 0x73cff450 (LWP 18928)]
[New Thread 0x734ff450 (LWP 18929)]
[New Thread 0x72cff450 (LWP 18930)]
[New Thread 0x724ff450 (LWP 18931)]
RTPS Participant matched 1.f.3c.95.f0.68.0.0.0.0.0.0|0.0.1.c1

Thread 7 "MicroRTPSAgent" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x73cff450 (LWP 18928)]
0x76f5ce70 in unsigned int asio::detail::epoll_reactor::cancel_timer<asio::detail::chrono_time_traits<std::chrono::_V2::steady_clock, asio::wait_traits<std::chrono::_V2::steady_clock> > >(asio::detail::time     r_queue<asio::detail::chrono_time_traits<std::chrono::_V2::steady_clock, asio::wait_traits<std::chrono::_V2::steady_clock> > >&, asio::detail::timer_queue<asio::detail::chrono_time_traits<std::chrono::_V2::     steady_clock, asio::wait_traits<std::chrono::_V2::steady_clock> > >::per_timer_data&, unsigned int) () from /usr/local/lib/libmicrortps_agent.so

```

OpusK, 2018-08-28:
+Update:
I have found a way to solve this!

I thought that there was a problem with the libfastrtps.so.1 file in ROS2.0 and replaced it with the libfastrtps.so.1 file generated when FastRTPS was installed, and it was executed well without a "segmentation fault" error.

If have installed FastRTPS and have installed ROS2.0(bouncy) in a binary way:
$ cd /opt/ros/bouncy/lib
$ sudo mv libfastrtps.so.1 libfastrtps.so.2   #for backup
$ sudo cp /usr/local/lib/libfastrtps.so.1 ./

julianbermudez, 2018-09-03:
Hi @OpusK 

We are working on a more appropriate solution. The bug was related with the Asio distribution.
Once we test we will update it.

julianbermudez, 2018-09-21:
Hi Key, 
We fixed this bug on pull #8. As I mentioned before the problem was related with the Asio library.
We tested it in a clean docker with ROS2 distribution.

julianbermudez, 2018-11-15:
I close this issue, if some else has this problem, please reopen it.

closed 2018-11-15 by julianbermudez