eProsima/Micro-XRCE-DDS-Agent #4: [RTPS_HISTORY Error] resize payload size error

OpusK, 2018-09-03:
Hi,

```
UDP agent initialization... OK
Error written data on DDS entity.
[RTPS_HISTORY Error] Change payload size of '1500' bytes is larger than the history payload size of '1103' bytes and cannot be resized. -> Function add_change
Segmentation fault (core dumped)
```
Here, is the history payload size of '1103' bytes set to the maximum size?
Is this a protocol limitation? Or is there another way?

richiware, 2018-09-03:
Currently MicroRTPS agent registers a default generic type in FastRTPS. This generic type has a maximum size of 1028 and 
by default FastRTPS entities uses the memory policy PREALLOCATED_MEMORY_MODE.

In next releases we will make MicroRTPS agent more aware of types (we are developing dynamic types for this purpose) and 
it will know the maximum size of a type. But for this release you can solve this problem configuring FastRTPS entities 
to use PREALLOCATION_WITH_REALLOC_MEMORY_MODE. With this memory mode FastRTPS will realloc memory if a sample is higher 
of the maximum size of the type.

To configure a DataWriter using xml:

```
<publisher>
    <historyMemoryPolicy>PREALLOCATED_WITH_REALLOC</historyMemoryPolicy>
</publisher>
```

To configure a DataReader using xml:

```
<subscriber>
    <historyMemoryPolicy>PREALLOCATED_WITH_REALLOC</historyMemoryPolicy>
</subscriber>
```

OpusK, 2018-09-03:
@richiware,

Thank you! I solved it thanks to you.

closed 2018-09-03 by OpusK

PYRET1C, 2023-07-24:
where do we find this xml file