eProsima/Micro-XRCE-DDS-Agent #10: Profile 'default_xrce_participant_profile' not found

iluetkeb, 2018-09-21:
When I run the agent using a path (e.g., `micrortps/bin/microRTPSAgent`), I cannot connect to it with a client. It always reports
```
[XMLPARSER Error] Profile 'default_xrce_participant_profile' not found ' -> Function fillParticipantAttributes
[PARTICIPANT Error] Problem loading profile 'default_xrce_participant_profile' -> Function createParticipant

```
Instead, when I cd to the "bin" directory and execute it as `./microRTPSAgent`, everything works. I'm assuming it doesn't resolve the configuration file correctly. Maybe it could look at `argv[0]` to determine the path to load the config from.

btw, installing the configuration in the same place as the binary is not standard practice. Just saying.

julianbermudez, 2018-09-24:
Hi @iluetkeb,

There is a mistake in the [manual](https://micro-rtps.readthedocs.io/en/latest/agent.html), it is the current directory which shall contain the `DEFAULT_FASTRTPS_PROFILE.xml` ant not the `MicroRTPSAgent`. I will fix it today. 

On the other hand you are right, it is on the roadmap to remove the `DEFAULT_FASTRTPS_PROFILE.xml` installation and to add a third argument in the `MicroRTPSAgent` for the XML profile path.
We will take into account the possibility to use the `argv[0]` as reference path for the XML profile.

julianbermudez, 2018-11-15:
This issue was solved in the Release 1.0.0. Manual documentation was fixed and the installation of `DEFAULT_FASTRTPS_PROFILE.xml` was removed.

closed 2018-11-15 by julianbermudez