eProsima/Micro-XRCE-DDS-Agent #25: Compilation problems.

jfm92, 2018-11-12:
Hi,

I'm triying to compile micro-XRCE-DDS but I'm having the next problem:
```
[  3%] Building CXX object CMakeFiles/microxrcedds_agent.dir/src/cpp/Root.cpp.o
[  7%] Building CXX object CMakeFiles/microxrcedds_agent.dir/src/cpp/processor/Processor.cpp.o
[ 10%] Building CXX object CMakeFiles/microxrcedds_agent.dir/src/cpp/client/ProxyClient.cpp.o
[ 14%] Building CXX object CMakeFiles/microxrcedds_agent.dir/src/cpp/participant/Participant.cpp.o
/home/juan/micro-XRCE-DDS-agent/src/cpp/participant/Participant.cpp: In member function ‘virtual bool eprosima::uxr::Participant::matched(const dds::xrce::ObjectVariant&) const’:
/home/juan/micro-XRCE-DDS-agent/src/cpp/participant/Participant.cpp:152:43: error: no match for ‘operator==’ (operand types are ‘eprosima::fastrtps::ParticipantAttributes’ and ‘const eprosima::fastrtps::ParticipantAttributes’)
     return parser_cond && (new_attributes == old_attributes);
                                           ^
CMakeFiles/microxrcedds_agent.dir/build.make:134: fallo en las instrucciones para el objetivo 'CMakeFiles/microxrcedds_agent.dir/src/cpp/participant/Participant.cpp.o'
make[2]: *** [CMakeFiles/microxrcedds_agent.dir/src/cpp/participant/Participant.cpp.o] Error 1
CMakeFiles/Makefile2:67: fallo en las instrucciones para el objetivo 'CMakeFiles/microxrcedds_agent.dir/all'
make[1]: *** [CMakeFiles/microxrcedds_agent.dir/all] Error 2
Makefile:127: fallo en las instrucciones para el objetivo 'all'
make: *** [all] Error 2
```

I've installed the last version of fast-rtps (master) and I've followed the next steps to install micro-XRCE
- git clone https://github.com/eProsima/micro-XRCE-DDS-agent.git
- cd micro-XRCE-DDS-agent
- mkdir build && cd build
- cmake -DTHIRDPARTY=ON ..
- make
But the problem raise.

lemunozm, 2018-11-12:
Hi @jfm92, probably your problem is the same as: https://github.com/eProsima/Micro-XRCE-DDS/issues/14 
You are linking with a different version of FastRTPS. Maybe you have a global installation of FastRTPS, and the agent found these headers. Try to unistall FastRTPS. The `cmake -DTHIRDPARTY=ON ..` command you are using will install the newer one for the agent.

jfm92, 2018-11-12:
HI @lemunozm ,
I've done a clean installation of the master branch of the repository  fast-rtps:
https://github.com/eProsima/Fast-RTPS
And I'm using the master branch of micro-XrCE-DDS Agent, but the problem persist. 
I've followed the next instructions:
https://micro-xrce-dds.readthedocs.io/en/latest/installation.html
http://docs.eprosima.com/en/latest/sources.html

lemunozm, 2018-11-12:
The clean installation that you are following into http://docs.eprosima.com/en/latest/sources.html is installing Fast RTPS globally in your OS. Currently this version of Fast RTPS is not compatible with the las version of Micro XRCE-DDS Agent, that uses a particular commit of Fast RTPS (which is not in master yet).

The solution is unistall your global installation of Fast RTPS and allow that Micro XRCE-DDS Agent donwload and install locally the version that it needs with the cmake flag `-DTHIRDPARTY=ON` as you were doing.