eProsima/Micro-XRCE-DDS-Agent #26: Automated Docker Image on Docker Hub

raveious, 2018-11-14:
The image on Docker Hub seems to be a few months out of date. It might be worth while to setup automated building on Docker Hub that watches this repo, and automatically tags each build with the release tags. This would the require the dockerfile being included in this repository.

raveious, 2018-11-22:
If the entry on Docker Hub were to be placed with an Automated Build entry, instructions can be found [here](https://docs.docker.com/docker-hub/builds/). You can also have it automatically tag images with the release tag used on this repository, the [documentation](https://docs.docker.com/docker-hub/builds/#create-an-automated-build) has an example of how to set this up.

If the automated build gets setup this way, then no one would have to manually push images to Docker Hub. With every release, a new image will be made and tagged with the release version, as well as the latest.

julianbermudez, 2018-12-17:
Hi @raveious, 

I followed your advice and set up an automatic build in [Docker Hub](https://cloud.docker.com/u/eprosima/repository/docker/eprosima/micro-xrce-dds-agent). Thank you so much!

closed 2018-12-17 by julianbermudez