eProsima/Micro-XRCE-DDS-Agent #28: [1.0.1 Version] When closing the Agent with 'q' input, I wish to remove the topic entities managed by the Agent.

OpusK, 2018-11-16:
Hi,

Currently, we are using Micro-XRCE-DDS 1.0.1 with our [XEL Network](https://github.com/ROBOTIS-GIT/XelNetwork_CommXel/tree/develop) project and [ros2arduino](https://github.com/ROBOTIS-GIT/ros2arduino/tree/develop) project.

It's still working fine as we used it in the previous 1.0.0beta.
I saw the addition of the ability to shut down the Agent (via the 'q' input to the Agent).
However, when I shut down the Agent, I get a "segmentation fault" error as follows, and the DDS topic still exists. (I checked with ros2 topic list command)
```bash
kei@ROBOTIS-9560:~$ MicroXRCEAgent serial /dev/ttyACM0
Serial agent initialization... OK
Enter 'q' for exit
RTPS Participant matched 1.f.3c.58.2e.7a.0.0.0.0.0.0|0.0.1.c1
RTPS Subscriber matched 1.f.3c.58.c0.2.0.0.0.0.0.0|0.0.2.4
RTPS Publisher matched 1.f.3c.58.c0.2.0.0.0.0.0.0|0.0.1.3
RTPS Participant matched 1.f.3c.58.dc.2.0.0.0.0.0.0|0.0.1.c1
RTPS Subscriber matched 1.f.3c.58.dc.2.0.0.0.0.0.0|0.0.1.4
q
Segmentation fault (core dumped)
```

Anyway, when closing the Agent with 'q' input, I wish to remove the topic entities managed by the Agent.

Is this functionality already in place, and is it not working because of segmentation fault errors? 
Or have not you added these features yet? If not, will you add it in the future?

julianbermudez, 2018-11-19:
Hi @OpusK, 

I'm going to debug this issue. Please, could you say me if it also happens with UDP transport of only with the Serial? Thanks!

OpusK, 2018-11-19:
Hi, @julianbermudez

It happens with both UDP and Serial.
And, I don't try with TCP.

julianbermudez, 2018-11-19:
Hi @OpusK, 

I'm not able to reproduce the `core dumped` exception, could you give me more details about the issue (which is the Client application, when you close the agent, ...). Anyway, I solved the entities destruction when the server is close by `q`.

OpusK, 2018-11-20:
@julianbermudez ,

I have tested the conditions in various environments and found the conditions to reproduce the problem.
The environment in which the problem occurs is as follows.
  - Application of ROS2 configuration on bash. (FastRTPS(I know it is not 1.7.0) of ROS2 is used)

In this case, an unconditional segmentation fault occurs at the end.
Maybe it is related to the issue below.
  - [[1.0.1 Version] Related to XML and entity deletion](https://github.com/eProsima/Micro-XRCE-DDS-Client/issues/52#issuecomment-440091649)

Will it be difficult to use with ROS2 until rmw of ROS2 is updated to 1.7.0?

<br>

In addition, the following phenomenon occurs when the ROS2 setting is not applied.
  - It is terminated normally immediately before the session is created.
  - If I enter 'q' after creating a session, participant, topic, writer or reader, the Agent will not quit and continue to print the log information as below. Of course, I can force it to terminate with Ctrl + C.

```bash
kei@ROBOTIS-9560:~$ MicroXRCEAgent serial /dev/ttyACM0
Serial agent initialization... OK
Enter 'q' for exit
RTPS Participant matched 1.f.3c.58.2d.28.0.0.0.0.0.0|0.0.1.c1
RTPS Participant matched 1.f.3c.58.d8.31.0.0.0.0.0.0|0.0.1.c1
RTPS Publisher matched 1.f.3c.58.d8.31.0.0.0.0.0.0|0.0.1.3
q
RTPS Publisher unmatched 1.f.3c.58.d8.31.0.0.0.0.0.0|0.0.1.3
RTPS Participant unmatched 1.f.3c.58.d8.31.0.0.0.0.0.0|0.0.1.c1
q
q
```

OpusK, 2018-11-21:
@julianbermudez 

As I mentioned in the following issue, I tried to change the version of rmw_fastrtps and fastrtps in ROS2 in the way @BorjaOuterelo told me, and then try to run the Agent.

 - [[1.0.1 Version] Related to XML and entity deletion.](https://github.com/eProsima/Micro-XRCE-DDS-Client/issues/52#issuecomment-440495326)

However, there were some problems in the compilation process, which was caused by the difference between the [submodule commit information](https://github.com/eProsima/Fast-RTPS/tree/d52a4a94e29cf50033bb74ae9d49cb4d1a98acee/include/fastrtps/participant) of the fastrtps currently used by the agent and the [commit information of the fastrtps release/1.7.0 branch](https://github.com/eProsima/Fast-RTPS/tree/release/1.7.0/include/fastrtps/participant).

The first error I saw was that the compiler could not find the header file due to a change in the include file location.
The header file is ParticipantDiscoveryInfo.h.
The header file no longer exists in the include path in the Agent code (from the release 1.7.0 branch).

Compiling and installing through the THIRDPARTY flag works fine, but when I work with the FastRTPS library installed on ROS2, another problem occurs as shown below.
```bash
$ MicroXRCEAgent serial /dev/ttyACM0
Serial agent initialization... OK
Enter 'q' for exit
RTPS Participant unmatched 3c.7f.0.0.2.2.1.f.3c.d2.f9.33|0.0.0.0
Segmentation fault (core dumped)
```

@BorjaOuterelo,
You said uROS Agent also uses Agent 1.0.1, do not you have the same problem as me?
Is it correct to match the commit version of fastrtps on ROS2 with the agent's submodule commit?

BorjaOuterelo, 2018-11-21:
Hi @OpusK.

Sorry for the misleading comments on discourse.ros.org.

>Is it correct to match the commit version of fastrtps on ROS2 with the agent's submodule commit?

I have been doing some further testing:

- ROS 2 Bouncy changed to use Fast RTPS commit used on Micro XRCE-DDS third-party leads to compilation errors due to ASIO incompatibility in Fast RTPS. There is a later commit (On Fast RTPS solving these incompatibilities).

>You said uROS Agent also uses Agent 1.0.1, do not you have the same problem as me?

We on micro-ROS don't have the same issues as If you check the micro-ROS repositories we use:

```
eProsima/Fast-RTPS:
    type: git
    url: https://github.com/eProsima/Fast-RTPS.git
    # micro-ROS-agent dependency.
    version: 9c6802b2155893b746bc0542645d9d2fc0c852d2
```

We are pointing to a commit in between 1.7.0 and the commit on the third party of Micro XRCE-DDS Agent. That in-between Fast RTPS commit is compatible with both Micro XRCE-DDS Agent and Bouncy ROS 2 packages and environment (Asio).

However, be careful because of Micro XRCE-DDS Agent v1.0.1 won't compile against Fast RTPS 1.7.0 (Sorry for confusing you here). My recommendation would be to use Fast-RTPS used on micro-ROS repositories list and upon official Fast-RTPS release 1.7.0 proceed to sync with that (We will make Micro XRCE-DDS Agent compatible with that release).

Note: micro-ROS repos on master have an error listing Fast-CDR on develop, that should be master.

OpusK, 2018-11-21:
Hi, @BorjaOuterelo,

Fine. I can make various attempts and take direction thanks to your comments.
As you mentioned, I will refer to the microROS repository and test it.

Can I get an approximate release schedule for the Agent and FastRTPS 1.7.0, including rmw_fastrtps?

Thanks again for your support :)

BorjaOuterelo, 2018-11-22:
> Can I get an approximate release schedule for the Agent and FastRTPS 1.7.0, including rmw_fastrtps?

Currently, the code is already frozen, and it is in the testing phase. It is on the ROS CI pipeline.
Our target is next week but depends on possible complications on the CI side.

OpusK, 2018-11-23:
@BorjaOuterelo,

Thanks for your information!
Anyway, when I run the agent alone, I check that the Entity is normally deleted at the end. 
So I will close this issue.

Thanks for your support! @julianbermudez, @BorjaOuterelo :)

closed 2018-11-23 by OpusK