eProsima/Micro-XRCE-DDS-Agent #31: [1.0.1 Version] Serial reader problem

OpusK, 2018-11-19:
Hi, 

I confirmed that both topic writer and reader work with UDP.

However, in the case of Serial, the topic writer works well, but the reader does not work well.
UDP works fine with the same code, but serial does not work well, so it raises the issue that the agent is having problems.

Below is the link of the code I used for the actual test. (This works with OpenCR, Arduino MKR Zero.)
 - [Arduino subscriber example](https://github.com/ROBOTIS-GIT/ros2arduino/blob/develop/examples/subscriber/subscriber.ino)

Topic and Reader are created normally, but Agent does not send any data to the client even if I send data to publish of ros2. On the other hand, when using UDP, the agent sends data to the client, and the callback is normally executed.

I would like to ask you to check if the reader works properly through the serial in the basic example.

julianbermudez, 2018-11-19:
Hi @OpusK, 

I'm going to debug this issue. I will create a `feature/serialdebug` branch, in both _Client_ and _Agent_ in order to test this example. I think the problem could be related to the custom `serial_protocol` that we use. I hope to give you an answer as soon as possible.

OpusK, 2018-11-20:
Thanks @julianbermudez ,

Still no data is transferred to Serial (USB).
However, when I test this, I use the publisher (writer) with ROS2, and I do not know if the same problem I mentioned in the issue below could lead to this phenomenon.
 - #28

OpusK, 2018-11-21:
@julianbermudez 

As I mentioned in the following issue, I tried to change the version of rmw_fastrtps and fastrtps in ROS2 in the way @BorjaOuterelo told me, and then try to run the Agent.

 - [[1.0.1 Version] Related to XML and entity deletion.](https://github.com/eProsima/Micro-XRCE-DDS-Client/issues/52#issuecomment-440495326)

However, there were some problems in the compilation process, which was caused by the difference between the [submodule commit information](https://github.com/eProsima/Fast-RTPS/tree/d52a4a94e29cf50033bb74ae9d49cb4d1a98acee/include/fastrtps/participant) of the fastrtps currently used by the agent and the [commit information of the fastrtps release/1.7.0 branch](https://github.com/eProsima/Fast-RTPS/tree/release/1.7.0/include/fastrtps/participant).

The first error I saw was that the compiler could not find the header file due to a change in the include file location.
The header file is ParticipantDiscoveryInfo.h.
The header file no longer exists in the include path in the Agent code (from the release 1.7.0 branch).

Compiling and installing through the THIRDPARTY flag works fine, but when I work with the FastRTPS library installed on ROS2, another problem occurs as shown below.
```bash
$ MicroXRCEAgent serial /dev/ttyACM0
Serial agent initialization... OK
Enter 'q' for exit
RTPS Participant unmatched 3c.7f.0.0.2.2.1.f.3c.d2.f9.33|0.0.0.0
Segmentation fault (core dumped)
```

@BorjaOuterelo,
You said uROS Agent also uses Agent 1.0.1, do not you have the same problem as me?
Is it correct to match the commit version of fastrtps on ROS2 with the agent's submodule commit?

OpusK, 2018-12-11:
Hi, @julianbermudez 

I have now confirmed that the mechanism of the callback function has changed.
Previously, topic_id was received with object_id, and now reader_id is received.

Anyway, I confirmed that it works normally.

I'm sorry to bother you. Thanks for your support!

closed 2018-12-11 by OpusK