eProsima/Micro-XRCE-DDS-Agent #41: Multiple Agent

OpusK, 2018-12-19:
Hi,

 - Micro-XRCE-DDS Agent 1.0.1 (feature/serialdebug)
 - Micro-XRCE-Client 1.0.1
 - ROS2 Crystal

When I try to run the agent using serial first and then udp, the following error occurs.

[image]

It simply says ERROR.
Conversely, if I run udp first and then serial, I can run both, but no Topic will be generated.

julianbermudez, 2018-12-19:
Hi @OpusK, 

I fixed this issue in feature/serialdebug. The problem was that Fast RTPS made use of the port 7400, and this is the discovery port established by the XRCE. Now both Agent and Fast RTPS can shared this port. I hope it helps you!

OpusK, 2018-12-20:
Hi, @julianbermudez 

Thanks for your support!
Resolved!

closed 2018-12-20 by OpusK