eProsima/Micro-XRCE-DDS-Agent #59: [ROS2, Agent, Client] fastcdr : Not enough memory in the buffer stream

OpusK, 2019-02-26:
Hi.

Since this issue is related to ros2, I was wondering whether I should leave an issue in rmw or leave an issue in the agent.
Anyway, I am using the Agent, so I leave an issue here.
Please let me know if you have a better location.

---

I am now facing the following error.

```bash
$ ros2 topic echo /arduino_tf2 

terminate called after throwing an instance of 'eprosima::fastcdr::exception::NotEnoughMemoryException'
  what():  Not enough memory in the buffer stream
Aborted (core dumped)
```

The current relationship is as follows.
- `ESP32(client)` - `Router` - `Agent` - `ROS2(fastrtps)`

Here, I attached a file capturing the logs of the data sent by the Client to the Agent as wireshark.
[wireshark_tf2.pcapng.zip](https://github.com/eProsima/Micro-XRCE-DDS-Agent/files/2904473/wireshark_tf2.pcapng.zip)

What should I do?

OpusK, 2019-07-02:
I'll change the question.

How do I handle the following special types of sequences?
 - [TFMessage](https://github.com/ros2/geometry2/blob/dashing/tf2_msgs/msg/TFMessage.msg)

I was able to make up to  `geometry_msgs/TransformStamped` that I use in the message above.

However, in order to handle this as a sequence and transmit it, I think it is necessary to add the number of this sequence, that is, the number of arrays. What should I do?

When I use the `ucdr_serialize_uint32_t` function to serialize, I add the number of the array to the front, which still causes the same error as the previous issue.

I think this is happening because I do not know. How do I convert these IDL files?

julibert, 2020-03-10:
Hi @OpusK,

Does this issue persist in new Micro XRCE-DDS Agent versions?

OpusK, 2020-03-11:
Hi, @julibert 

I have not been testing recently on this.
If I have a chance to test it later, I will create additional issues at that time.

Thank you :)

closed 2020-03-11 by OpusK

Jeisongarcia9713, 2021-01-26:
i have the same problem working with olimex E407, with a  temperature sensor mesagge anyone, could solve this?