eProsima/Micro-XRCE-DDS-Agent #63: FastRTPS(master branch) API problem

routiful, 2019-03-12:
I've got message which shows build fail from this repo after install ROS2.
I assume that the FastRTPS API might be modified.

https://github.com/eProsima/Micro-XRCE-DDS-Agent/blob/a495c65faa964ddc068ac6e1249f17f5c9f92787/src/cpp/Root.cpp#L42

julianbermudez, 2019-03-13:
Hi @routiful,

Yes, it could be due to a change in Fast RTPS API.
Could you say me which version of ROS2 you are using?

routiful, 2019-03-13:
@julianbermudez 

Thanks you for reply.
I was downloaded source files to use wget command. 
The repos files in master branch includes Fast-RTPS master branch

https://raw.githubusercontent.com/ros2/ros2/master/ros2.repos

julianbermudez, 2019-03-18:
Hi @routiful,

We have already merged this issue into the master branch. Now the Agent is compatible with Fast RTPS version 1.7.2.

routiful, 2019-03-19:
Wonderful :)
Thanks for handling it quickly!

closed 2019-03-19 by routiful