eProsima/Micro-XRCE-DDS-Agent #66: No topic from MicroXRECAgent after set ROS_DOMAIN_ID

yechun1, 2019-04-02:
We have two turtlebot3 in the same network, we want to set domain id to separate two robots, so that they will not impact each other's topics. But after set ROS_DOMAIN_ID, there is no topic lists published by MicroXRCEAgent. 

**Test 1:** (this is correct)
>  $ cd ~/turtlebot3 && MicroXRCEAgent serial /dev/ttyACM0
>  $ export ROS_DOMAIN_ID=0
>  $ ros2 topic list
```
/cmd_vel
/imu
/motor_power
/parameter_events
/reset
/rosout
/sensor_state
/sound
/time_sync
/version_info
```

**Test 2:** (The topic lists are not as expected)
>  $ export ROS_DOMAIN_ID=1
>  $ cd ~/turtlebot3 && MicroXRCEAgent serial /dev/ttyACM0
>  $ export ROS_DOMAIN_ID=1
>  $ ros2 topic list
```
/parameter_events
/rosout
```
For the test2, there is not cmd_vel, imu.., and other topics, this is not as expected.

julianbermudez, 2019-04-02:
Hi @yechun1,

Just a simple question, are you creating the XRCE object by reference or by XML?

In case you use the reference representation, you should change the _agent.refs_ to select the domainId.
According to the Fast RTPS [documentation](https://fast-rtps.docs.eprosima.com/en/latest/xmlprofiles.html) it is done adding the following to the participant profile:

```
<participant .....>
    <rtps>
        .....
        <builtin>
            <domainId>X</domainId>
        </buitin>
    </rtps>
</participant>
```

In case of XML representation, you should modify the XML in that sense also.

Is it solved your problem?

yechun1, 2019-04-03:
Hi, @julianbermudez,
Thanks, I have used the default XML, after added domainId in it, the issue still exist. 

```
$ cat ~/turtlebot3/DEFAULT_FASTRTPS_PROFILES.xml 
<profiles>
    <participant profile_name="default_xrce_participant">
        <rtps>
            <name>default_xrce_participant</name>
            <builtin>
                <domainId>4</domainId>
            </builtin>
        </rtps>
    </participant>
    <data_writer profile_name="shapetype_data_writer">
        <topic>
            <kind>WITH_KEY</kind>
            <name>Square</name>
            <dataType>ShapeType</dataType>
        </topic>
    </data_writer>
    <data_reader profile_name="shapetype_data_reader">
        <topic>
            <kind>WITH_KEY</kind>
            <name>Square</name>
            <dataType>ShapeType</dataType>
        </topic>
    </data_reader>
    <topic profile_name="shapetype_topic">
        <kind>WITH_KEY</kind>
        <name>Square</name>
        <dataType>ShapeType</dataType>
    </topic>
</profiles>
```

```
chris@yechun1:~/turtlebot3$ MicroXRCEAgent serial /dev/ttyACM0
Serial agent initialization... OK
```

```
$ export ROS_DOMAIN_ID=4
$ ros2 topic list
/parameter_events
/rosout
```

`$ ps -aux |grep ros2`

> chris     8370  0.2  0.2 420368 39920 pts/1    Sl   10:29   0:00 /usr/bin/python3 /data/proj/modules/robotsdk/robot_sdk.build/build/install/ros2cli/bin/_ros2_daemon --rmw-implementation rmw_fastrtps_cpp --ros-domain-id 4

julianbermudez, 2019-04-03:
Hi @yechun1,

I suppose you followed this [instruction](http://emanual.robotis.com/docs/en/platform/turtlebot3/ros2/#ros2) to run the TurtleBot3, is it correct?

One question, do you export the domain id when you launch the TurtleBot3? That is,

```
$ export ROS_DOMAIN_ID=X
$ export TURTLEBOT3_MODEL=${TB3_MODEL}
$ ros2 launch turtlebot3_bringup robot.launch.py
```

yechun1, 2019-04-10:
Hi @julianbermudez,
Yes, we followed the instruction from http://emanual.robotis.com/docs/en/popup/turtlebot3_ros2_sbc_setting/
and added domain id in` DEFAULT_FASTRTPS_PROFILES.xml`
```
     <builtin>
                <domainId>4</domainId>
      </builtin>
```
I exported the domain id when launch trutlebot3. The domainID in xml did not fix the issue.

OpusK, 2019-04-10:
@yechun1 

As described in @julianbermudez, this is an issue caused by the implementation of ros2arduino in TB3.
The detailed answer to this question answered the issue you left in ros2xrcedds.

closed 2019-06-12 by julianbermudez

lxc777, 2022-04-12:
@yechun1 I would like to ask this question. Now you can directly modify the configuration file to realize ROS_ DOMAIN_ ID setting?

lxc777, 2022-04-12:
Hi. @julionce   When I try to set ROS_ DOMAIN_ When 
[image]
[image]

Ryanf55, 2024-05-22:
Hey there, 

What do you think about adding a warning on bootup if it detects ROS_DOMAIN_ID is set by the environment AND you launch by a refs file, it should issue  a warning that the environment variable is ignored? We have gotten numerous support requests in ArduPilot from people who don't know you need to do this when they have configured a non-default domain ID globally on their laptop to not conflict with their lab mates or coworkers.