eProsima/Micro-XRCE-DDS-Agent #70: Turtlebot3 not work while Micro-XRCE-DDS-Agent install again.

congliu0913, 2019-04-09:
**reproduce step:**
* First-time installation
```
$ sudo apt update && sudo apt install build-essential cmake git

$ cd test1 && git clone https://github.com/eProsima/Micro-XRCE-DDS-Agent.git
$ cd Micro-XRCE-DDS-Agent && git checkout a495c65faa964ddc068ac6e1249f17f5c9f92787
$ mkdir build && cd build
$ cmake -DTHIRDPARTY=ON -DCONFIG_UDP_TRANSPORT_MTU=8192 -DCONFIG_SERIAL_TRANSPORT_MTU=8192 ..
$ sudo make install
$ mkdir ~/turtlebot3_1
$ cp ../DEFAULT_FASTRTPS_PROFILES.xml ~/turtlebot3_1
$ sudo ldconfig /usr/local/lib/
```
* Running successfully
```
$ cd ~/turtlebot3_1 && MicroXRCEAgent serial /dev/ttyACM0
Serial agent initialization... OK
Enter 'q' for exit
RTPS Participant matched 1.f.59.12.60.65.0.0.1.0.0.0|0.0.1.c1
RTPS Participant matched 1.f.59.99.1f.59.0.0.1.0.0.0|0.0.1.c1
RTPS Participant matched 1.f.59.99.24.59.0.0.1.0.0.0|0.0.1.c1
```
* Second installation
```
$ cd test2 && git clone https://github.com/eProsima/Micro-XRCE-DDS-Agent.git
$ cd Micro-XRCE-DDS-Agent && git checkout a495c65faa964ddc068ac6e1249f17f5c9f92787
$ mkdir build && cd build
$ cmake -DTHIRDPARTY=ON -DCONFIG_UDP_TRANSPORT_MTU=8192 -DCONFIG_SERIAL_TRANSPORT_MTU=8192 ..
$ sudo make install
$ mkdir ~/turtlebot3_2
$ cp ../DEFAULT_FASTRTPS_PROFILES.xml ~/turtlebot3_2
$ sudo ldconfig /usr/local/lib/
```
* Running failed
```
$ cd ~/turtlebot3_2 && MicroXRCEAgent serial /dev/ttyACM0
Serial agent initialization... OK
Enter 'q' for exit
Segmentation fault (core dumped)
```
I accidentally discovered this problem, the above is my experimental steps.

system info:	Ubuntu 18.04.2 LTS
link: http://emanual.robotis.com/docs/en/platform/turtlebot3/ros2/#sbc-setup

yechun1, 2019-04-09:
I met the same issue when rebuild MicroXRCEAgent. MicroXRCEAgent could not work after delete `build` folder and run cmake/make again.

OpusK, 2019-04-09:
Hi, @congliu0913, @yechun1 

What is ros2 version you using?
Depending on the ros2 version, the version of FastRTPS may be different. Also, depending on this, the version of the Agent is also important.

`git checkout a495c65faa964ddc068ac6e1249f17f5c9f92787`
Do not use the above command, but use it according to ros2 version.
Please refer to the README below for links.
- [ros2arduino](https://github.com/ROBOTIS-GIT/ros2arduino#version-specific-dependencies)

yechun1, 2019-04-09:
@OpusK 
Actually, We did not source ros2 environment while build Agent. The agent will download and build FastRTPS itself and build, on first time build, the Agent works, but second time build, the Agent will Segmentation fault.

We follow the guide from [turtlebot3 manual](http://emanual.robotis.com/docs/en/popup/turtlebot3_ros2_sbc_setting/)

congliu0913, 2019-04-10:
I don't know if this is my environment causing it!
I found this information.
```
$ ldd /usr/local/lib/libmicroxrcedds_agent.so
	linux-vdso.so.1 (0x00007ffceffa3000)
	libfastrtps.so.1 => /usr/local/lib/libfastrtps.so.1 (0x00007fe0e5a41000)
	libfastcdr.so.1 => /usr/local/lib/libfastcdr.so.1 (0x00007fe0e582e000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007fe0e560f000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007fe0e5286000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007fe0e506e000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fe0e4c7d000)
	/lib64/ld-linux-x86-64.so.2 (0x00007fe0e635a000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007fe0e4a79000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007fe0e46db000)
```
```
$ ls -l  /usr/local/lib/libfastcdr.so.1
 /usr/local/lib/libfastrtps.so.1 -> libfastrtps.so.1.7.0
$ ldd /usr/local/lib/libfastcdr.so.1
	linux-vdso.so.1 (0x00007ffd2b82a000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f87cd72a000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f87cd512000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f87cd121000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f87ccd83000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f87cdcc6000)
```
```
$ ls -l /usr/local/lib/libfastcdr.so.1
/usr/local/lib/libfastcdr.so.1 -> libfastcdr.so.1.0.8
$ ldd /usr/local/lib/libfastcdr.so.1
	linux-vdso.so.1 (0x00007fff587d8000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007fae53c42000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007fae53a2a000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fae53639000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007fae5329b000)
	/lib64/ld-linux-x86-64.so.2 (0x00007fae541de000)
```

congliu0913, 2019-04-10:
```
ldd /usr/local/bin/MicroXRCEAgent
	linux-vdso.so.1 (0x00007ffd989fb000)
	libmicroxrcedds_agent.so.1 => /usr/local/lib/libmicroxrcedds_agent.so.1 (0x00007f8bb2eae000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f8bb2c8f000)
	libfastcdr.so.1 => /usr/local/lib/libfastcdr.so.1 (0x00007f8bb2a7c000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f8bb26f3000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f8bb24db000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f8bb20ea000)
	libfastrtps.so.1 => /usr/local/lib/libfastrtps.so.1 (0x00007f8bb1a89000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f8bb336e000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f8bb16eb000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f8bb14e7000)
```

yechun1, 2019-04-10:
I could easily reproduce this issue on my local. Same result while use the commit  `b3c43bd20b16bd3570a128f05a9bf1a164883435` from ros2arduino. 

On first time build, libfastrtps and libfastcdc will be built and installed to `/usr/local/lib`,  second time build, the agent will use those two libraries under` /usr/local/lib`, just delete `build` folder and rebuild, the MicroXRCEAgent will have different behavior. Not found Segmentation fault this time, but no topic published by Agent. When delete libraries in `/usr/local`, the rebuild and execution will work fine.

 We sometimes want to uninstall or reinstall MicroXRCEAgent, is it possible to add "make uninstall" to delete the files listed in `install_manifest.txt` ?

Here are steps we run:
```
# clean files, delete previously installed libraries 
find /usr/local/ |grep fastrtps | sudo xargs rm -rf
find /usr/local/ |grep fastcdr | sudo xargs rm -rf
find /usr/local/ |grep microxrcedds_agent | sudo xargs rm -rf

# first build 
 cd && git clone https://github.com/eProsima/Micro-XRCE-DDS-Agent.git
 cd Micro-XRCE-DDS-Agent && git checkout a495c65faa964ddc068ac6e1249f17f5c9f92787
 mkdir build && cd build
 cmake -DTHIRDPARTY=ON -DCONFIG_UDP_TRANSPORT_MTU=8192 -DCONFIG_SERIAL_TRANSPORT_MTU=8192 ..
 sudo make install
 mkdir ~/turtlebot3
 cp ../DEFAULT_FASTRTPS_PROFILES.xml ~/turtlebot3
 sudo ldconfig /usr/local/lib/

# second build
 cd Micro-XRCE-DDS-Agent
rm -rf build
mkdir build && cd build
 cmake -DTHIRDPARTY=ON -DCONFIG_UDP_TRANSPORT_MTU=8192 -DCONFIG_SERIAL_TRANSPORT_MTU=8192 ..
 sudo make install
 mkdir ~/turtlebot3
 cp ../DEFAULT_FASTRTPS_PROFILES.xml ~/turtlebot3
 sudo ldconfig /usr/local/lib/

```
the result is no topic published by MicroXRCEAgent

$ cd ~/turtlebot3 && MicroXRCEAgent serial /dev/ttyACM0
```
Serial agent initialization... OK
Enter 'q' for exit
```
$ ros2 topic list
```
/parameter_events
/rosout
```

julianbermudez, 2019-04-22:
Hi @congliu0913 and @yechun1,

Your environment looks properly. Even though I was not able to reproduce your issue in a clean Docker. Did you try in a clean Docker?
We will add an uninstall for the next release, as well as, we will change the `find_package` in order to link with a specific version of Fast RTPS and Fast CDR.

julianbermudez, 2019-07-01:
Hi @congliu0913 and @yechun1,

We launched a new release 1.1.0 in which the handling of dependencies has been changed.
I close the issue, please reopen it in case of this error persists in the new release.

closed 2019-07-01 by julianbermudez