eProsima/Micro-XRCE-DDS-Agent #81: Do you have documentation on the detailed protocols of the Agent and Client?

OpusK, 2019-05-20:
Hi,

Using Client and Agent, I am experiencing some problems.
The environment that I mainly use is a low-end MCU, and it applies to both environments that use or do not use RTOS (FreeRTOS).
I use Serial (USB) and UDP.

Simple configuration (one writer) does not cause any problems.
However, when I use several DataWriters and DataReaders in one participant, I sometimes have problems communicating with the Agent. (This phenomenon is random depending on the environment.)
I want to know the cause of this and I want to find ways to reduce this phenomenon.

This does not mean that the MCU itself is in trouble. In particular, when I checked the packet with Wireshark, I could see from the moment that the DataWriter packet was not seen, but only the packets like Heartbeat were sent between the Client and the Agent.

I believe that if there is a problem with mutual communication according to the mutual protocol, there will be such a process or other causes.

Can I get information about these detailed protocols in OMG's XRCE-DDS documentation? Do MicroXRCE-DDS-Agent and Client fully observe this document? Do not you include anything not in this document? If so, can I get relevant information?

julianbermudez, 2019-06-05:
Hi @OpusK, 

Sorry for the delay but we are launching a new release (1.1.0) and it consumes most of our time.

> However, when I use several DataWriters and DataReaders in one participant, I sometimes have problems communicating with the Agent. (This phenomenon is random depending on the environment.)
I want to know the cause of this and I want to find ways to reduce this phenomenon.

Do you find this issue in every environment? In which environment do you find it more frequently?

> This does not mean that the MCU itself is in trouble. In particular, when I checked the packet with Wireshark, I could see from the moment that the DataWriter packet was not seen, but only the packets like Heartbeat were sent between the Client and the Agent.

Could you share with us the Wireshark result?

> Can I get information about these detailed protocols in OMG's XRCE-DDS documentation? Do MicroXRCE-DDS-Agent and Client fully observe this document? Do not you include anything not in this document? If so, can I get relevant information?

Yes, Micro XRCE-DDS is the first implementation of the DDS-XRCE wire protocol.
 We support most of the standard except for some minor features like the `BINARY_REPRESENTATION`.
On the other hand, Micro XRCE-DDS does not include any extra feature apart from the aforementioned standard.

OpusK, 2019-06-11:
@julianbermudez 

After testing with 1.1.0, I will reply back here.

OpusK, 2019-06-13:
When I tested it with ROS2 Dashing, pre-release 1.1.0, Client 1.1.1, I saw that it worked well after one day.

The version of several things was updated at once, so it is hard to know where the problem was. If the problem recurs later, I will raise the issue again.

Thank you :)

closed 2019-06-13 by OpusK