eProsima/Micro-XRCE-DDS-Agent #88: CentOS / RedHat support

raveious, 2019-06-03:
When compiling the source on CentOS or RedHat, GCC is configured to compile against C90 and not C99, causing errors while compiling. This can be fixed by explicitly setting the standard in the cmake, forcing the use of C99.

This issue also effects microxrceddsgen as it builds and tests the agent while building the generator, which is how I found the problem.

julianbermudez, 2019-06-03:
Hi @raveious, 

I added the C required version to the new [release 1.1.0](https://github.com/eProsima/Micro-XRCE-DDS-Agent/tree/release/1.1.0) which will be available during this week.

raveious, 2019-06-03:
I'm now realizing that this actually should have been against the client, and not the agent. The patch seems to still be release 1.1.0

julianbermudez, 2019-07-01:
Hi @raveious,

It should be resolved in the release 1.1.0 of the Client.
I close the issue, please reopen in the Client side in case the issue persists.

closed 2019-07-01 by julianbermudez