eProsima/Micro-XRCE-DDS-Agent #91: [Agent 1.1.0] XML reference error.

OpusK, 2019-06-11:
Hi,

The xml file that I used well in 1.0.3 has the following error in 1.1.0.
If there is anything I am using wrongly, I would be grateful to let me know.

```
kei@kei-NUC8i7BEH:~$ MicroXRCEAgent udp -p 2019 -v 6 -r tb3_fastrtps_profile.refs 
Enter 'q' for exit
[1560231733.214351] info     | UDPServerLinux.cpp | init                     | running...             | port: 2019
2019-06-11 14:42:13.214 [XMLPARSER Error] Invalid element found into 'publisherProfileType'. Name: publisher -> Function fillDataNode
2019-06-11 14:42:13.214 [XMLPARSER Error] Error parsing publisher profile -> Function parseXMLPublisherProf
2019-06-11 14:42:13.214 [XMLPARSER Error] Error parsing profile's tag data_writer -> Function parseProfiles
2019-06-11 14:42:13.214 [XMLPARSER Error] Error parsing 'tb3_fastrtps_profile.refs' -> Function loadXMLFile
[1560231733.214912] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6
[1560231739.532029] debug    | UDPServerLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x00000000, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F AA BB CC DD 81 00 FC 07
[1560231739.532441] info     | Root.cpp           | create_client            | create                 | client_key: 0xAABBCCDD, session_id: 0x81
[1560231739.532594] info     | UDPServerBase.cpp  | on_create_client         | session established    | client_key: 0xAABBCCDD, address: 192.168.0.3:47138
[1560231739.532899] debug    | UDPServerLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0xAABBCCDD, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1560231739.545786] debug    | UDPServerLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0xAABBCCDD, len: 48, data: 
0000: 81 80 00 00 01 05 28 00 00 0A 00 11 01 01 00 00 19 00 00 00 72 6F 73 32 5F 78 72 63 65 64 64 73
0020: 5F 70 61 72 74 69 63 69 70 61 6E 74 00 00 00 00
2019-06-11 14:42:19.546 [XMLPARSER Error] Profile 'ros2_xrcedds_participant' not found ' -> Function fillParticipantAttributes[1560231739.546591] 
2019-06-11 14:42:19.546 [PARTICIPANT Error] Problem loading profile 'ros2_xrcedds_participant' -> Function createParticipant
debug    | ProxyClient.cpp    | create_participant       | dds error              | client_key: 0xAABBCCDD, participant_id: 0x001(1)
[1560231739.547157] debug    | UDPServerLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0xAABBCCDD, len: 14, data: 
0000: 81 80 00 00 05 01 06 00 00 0A 00 11 80 00
[1560231739.547272] debug    | UDPServerLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0xAABBCCDD, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1560231739.556661] debug    | UDPServerLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0xAABBCCDD, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
```

- xml file.
```
<profiles>
    <participant profile_name="ros2_xrcedds_participant">
        <rtps>
            <name>ros2_xrcedds_participant</name>
            <builtin>
                <domainId>0</domainId>
            </builtin>
        </rtps>
    </participant>

    <data_writer profile_name="sensor_state">
        <topic>
            <kind>NO_KEY</kind>
            <name>rt/sensor_state</name>
            <dataType>turtlebot3_msgs::msg::dds_::SensorState_</dataType>
            <historyQos>
                <kind>KEEP_LAST</kind>
                <depth>10</depth>
            </historyQos>
        </topic>
    </data_writer>

    <data_writer profile_name="imu">
        <topic>
            <kind>NO_KEY</kind>
            <name>rt/imu</name>
            <dataType>sensor_msgs::msg::dds_::Imu_</dataType>
            <historyQos>
                <kind>KEEP_LAST</kind>
                <depth>10</depth>
            </historyQos>
        </topic>
    </data_writer>

    <data_writer profile_name="scan_half">
        <topic>
            <kind>NO_KEY</kind>
            <name>rt/scan_half</name>
            <dataType>sensor_msgs::msg::dds_::LaserScan_</dataType>
            <historyQos>
                <kind>KEEP_LAST</kind>
                <depth>10</depth>
            </historyQos>
        </topic>
        <publisher>
            <historyMemoryPolicy>PREALLOCATED_WITH_REALLOC</historyMemoryPolicy>
        </publisher>
    </data_writer>

    <data_writer profile_name="version_info">
        <topic>
            <kind>NO_KEY</kind>
            <name>rt/version_info</name>
            <dataType>turtlebot3_msgs::msg::dds_::VersionInfo_</dataType>
            <historyQos>
                <kind>KEEP_LAST</kind>
                <depth>10</depth>
            </historyQos>
        </topic>
    </data_writer>

    <data_reader profile_name="cmd_vel">
        <topic>
            <name>rt/cmd_vel</name>
            <dataType>geometry_msgs::msg::dds_::Twist_</dataType>
        </topic>
    </data_reader>

    <data_reader profile_name="motor_power">
        <topic>
            <name>rt/motor_power</name>
            <dataType>std_msgs::msg::dds_::Bool_</dataType>
        </topic>
    </data_reader>

    <data_reader profile_name="reset">
        <topic>
            <name>rt/reset</name>
            <dataType>std_msgs::msg::dds_::Empty_</dataType>
        </topic>
    </data_reader>

    <data_reader profile_name="time_sync">
        <topic>
            <name>rt/time_sync</name>
            <dataType>builtin_interfaces::msg::dds_::Time_</dataType>
        </topic>
    </data_reader>

    <topic profile_name="Imu">
        <kind>NO_KEY</kind>
        <name>Imu</name>
        <dataType>sensor_msgs::msg::dds_::Imu_</dataType>
    </topic>

    <topic profile_name="LaserScan">
        <kind>NO_KEY</kind>
        <name>LaserScan</name>
        <dataType>sensor_msgs::msg::dds_::LaserScan_</dataType>
    </topic>

    <topic profile_name="Twist">
        <kind>NO_KEY</kind>
        <name>Twist</name>
        <dataType>geometry_msgs::msg::dds_::Twist_</dataType>
    </topic>

    <topic profile_name="SensorState">
        <kind>NO_KEY</kind>
        <name>SensorState</name>
        <dataType>turtlebot3_msgs::msg::dds_::SensorState_</dataType>
    </topic>

    <topic profile_name="Bool">
        <kind>NO_KEY</kind>
        <name>Bool</name>
        <dataType>std_msgs::msg::dds_::Bool_</dataType>
    </topic>

    <topic profile_name="Empty">
        <kind>NO_KEY</kind>
        <name>Empty</name>
        <dataType>std_msgs::msg::dds_::Empty_</dataType>
    </topic>

    <topic profile_name="Time">
        <kind>NO_KEY</kind>
        <name>Time</name>
        <dataType>builtin_interfaces::msg::dds_::Time_</dataType>
    </topic>

    <topic profile_name="VersionInfo">
        <kind>NO_KEY</kind>
        <name>VersionInfo</name>
        <dataType>turtlebot3_msgs::msg::dds_::VersionInfo_</dataType>
    </topic>

    <participant profile_name="template_participant">
        <rtps>
            <name>template_participant</name>
        </rtps>
    </participant>

    <data_writer profile_name="template_writer">
        <topic>
            <kind>WITH_KEY</kind>
            <name>topic_name</name>
            <dataType>topic_type</dataType>
        </topic>
    </data_writer>

    <data_reader profile_name="template_reader">
        <topic>
            <kind>WITH_KEY</kind>
            <name>topic_name</name>
            <dataType>topic_type</dataType>
        </topic>
    </data_reader>

    <topic profile_name="template_topic">
        <kind>WITH_KEY</kind>
        <name>topic_name</name>
        <dataType>topic_type</dataType>
    </topic>
</profiles>
```

julianbermudez, 2019-06-11:
Hi @OpusK,

At first glance, it seems that the `data_writer` profile `scan_half` is not correct. Please remove the tag `publisher` and try again. Anyway, I will try to reproduce the issue.

OpusK, 2019-06-11:
Hi, @julianbermudez 

Resolved! Thank you :)

But should I use this profile to increase the Agent's memory?
(https://github.com/eProsima/Micro-XRCE-DDS-Agent/issues/4#issuecomment-418011213)

julianbermudez, 2019-06-11:
Hi @OpusK,

Yes, you should use it to increase the Fast RTPS buffers, just remove the `<publisher>` and `</publisher>` tags.

OpusK, 2019-06-11:
@julianbermudez 

Thanks for your reply.
I'll test this later :)

closed 2019-06-11 by OpusK