eProsima/Micro-XRCE-DDS-Agent #92: [Agent 1.1.0] geometry::Vector3 message not work.

OpusK, 2019-06-11:
Hi,

I was using messages from ros2 generated by IDL Generator 1.0.0 very well.
However, since the 1.1.0 update, there is a phenomenon that the registered reader callback is invoked infinitely (almost 200 times per second?) With the infinite repetition packet as below using geometry::Vector3 message.
And the same thing happens in messages like Twist using this Vector3.

I have not tested every message, but std_msgs::Bool and std_msgs::Empty work fine.
Has the IDL conversion method changed?

```
kei@kei-NUC8i7BEH:~$ MicroXRCEAgent udp -p 2019 -v 6
Enter 'q' for exit
[1560239183.121138] info     | UDPServerLinux.cpp | init                     | running...             | port: 2019
[1560239183.121282] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6
[1560239203.909175] debug    | UDPServerLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x00000000, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F AA BB CC DD 81 00 FC 07
[1560239203.909739] info     | Root.cpp           | create_client            | create                 | client_key: 0xAABBCCDD, session_id: 0x81
[1560239203.909933] info     | UDPServerBase.cpp  | on_create_client         | session established    | client_key: 0xAABBCCDD, address: 192.168.0.5:47138
[1560239203.910277] debug    | UDPServerLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0xAABBCCDD, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1560239203.924143] debug    | UDPServerLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0xAABBCCDD, len: 112, data: 
0000: 81 80 00 00 01 05 68 00 00 0A 00 11 01 02 00 00 59 00 00 00 3C 64 64 73 3E 3C 70 61 72 74 69 63
0020: 69 70 61 6E 74 3E 3C 72 74 70 73 3E 3C 6E 61 6D 65 3E 72 6F 73 32 5F 78 72 63 65 64 64 73 5F 70
0040: 61 72 74 69 63 69 70 61 6E 74 3C 2F 6E 61 6D 65 3E 3C 2F 72 74 70 73 3E 3C 2F 70 61 72 74 69 63
0060: 69 70 61 6E 74 3E 3C 2F 64 64 73 3E 00 00 00 00
[1560239203.928240] debug    | ProxyClient.cpp    | create_participant       | participant created    | client_key: 0xAABBCCDD, participant_id: 0x001(1)
[1560239203.928543] debug    | UDPServerLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0xAABBCCDD, len: 14, data: 
0000: 81 80 00 00 05 01 06 00 00 0A 00 11 00 00
[1560239203.928577] debug    | UDPServerLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0xAABBCCDD, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1560239203.929208] trace    | FastEntities.cpp   | onParticipantDiscovery   | matched                | entity_id: 0.0.1.c1, guid_prefix: 1.f.3b.9f.3a.1.0.0.1.0.0.0
[1560239203.929751] trace    | FastEntities.cpp   | onParticipantDiscovery   | matched                | entity_id: 0.0.1.c1, guid_prefix: 1.f.3b.9f.7d.7d.0.0.1.0.0.0
[1560239203.956886] debug    | UDPServerLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0xAABBCCDD, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1560239203.957243] debug    | UDPServerLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0xAABBCCDD, len: 24, data: 
0000: 81 80 01 00 01 05 0F 00 00 0B 00 13 03 02 00 00 01 00 00 00 00 00 11 00
[1560239203.957357] debug    | ProxyClient.cpp    | create_publisher         | publisher created      | client_key: 0xAABBCCDD, publisher_id: 0x001(3), participant_id: 0x001(1)
[1560239203.957420] debug    | UDPServerLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0xAABBCCDD, len: 14, data: 
0000: 81 80 01 00 05 01 06 00 00 0B 00 13 00 00
[1560239203.957446] debug    | UDPServerLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0xAABBCCDD, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 02 00 00 00 80
[1560239203.961538] debug    | UDPServerLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0xAABBCCDD, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 02 00 00 00 80
[1560239203.963709] debug    | UDPServerLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0xAABBCCDD, len: 24, data: 
0000: 81 80 02 00 01 05 0F 00 00 0C 00 14 04 02 00 00 01 00 00 00 00 00 11 73
[1560239203.963842] debug    | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0xAABBCCDD, subscriber_id: 0x001(4), participant_id: 0x001(1)
[1560239203.963953] debug    | UDPServerLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0xAABBCCDD, len: 14, data: 
0000: 81 80 02 00 05 01 06 00 00 0C 00 14 00 00
[1560239203.963990] debug    | UDPServerLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0xAABBCCDD, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 03 00 00 00 80
[1560239203.968653] debug    | UDPServerLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0xAABBCCDD, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 03 00 00 00 80
[1560239203.971956] debug    | UDPServerLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0xAABBCCDD, len: 124, data: 
0000: 81 80 03 00 01 03 74 00 00 0D 01 32 02 02 00 00 66 00 00 00 3C 64 64 73 3E 3C 74 6F 70 69 63 3E
0020: 3C 6E 61 6D 65 3E 56 65 63 74 6F 72 33 3C 2F 6E 61 6D 65 3E 3C 64 61 74 61 54 79 70 65 3E 67 65
0040: 6F 6D 65 74 72 79 5F 6D 73 67 73 3A 3A 6D 73 67 3A 3A 64 64 73 5F 3A 3A 56 65 63 74 6F 72 33 5F
0060: 3C 2F 64 61 74 61 54 79 70 65 3E 3C 2F 74 6F 70 69 63 3E 3C 2F 64 64 73 3E 00 00 11
[1560239203.972281] debug    | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0xAABBCCDD, topic_id: 0x013(2), participant_id: 0x001(1)
[1560239203.972418] debug    | UDPServerLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0xAABBCCDD, len: 14, data: 
0000: 81 80 03 00 05 01 06 00 00 0D 01 32 00 00
[1560239203.972467] debug    | UDPServerLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0xAABBCCDD, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 04 00 00 00 80
[1560239203.976369] debug    | UDPServerLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0xAABBCCDD, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 04 00 00 00 80
[1560239203.979457] debug    | UDPServerLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0xAABBCCDD, len: 184, data: 
0000: 81 80 04 00 01 05 AE 00 00 0E 00 16 06 02 00 00 A0 00 00 00 3C 64 64 73 3E 3C 64 61 74 61 5F 72
0020: 65 61 64 65 72 3E 3C 74 6F 70 69 63 3E 3C 6B 69 6E 64 3E 4E 4F 5F 4B 45 59 3C 2F 6B 69 6E 64 3E
0040: 3C 6E 61 6D 65 3E 72 74 2F 61 72 64 75 69 6E 6F 5F 63 61 6C 6C 62 61 63 6B 3C 2F 6E 61 6D 65 3E
0060: 3C 64 61 74 61 54 79 70 65 3E 67 65 6F 6D 65 74 72 79 5F 6D 73 67 73 3A 3A 6D 73 67 3A 3A 64 64
0080: 73 5F 3A 3A 56 65 63 74 6F 72 33 5F 3C 2F 64 61 74 61 54 79 70 65 3E 3C 2F 74 6F 70 69 63 3E 3C
[... 97 lines cut ...]
    return !writer->error;
  }

  bool deserialize(void* msg_buf, Vector3* topic)
  {
    ucdrBuffer* reader = (ucdrBuffer*)msg_buf;
    (void) ucdr_deserialize_double(reader, &topic->x);
    (void) ucdr_deserialize_double(reader, &topic->y);
    (void) ucdr_deserialize_double(reader, &topic->z);

    return !reader->error;
  }

  uint32_t size_of_topic(const Vector3* topic, uint32_t size)
  {
    (void)(topic);

    uint32_t previousSize = size;
    size += (uint32_t)ucdr_alignment(size, 8) + 8;
    size += (uint32_t)ucdr_alignment(size, 8) + 8;
    size += (uint32_t)ucdr_alignment(size, 8) + 8;

    return size - previousSize;
  }
```

julianbermudez, 2019-06-11:
Hi @OpusK,

If I have understood properly, you have trouble because of the `read_callback` at Client side is call infinitely, I'm right? 
In such a case, did you set up the read at Client side taking into account the [delivery_control](https://micro-xrce-dds.readthedocs.io/en/latest/client.html#read-access-profile).
It allows you to control how the Agent delive to the Client the message, in particular you can set the following parameters:
```
typedef struct uxrDeliveryControl
{
    /** The maximum number of topics that the Agent shall send to the Client. */
    uint16_t max_samples;
    /** The maximum amount of time in seconds that shall be spent by the Agent delivering the topic. */
    uint16_t max_elapsed_time;
    /** The maximum transfer rate, in bytes per second, that the Agent shall use. */
    uint16_t max_bytes_per_second;
    /** The minimum elapsed time, in milliseconds, between two topics deliveries. */
    uint16_t min_pace_period;

} uxrDeliveryControl;
```

On the other hand, regarding the IDL generation, which version of the generator are you using?
Which version did you use before this issue?
Which change should you make over the generated code?

OpusK, 2019-06-11:
Hi, @julianbermudez 

I am using the configuration of uxrDeliveryControl like [this](https://github.com/ROBOTIS-GIT/ros2arduino/blob/master/src/ros2/xrcedds/xrcedds.cpp#L328).

> On the other hand, regarding the IDL generation, which version of the generator are you using?
Which version did you use before this issue?

I am using [1.0.1](https://github.com/eProsima/Micro-XRCE-DDS-Gen/releases/tag/v1.0.1) in the past or now.

> Which change should you make over the generated code?

Oh, it's just wrapped in ros2arduino's API.

NOTE: I have tested some of the messages and found that this is happening only when processing 8-byte data like Vector3, Int64.

OpusK, 2019-06-12:
@julianbermudez 

I applied [Client::#94](https://github.com/eProsima/Micro-XRCE-DDS-Client/pull/95#issuecomment-501147888) and confirmed that Int64 works normally.

closed 2019-06-12 by OpusK