eProsima/Micro-XRCE-DDS-Agent #93: TB3-Ros2 no response after Ubuntu 18.04.2 upgrade to latest patch.

rafaelhuang-intel, 2019-06-14:
ISSUE TEMPLATE ver. 0.2.0

Which TurtleBot3 you have?

Waffle & Waffle Pi
Which SBC(Single Board Computer) is working on TurtleBot3?

Intel NUC (I7-6700)
Which OS you installed in SBC?

Ubuntu 18.04 LTS (Bionic Beaver)
Below is execution steps:

install Ubuntu 18.04.2 LTS (Bionic Beaver) (kernal 4.15.x or 4.18.x)
sudo apt update
sudo apt upgrade
after all ubuntu patch updated, install all TB3 relate package, like TB3, Navigation2, Cartographer, etc.
run TB3 and Cartographer SLAM (or navigation)
All data show on terminal:

intel@intel-desktop:~/turtlebot3$ cd ~/turtlebot3 && MicroXRCEAgent serial /dev/ttyACM0
Serial agent initialization... OK
Enter 'q' for exit

cd ~/turtlebot3 && MicroXRCEAgent udp 2018
UDP agent initialization... OK
Enter 'q' for exit
RTPS Participant matched 1.f.59.93.f4.43.0.0.1.0.0.0|0.0.1.c1
RTPS Participant matched 1.f.59.93.fe.43.0.0.1.0.0.0|0.0.1.c1
RTPS Participant matched 1.f.59.93.fd.43.0.0.1.0.0.0|0.0.1.c1
RTPS Subscriber matched 1.f.59.93.fe.43.0.0.1.0.0.0|0.0.22.4
RTPS Participant matched 1.f.59.93.b.44.0.0.1.0.0.0|0.0.1.c1
RTPS Participant matched 1.f.59.93.15.44.0.0.1.0.0.0|0.0.1.c1
RTPS Participant matched 1.f.59.93.14.44.0.0.1.0.0.0|0.0.1.c1
RTPS Participant matched 1.f.59.93.14.44.0.0.2.0.0.0|0.0.1.c1
RTPS Participant matched 1.f.59.93.16.44.0.0.1.0.0.0|0.0.1.c1
RTPS Participant unmatched 1.f.59.93.15.44.0.0.1.0.0.0|0.0.1.c1
RTPS Participant unmatched 1.f.59.93.b.44.0.0.1.0.0.0|0.0.1.c1
RTPS Participant unmatched 1.f.59.93.14.44.0.0.2.0.0.0|0.0.1.c1
RTPS Participant unmatched 1.f.59.93.14.44.0.0.1.0.0.0|0.0.1.c1
RTPS Subscriber unmatched 1.f.59.93.fe.43.0.0.1.0.0.0|0.0.22.4
RTPS Participant unmatched 1.f.59.93.fd.43.0.0.1.0.0.0|0.0.1.c1
RTPS Participant unmatched 1.f.59.93.fe.43.0.0.1.0.0.0|0.0.1.c1
RTPS Participant unmatched 1.f.59.93.f4.43.0.0.1.0.0.0|0.0.1.c1

intel@intel-desktop:~$ ~/turtlebot3/turtlebot3_lidar
turtlebot3 lidar client setup complete!
1 - Sent topic: 2.101000, range_max: 361
2 - Sent topic: 2.127000, range_max: 361
3 - Sent topic: 2.128000, range_max: 361
4 - Sent topic: 2.114000, range_max: 361
5 - Sent topic: 2.128000, range_max: 361
6 - Sent topic: 2.144000, range_max: 361
7 - Sent topic: 2.133000, range_max: 361
8 - Sent topic: 2.121000, range_max: 361
9 - Sent topic: 2.133000, range_max: 361
10 - Sent topic: 2.102000, range_max: 361
11 - Sent topic: 2.129000, range_max: 361
12 - Sent topic: 2.091000, range_max: 361
13 - Sent topic: 2.100000, range_max: 355
14 - Sent topic: 2.140000, range_max: 361
15 - Sent topic: 2.116000, range_max: 361
16 - Sent topic: 2.115000, range_max: 361
17 - Sent topic: 2.128000, range_max: 361
18 - Sent topic: 2.123000, range_max: 361
19 - Sent topic: 2.134000, range_max: 361
20 - Sent topic: 2.125000, range_max: 361
21 - Sent topic: 2.123000, range_max: 361
22 - Sent topic: 2.128000, range_max: 361
23 - Sent topic: 2.108000, range_max: 361
[... 80 lines cut ...]
[cartographer_node-1] I0527 01:26:40.000000 17428 configuration_file_resolver.cc:41] Found '/home/intel/robot_sdk/sdk_ws/device_ws/install/cartographer/share/cartographer/configuration_files/trajectory_builder_3d.lua' for 'trajectory_builder_3d.lua'.
[cartographer_node-1] I0527 01:26:40.000000 17428 configuration_file_resolver.cc:41] Found '/home/intel/robot_sdk/sdk_ws/device_ws/install/cartographer/share/cartographer/configuration_files/trajectory_builder_3d.lua' for 'trajectory_builder_3d.lua'.
[cartographer_node-1] [WARN] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[cartographer_node-1] I0527 01:26:40.000000 17428 submap_2d.cc:187] Added submap 1
[cartographer_node-1] I0527 01:26:40.000000 17428 map_builder_bridge.cc:132] Added trajectory with ID '0'.
[rviz2-3] [INFO] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-3] [INFO] [rviz2]: OpenGl version: 3 (GLSL 1.3)
[rviz2-3] [WARN] [rviz2]: Could not find folder /opt/robot_sdk/core/rviz_rendering_tests/share/rviz_rendering_tests/ogre_media_resources/scripts
[rviz2-3] [WARN] [rviz2]: Could not find folder /opt/robot_sdk/core/rviz_rendering_tests/share/rviz_rendering_tests/ogre_media_resources/meshes
[rviz2-3] [INFO] [rviz2]: Stereo is NOT SUPPORTED
[occupancy_grid_node-2] [WARN] [occupancy_grid_node]: submap_slices and last_frame_id is empty
[occupancy_grid_node-2] [WARN] [occupancy_grid_node]: submap_slices and last_frame_id is empty
[occupancy_grid_node-2] [WARN] [occupancy_grid_node]: submap_slices and last_frame_id is empty
[occupancy_grid_node-2] [WARN] [occupancy_grid_node]: submap_slices and last_frame_id is empty
[occupancy_grid_node-2] [WARN] [occupancy_grid_node]: submap_slices and last_frame_id is empty
[occupancy_grid_node-2] [WARN] [occupancy_grid_node]: submap_slices and last_frame_id is empty
[occupancy_grid_node-2] [WARN] [occupancy_grid_node]: submap_slices and last_frame_id is empty
[occupancy_grid_node-2] [WARN] [occupancy_grid_node]: submap_slices and last_frame_id is empty
[occupancy_grid_node-2] [WARN] [occupancy_grid_node]: submap_slices and last_frame_id is empty
[occupancy_grid_node-2] [WARN] [occupancy_grid_node]: submap_slices and last_frame_id is empty
[occupancy_grid_node-2] [WARN] [occupancy_grid_node]: submap_slices and last_frame_id is empty
[occupancy_grid_node-2] [WARN] [occupancy_grid_node]: submap_slices and last_frame_id is empty
[occupancy_grid_node-2] [WARN] [occupancy_grid_node]: submap_slices and last_frame_id is empty

BTW, if not upgrade all ubuntu patch (no run step3), this issue disappear. Does anyone met the same issue before, would you please share some experience about how to fix or debug?

rafaelhuang-intel, 2019-06-14:
I also report this issue to TB3 (https://github.com/ROBOTIS-GIT/turtlebot3/issues/431), but it's seems this issue may cause by MicroXRCEAgent,. 

I tested on ROS2 Crystal, and double checked on Dashing release, the test result are same.

rafaelhuang-intel, 2019-06-14:
currently I used commit (TB3 version: 74ab7d5,
MicroXRCEAgent: version: a495c65faa964ddc068ac6e1249f17f5c9f92787) in my environment for build and test, if not upgrade all ubuntu patch (no run above step3), this issue disappear, so I think this issue may cause by Ubuntu patch conflict with MicroXRCEAgent.

BorjaOuterelo, 2019-11-04:
Hi, @rafaelhuang-intel is this still an open issue?

rafaelhuang-intel, 2019-11-05:
Hi , @BorjaOuterelo  after tested Dashing release (https://github.com/ROBOTIS-GIT/turtlebot3/issues/431), all workable, so I will close this issue.

closed 2019-11-05 by rafaelhuang-intel