eProsima/Micro-XRCE-DDS-Agent #95: How to uninstall Agent and Client

OpusK, 2019-06-15:
Hi,

I use Agent 1.1.0 very well.
However, during the installation for the previous version of the test, the following link error occurred.
Probably, this is a problem caused by not removing previously installed libraries.

This problem will be solved by deleting old libraries one by one, or by deleting the problematic library and reinstalling it, but it would be nice if there was a way to easily remove the old version.

Is there a good way to solve this?

```
...
[ 94%] Linking CXX shared library libmicroxrcedds_agent.so
/usr/bin/ld: /usr/local/lib/libmicroxrcedds_client.a(session.c.o): relocation R_X86_64_PC32 against symbol `uxr_flash_output_streams' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: final link failed: Bad value
collect2: error: ld returned 1 exit status
CMakeFiles/microxrcedds_agent.dir/build.make:1008: recipe for target 'libmicroxrcedds_agent.so.1.1.0' failed
make[5]: *** [libmicroxrcedds_agent.so.1.1.0] Error 1
CMakeFiles/Makefile2:104: recipe for target 'CMakeFiles/microxrcedds_agent.dir/all' failed
make[4]: *** [CMakeFiles/microxrcedds_agent.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make[3]: *** [all] Error 2
CMakeFiles/uagent.dir/build.make:112: recipe for target 'uagent-prefix/src/uagent-stamp/uagent-build' failed
make[2]: *** [uagent-prefix/src/uagent-stamp/uagent-build] Error 2
CMakeFiles/Makefile2:68: recipe for target 'CMakeFiles/uagent.dir/all' failed
make[1]: *** [CMakeFiles/uagent.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
```

julianbermudez, 2019-06-17:
Hi @OpusK,

Right now, there is not a suitable way to remove the old versions. I will look for a solution but in the meantime, you can try to use prefix (through `-DCMAKE_INSTALL_PREFIX` flag) to isolate versions.

OpusK, 2019-06-17:
Hi, @julianbermudez 

Can you tell me how to use it(-DCMAKE_INSTALL_PREFIX)?
It would be helpful to let me know the actual usage example.

julianbermudez, 2019-06-18:
Hi @OpusK,

The `CMAKE_INSTALL_PREFIX` just set the installation folder. Once installed, you can set the CMake variable `CMAKE_PREFIX_PATH` to `CMAKE_INSTALL_PREFIX` in order to find packages and the Linux `LD_LIBRARY_PATH` to `<CMAKE_INSTALL_PREFIX>/lib` to find the shared libraries.

Is it useful for you enviroment?

OpusK, 2019-06-28:
Hi, @julianbermudez 

It seems to be helpful enough. 
Thank you :)

closed 2019-06-28 by OpusK