eProsima/Micro-XRCE-DDS-Agent #106: Installing Micro-XRCE-DDS-Agent won't make if ros2 Dashing installed on Pi

zmk5, 2019-07-25:
Hey,

I am attempting to install Micro-XRCE-DDS-Agent and Client on a Turtlebot3 Burger running Ubuntu Server IoT 18.04, and I am running into an issue building the program after following this [guide](http://emanual.robotis.com/docs/en/popup/turtlebot3_ros2_sbc_setting/). I attempted the installation on two Turtlebot3's (turtlebot-00 has ros2 dashing installed, trutlebot-01 does not)
If ROS dashing is not installed, it attempts to compile fastRTPS and TinyXML2. The build (`sudo make install`) then succeeds.

```bash
turtlebot-01@tb-01-pi:~/Micro-XRCE-DDS-Agent/build$ cmake -DTHIRDPARTY=ON -DCONFIG_UDP_TRANSPORT_MTU=8192 -DCONFIG_SERIAL_TRANSPORT_MTU=8192 ..
-- Setting build type to 'Release' as none was specified.
-- The C compiler identification is GNU 7.4.0
-- The CXX compiler identification is GNU 7.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- fastcdr thirdparty is being updated...
Submodule 'thirdparty/fastcdr' (https://github.com/eProsima/Fast-CDR.git) registered for path 'thirdparty/fastcdr'
Cloning into '/home/turtlebot-01/Micro-XRCE-DDS-Agent/thirdparty/fastcdr'...
Submodule path 'thirdparty/fastcdr': checked out 'b395f44e896c8621ec754d91bcb9c561de41d9d6'
-- Configuring Fast CDR
-- Version: 1.0.8
-- To change de version modify the file configure.ac
-- Performing Test SUPPORTS_CXX14
-- Performing Test SUPPORTS_CXX14 - Success
-- Check if the system is big endian
-- Searching 16 bit integer
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of unsigned short
-- Check size of unsigned short - done
-- Using unsigned short
-- Check if the system is big endian - little endian
-- fastcdr library found...
-- fastrtps thirdparty is being updated...
Submodule 'thirdparty/fastrtps' (https://github.com/eProsima/Fast-RTPS.git) registered for path 'thirdparty/fastrtps'
Cloning into '/home/turtlebot-01/Micro-XRCE-DDS-Agent/thirdparty/fastrtps'...
Submodule path 'thirdparty/fastrtps': checked out 'b48ce9d2fba6fc94e756da01c58b72f2ad848238'
Submodule 'thirdparty/android-ifaddrs' (https://github.com/michalsrb/android-ifaddrs.git) registered for path 'thirdparty/fastrtps/thirdparty/android-ifaddrs'
Submodule 'thirdparty/asio' (https://github.com/chriskohlhoff/asio.git) registered for path 'thirdparty/fastrtps/thirdparty/asio'
Submodule 'thirdparty/fastcdr' (https://github.com/eProsima/Fast-CDR.git) registered for path 'thirdparty/fastrtps/thirdparty/fastcdr'
Submodule 'thirdparty/idl' (https://github.com/eProsima/IDL-Parser.git) registered for path 'thirdparty/fastrtps/thirdparty/idl'
Submodule 'thirdparty/tinyxml2' (https://github.com/leethomason/tinyxml2.git) registered for path 'thirdparty/fastrtps/thirdparty/tinyxml2'
Cloning into '/home/turtlebot-01/Micro-XRCE-DDS-Agent/thirdparty/fastrtps/thirdparty/android-ifaddrs'...
Cloning into '/home/turtlebot-01/Micro-XRCE-DDS-Agent/thirdparty/fastrtps/thirdparty/asio'...
Cloning into '/home/turtlebot-01/Micro-XRCE-DDS-Agent/thirdparty/fastrtps/thirdparty/fastcdr'...
Cloning into '/home/turtlebot-01/Micro-XRCE-DDS-Agent/thirdparty/fastrtps/thirdparty/idl'...
Cloning into '/home/turtlebot-01/Micro-XRCE-DDS-Agent/thirdparty/fastrtps/thirdparty/tinyxml2'...
Submodule path 'thirdparty/fastrtps/thirdparty/android-ifaddrs': checked out '7b1ce82817226e481d3cda0a5d06b66ebcc211f8'
Submodule path 'thirdparty/fastrtps/thirdparty/asio': checked out 'b3d2ab7255fabe46a49b24a584c9fd797c8248e5'
Submodule path 'thirdparty/fastrtps/thirdparty/fastcdr': checked out 'b395f44e896c8621ec754d91bcb9c561de41d9d6'
Submodule path 'thirdparty/fastrtps/thirdparty/idl': checked out '04930f7926f972142811dd0dbdc0019356385e74'
Submodule path 'thirdparty/fastrtps/thirdparty/tinyxml2': checked out '8c8293ba8969a46947606a93ff0cb5a083aab47a'
-- Configuring Fast RTPS
-- Version: 1.7.0
-- To change de version modify the file configure.ac
-- Found tinyxml2: /home/turtlebot-01/Micro-XRCE-DDS-Agent/thirdparty/fastrtps/thirdparty/tinyxml2  
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
[... 47 lines cut ...]
turtlebot-00@tb-00-pi:~/Micro-XRCE-DDS-Agent/build$ sudo make install
Scanning dependencies of target microxrcedds_agent
[  3%] Building CXX object CMakeFiles/microxrcedds_agent.dir/src/cpp/Root.cpp.o
In file included from /home/turtlebot-00/Micro-XRCE-DDS-Agent/include/uxr/agent/client/ProxyClient.hpp:18:0,
                 from /home/turtlebot-00/Micro-XRCE-DDS-Agent/include/uxr/agent/Root.hpp:18,
                 from /home/turtlebot-00/Micro-XRCE-DDS-Agent/src/cpp/Root.cpp:15:
/home/turtlebot-00/Micro-XRCE-DDS-Agent/include/uxr/agent/participant/Participant.hpp:51:69: error: ‘eprosima::fastrtps::ParticipantDiscoveryInfo’ has not been declared
     void onParticipantDiscovery(fastrtps::Participant* p, fastrtps::ParticipantDiscoveryInfo&& info) override;
                                                                     ^~~~~~~~~~~~~~~~~~~~~~~~
/home/turtlebot-00/Micro-XRCE-DDS-Agent/include/uxr/agent/participant/Participant.hpp:51:10: error: ‘void eprosima::uxr::Participant::onParticipantDiscovery(eprosima::fastrtps::Participant*, int&&)’ marked ‘override’, but does not override
     void onParticipantDiscovery(fastrtps::Participant* p, fastrtps::ParticipantDiscoveryInfo&& info) override;
          ^~~~~~~~~~~~~~~~~~~~~~
/home/turtlebot-00/Micro-XRCE-DDS-Agent/src/cpp/Root.cpp: In constructor ‘eprosima::uxr::Root::Root()’:
/home/turtlebot-00/Micro-XRCE-DDS-Agent/src/cpp/Root.cpp:42:47: error: no match for ‘operator!=’ (operand types are ‘eprosima::fastrtps::xmlparser::XMLP_ret’ and ‘void’)
     if (fastrtps::xmlparser::XMLP_ret::XML_OK != fastrtps::xmlparser::XMLProfileManager::loadDefaultXMLFile())
         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
CMakeFiles/microxrcedds_agent.dir/build.make:62: recipe for target 'CMakeFiles/microxrcedds_agent.dir/src/cpp/Root.cpp.o' failed
make[2]: *** [CMakeFiles/microxrcedds_agent.dir/src/cpp/Root.cpp.o] Error 1
CMakeFiles/Makefile2:104: recipe for target 'CMakeFiles/microxrcedds_agent.dir/all' failed
make[1]: *** [CMakeFiles/microxrcedds_agent.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
```

From the error, it seems that the fact that TinyXML2 library is not found makes the build fail.

zmk5, 2019-07-25:
If you attempt to build with the `-DTHIRDPARTY` option removed. I believe it fails to find the `ASIO_INCLUDE_DIR`:

```bash
turtlebot-00@tb-00-pi:~/Micro-XRCE-DDS-Agent/build$ cmake -DCONFIG_UDP_TRANSPORT_MTU=8192 -DCONFIG_SERIAL_TRANSPORT_MTU=8192 ..
-- Setting build type to 'Release' as none was specified.
-- The C compiler identification is GNU 7.4.0
-- The CXX compiler identification is GNU 7.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- fastcdr library found...
-- Found tinyxml2: /usr/lib/aarch64-linux-gnu/libtinyxml2.so  
-- Found OpenSSL: /usr/lib/aarch64-linux-gnu/libcrypto.so (found version "1.1.1") 
-- fastrtps library found...
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
/home/turtlebot-00/Micro-XRCE-DDS-Agent/ASIO_INCLUDE_DIR
   used as include directory in directory /home/turtlebot-00/Micro-XRCE-DDS-Agent

-- Configuring incomplete, errors occurred!
See also "/home/turtlebot-00/Micro-XRCE-DDS-Agent/build/CMakeFiles/CMakeOutput.log".
```

julianbermudez, 2019-07-29:
Hi @zmk5,

Which version are you using?

zmk5, 2019-07-29:
Hi!

The version installed using ros2 repositories is **1.8.02bionic.20190530.061940** as shown from the following command:

```bash
turtlebot-00@tb-00-pi:~$ apt show ros-dashing-fastrtps
Package: ros-dashing-fastrtps
Version: 1.8.0-2bionic.20190530.061940
Priority: extra
Section: misc
Maintainer: Steven! Ragnarök <stevenragnarok@osrfoundation.org>
Installed-Size: 6,715 kB
Depends: libc6 (>= 2.17), libgcc1 (>= 1:4.2), libssl1.1 (>= 1.1.0), libstdc++6 (>= 7), libtinyxml2-6 (>= 5.0.0), ros-dashing-fastcdr, libssl-dev, libtinyxml2-dev, ros-dashing-ros-workspace
Download-Size: 1,497 kB
APT-Manual-Installed: no
APT-Sources: http://packages.ros.org/ros2/ubuntu bionic/main arm64 Packages
Description: Implementation of RTPS standard.
```

The FastRTPS version that is used to compile the Micro-XRCE-DDS-Agent (located in this [guide](http://emanual.robotis.com/docs/en/popup/turtlebot3_ros2_sbc_setting/)) is from branch `4549ef06040db30532604473f9aac20f9ad1559f`.

OpusK, 2019-07-30:
@zmk5 

I have made an OpenCR firmware update, and have made an e-manual update.
I did not completely check everything I had to check, but I did update after the simple test.
Please refer to the [e-manual](http://emanual.robotis.com/docs/en/platform/turtlebot3/ros2/#ros2).

This is a temporary update, and since most developers are currently on vacation, it is likely that developers will be able to begin detailed checks next week.

zmk5, 2019-07-30:
@OpusK This is great, I really appreciate it!

If you guys are on vacation don't worry about it and please enjoy your vacation! I can definitely wait until you guys are back and situated.

BorjaOuterelo, 2019-09-17:
Hi @zmk5, @OpusK 
Can you update me with the status of this issue? Can we close it?

OpusK, 2019-09-19:
Hi, @BorjaOuterelo 

This issue no longer occurs in the latest TB3.

BorjaOuterelo, 2019-09-19:
@OpusK thanks for the update.

closed 2019-09-19 by BorjaOuterelo

zmk5, 2019-09-20:
Since I used the updated guide provided by @OpusK I haven't run into the issue! Thank you for your help I really appreciate it!