eProsima/Micro-XRCE-DDS-Agent #107: IPv6 support

adamkrawczyk, 2019-08-06:
Hi, 

I'm trying to match this with [husarnet](https://husarnet.com/) ,  (esp32) , but it seems like this version has no ipv6 support. Does it? and how to use that? . I had similar problem with rosserial but I found what needs to be changed to enable that. Thank you in advance :)

versions: 

|ros2arduino|ROS2|Micro-XRCE-DDS Agent|
|:-:|:-:|:-:|
|0.1.2|[Dashing Diademata Patch1](https://github.com/ros2/ros2/releases/tag/release-dashing-20190614)|[1.1.0](https://github.com/eProsima/Micro-XRCE-DDS-Agent/releases/tag/v1.1.0)|

julianbermudez, 2019-08-06:
Hi @adamkrawczyk,

There is a branch under development (in both [Client](https://github.com/eProsima/Micro-XRCE-DDS-Client/tree/feature/ipv6) and [Agent](https://github.com/eProsima/Micro-XRCE-DDS-Agent/tree/feature/ipv6)) which implements IPv6. You can try with it a give us feedback. Thanks!

adamkrawczyk, 2019-08-06:
There is problem with make - after cloning this branch 

`git clone --single-branch --branch feature/ipv6 https://github.com/eProsima/Micro-XRCE-DDS-Agent.git` 

when i try to `make` i have following error:

[image]

julianbermudez, 2019-08-06:
Do you have Micro XRCE-DDS Client installed on your computer?
It seems to be a compatibility version error. Try to uninstall it and install the IPv6 branch version.

adamkrawczyk, 2019-08-06:
No I don't have. I even installed Client and everything compiled correctly but agent is still not working

adamkrawczyk, 2019-08-06:
I've tested that on another computer where any version was never downloaded and I have following error:
```
[ 92%] Building CXX object CMakeFiles/microxrcedds_agent.dir/src/cpp/p2p/InternalClient.cpp.o
/home/byq77/workspace/tmp/Micro-XRCE-DDS-Agent/src/cpp/p2p/InternalClient.cpp: In member function ‘bool eprosima::uxr::InternalClient::run()’:
/home/byq77/workspace/tmp/Micro-XRCE-DDS-Agent/src/cpp/p2p/InternalClient.cpp:107:83: error: invalid conversion from ‘uint16_t {aka short unsigned int}’ to ‘const char*’ [-fpermissive]
         if (uxr_init_udp_transport(&transport_, &platform_, UXR_IPv4, ip.c_str(), port_))
                                                                                   ^~~~~
In file included from /home/byq77/workspace/tmp/Micro-XRCE-DDS-Agent/build/temp_install/include/uxr/client/transport.h:21:0,
                 from /home/byq77/workspace/tmp/Micro-XRCE-DDS-Agent/build/temp_install/include/uxr/client/client.h:30,
                 from /home/byq77/workspace/tmp/Micro-XRCE-DDS-Agent/include/uxr/agent/p2p/InternalClient.hpp:18,
                 from /home/byq77/workspace/tmp/Micro-XRCE-DDS-Agent/src/cpp/p2p/InternalClient.cpp:15:
/home/byq77/workspace/tmp/Micro-XRCE-DDS-Agent/build/temp_install/include/uxr/client/profile/transport/ip/udp/udp_transport.h:48:16: note:   initializing argument 5 of ‘bool uxr_init_udp_transport(uxrUDPTransport*, uxrUDPPlatform*, uxrIpProtocol, const char*, const char*)’
 UXRDLLAPI bool uxr_init_udp_transport(
                ^~~~~~~~~~~~~~~~~~~~~~
CMakeFiles/microxrcedds_agent.dir/build.make:878: recipe for target 'CMakeFiles/microxrcedds_agent.dir/src/cpp/p2p/InternalClient.cpp.o' failed
make[5]: *** [CMakeFiles/microxrcedds_agent.dir/src/cpp/p2p/InternalClient.cpp.o] Error 1
CMakeFiles/Makefile2:104: recipe for target 'CMakeFiles/microxrcedds_agent.dir/all' failed
make[4]: *** [CMakeFiles/microxrcedds_agent.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make[3]: *** [all] Error 2
CMakeFiles/uagent.dir/build.make:115: recipe for target 'uagent-prefix/src/uagent-stamp/uagent-build' failed
make[2]: *** [uagent-prefix/src/uagent-stamp/uagent-build] Error 2
CMakeFiles/Makefile2:256: recipe for target 'CMakeFiles/uagent.dir/all' failed
make[1]: *** [CMakeFiles/uagent.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
```

the same file but this time the problem is with port_

julianbermudez, 2019-08-07:
Hi @adamkrawczyk,

That compilation error should be fixed now.

adamkrawczyk, 2019-08-07:
It's working now :) I'm waiting for tcp6 for ESP32

adamkrawczyk, 2019-08-08:
If only this will be useful I have following troubles:
[image]

julianbermudez, 2019-08-27:
Hi @adamkrawczyk, 

Sorry for the delay but we were on vacations. Are you able to reproduce the same error using IPv4?

adamkrawczyk, 2019-08-28:
I've tried and I have the same error when using tcp4

adamkrawczyk, 2019-08-28:
I also have this error when using serial connection, but I'm able to list topic and see this topic 

```
adam@adamlaptop:~$ ros2 node list
/ros2arduino_pub_node
adam@adamlaptop:~$ ros2 node info /ros2arduino_pub_node 
/ros2arduino_pub_node
[ERROR] [rmw_fastrtps_shared_cpp]: Unable to find GUID for node: ros2arduino_pub_node
Failed to get_subscriber_names_and_types: Unable to find GUID for node , at /tmp/binarydeb/ros-dashing-rmw-fastrtps-shared-cpp-0.7.5/src/rmw_node_info_and_types.cpp:87
adam@adamlaptop:~$ ros2 topic list
/arduino_chatter
/parameter_events
/rosout
adam@adamlaptop:~$ ros2 topic echo /arduino_chatter 
data: Hello ros2arduino 69
---
data: Hello ros2arduino 70
---

BorjaOuterelo, 2019-11-18:
Hi @adamkrawczyk are you still having this issue?

julibert, 2020-03-06:
Hi @adamkrawczyk, I close this issue. In case it persists, please reopen it.

closed 2020-03-06 by julibert