eProsima/Micro-XRCE-DDS-Agent #109: Is there a way to support DDS-Security?

OpusK, 2019-08-27:
Hi,

For FastRTPS, I can use DDS-Security with ROS2.
Is there a way for Agent to use this function?

julianbermudez, 2019-08-27:
Hi @OpusK,

I'm going to create a branch (feature/security) which allows compiling the Agent with the Fast RTPS' security plugging. In principle, once the Agent is recompiled with `UAGENT_SECURITY_PROFILE` enable, you just need to modify the references XML according to the Fast RTPS [security documentation](https://fast-rtps.docs.eprosima.com/en/latest/security.html#example-configuring-the-participant).

OpusK, 2019-08-29:
@julianbermudez,

Thank you!
I will inform you of the result after the test.

OpusK, 2019-09-17:
Hi, @julianbermudez 

I've now tested the method you gave me.
However, I get the following error.

[image]

Maybe I don't understand how to use it so I think this happens.
I need your teaching.

The xml I used is below.
```
<profiles>
    <participant profile_name="xelnetwork">
        <rtps>
            <name>xelnetwork</name>
            <builtin>
                <domainId>0</domainId>
            </builtin>
            <propertiesPolicy>
                <properties>
                    <!-- Activate Auth:PKI-DH plugin -->
                    <property>
                        <name>dds.sec.auth.plugin</name>
                        <value>builtin.PKI-DH</value>
                    </property>

                    <!-- Configure Auth:PKI-DH plugin -->
                    <property>
                        <name>dds.sec.auth.builtin.PKI-DH.identity_ca</name>
                        <value>~/workspace/ros2/demo_examples/demo_keys/xelnetwork/identity_ca.cert.pem</value>
                    </property>
                    <property>
                        <name>dds.sec.auth.builtin.PKI-DH.identity_certificate</name>
                        <value>~/workspace/ros2/demo_examples/demo_keys/xelnetwork/cert.pem</value>
                    </property>
                    <property>
                        <name>dds.sec.auth.builtin.PKI-DH.private_key</name>
                        <value>~/workspace/ros2/demo_examples/demo_keys/xelnetwork/key.pem</value>
                    </property>
                </properties>
            </propertiesPolicy>
        </rtps>
    </participant>

    <data_writer profile_name="XEL_ANALOG">
        <topic>
            <kind>NO_KEY</kind>
            <name>rt/XEL_ANALOG</name>
            <dataType>std_msgs::msg::dds_::Int16_</dataType>
            <historyQos>
                <kind>KEEP_LAST</kind>
                <depth>10</depth>
            </historyQos>
        </topic>
    </data_writer>

    <data_reader profile_name="XEL_LED">
        <topic>
            <name>rt/XEL_LED</name>
            <dataType>std_msgs::msg::dds_::UInt8_</dataType>
        </topic>
    </data_reader>

    <topic profile_name="UInt8">
        <kind>NO_KEY</kind>
        <name>UInt8</name>
        <dataType>std_msgs::msg::dds_::UInt8_</dataType>
    </topic>

    <topic profile_name="Int16">
        <kind>NO_KEY</kind>
        <name>Int16</name>
        <dataType>std_msgs::msg::dds_::Int16_</dataType>
    </topic>
</profiles>
```

And below are the authentication files created using [sros2 function](https://github.com/ros2/sros2/blob/master/SROS2_Linux.md#generating-a-keystore-keys-and-certificates).

[image]

BorjaOuterelo, 2019-09-17:
Hi @OpusK 

It looks like the XML you are using is not right, plese refer to https://fast-rtps.docs.eprosima.com/en/v1.8.1/security.html

As you can see in the documentation, you are missing `file://` in the propertiesPolicy.
Apart from that, `~` is replaced by your shell, so I think it is not going to work in a configuration file.

Can you fix the XML and try again?

OpusK, 2019-09-17:
Hi, @BorjaOuterelo 

Your point was helpful!!
This problem is solved.
thank you :)

closed 2019-09-17 by OpusK