eProsima/Micro-XRCE-DDS-Agent #120: Segfault on quitting?

rnvandemark, 2019-11-12:
**Environment**
Ubuntu 18.04.3 LTS 64-bit
Micro-XRCE-DDS-Agent v1.1.0

**Problem**
When starting and then quitting the `MicroXRCEAgent` command, I get a segfault every time. I attempted a fresh install too and get the same result.

```
rnvandemark@ideapad:~/$ MicroXRCEAgent
A subcommand is required
Run with --help for more information.

Enter command: serial --dev /dev/ttyACM0 -b 115200
Enter 'q' for exit
[1573568617.797831] info     | SerialServerLinux.cpp | init                     | running...             | fd: 3
q
[1573568619.856496] info     | SerialServerLinux.cpp | close                    | server stopped         | fd: 3
Segmentation fault (core dumped)
```

Even if I do nothing, I get a segfault:

```
rnvandemark@ideapad:~/$ MicroXRCEAgent
A subcommand is required
Run with --help for more information.

Enter command: q
Segmentation fault (core dumped)
```

This comes from another issue I started: ROBOTIS-GIT/ros2arduino#34

BorjaOuterelo, 2019-11-26:
Hi @rnvandemark 
I am not able to reproduce this issue.
I tried it in a clean Ubuntu container.
```
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=18.04
DISTRIB_CODENAME=bionic
DISTRIB_DESCRIPTION="Ubuntu 18.04.3 LTS"
```
Can you provide more info?
It is a clean machine?
Can you test it in a Docker for example?
`ldd` output?

julibert, 2020-03-06:
Hi @rnvandemark, I close this issue. In case it persists, please reopen it.

closed 2020-03-06 by julibert