eProsima/Micro-XRCE-DDS-Agent #150: [Question] Specify bind interfaces

driftregion, 2020-04-29:
I and a couple other colleagues have had issues with crosstalk.  Our machines have multiple network adapters and the Agent binds to all of them by default.  This leads to scenarios where clients associate with incorrect agents.

I realize this could be classified as any of:
1. DDS object model issue (solution: specify different DOMAIN_ID, CLIENT_ID, ...)
1. Host networking configuration issue (solution: don't allow multiple Agents on the same subnet)
1. issue with Micro-XRCE-DDS-Agent itself (solution: specify bind interfaces explicitly)

We're working around this by running the Agent inside a Docker container with specific networking configuration (perhaps this is how others are doing it?)

I'm opening this issue to see if it resonates with others, and to see if anyone has found a better way to bypass or solve it.

Thanks!

julibert, 2020-04-30:
Hi @driftregion,

It is at least a lack of feature in Micro XRCE-DDS Agent.
I create a new `feature/interface` branch which allows you to select an interface for `udp4` subcommand. You can see the available interfaces with `udp4 --help`, and then select one of this with `udp4 -i <interface-name>`.
Please, take into account that it is under developing but I think it could help you.

FranFin, 2020-07-20:
@driftregion Hi! Thank you for your interest in Micro XRCE-DDS. Notice that we’ll discuss the recent contributions and developments in the next [Embedded WG meeting](https://discourse.ros.org/t/ros-2-embedded-wg-meetings/15460). You’re invited to bring your Micro XRCE-DDS related issue to this platform. We’ll be very happy to have you join us and discuss your questions with the community.

jamoralp, 2020-11-19:
Closing since long inactivity time has been elapsed.

closed 2020-11-19 by jamoralp