eProsima/Micro-XRCE-DDS-Agent #166: Binaries not found due to UAGENT_ISOLATED_INSTALL

karl-schulz, 2020-06-01:
Hi!
After building and installing the MicroXRCEAgent using the official [tutorial](https://micro-xrce-dds.docs.eprosima.com/en/latest/installation.html#installing-the-agent-stand-alone), the binaries are not found when executing `MicroXRCEAgent`.

[This Guide](https://micro-xrce-dds.docs.eprosima.com/en/latest/quickstart.html) also assumes the binaries are located in `/usr/local/bin`. They are actually in `/usr/local/microxrcedds_agent-1.3.0/bin`.

Correct me if I'm wrong, but defaulting `UAGENT_ISOLATED_INSTALL` to `ON` breaks the tutorial in this form. Was it added for development purposes and left on ON by accident? Or is this on purpose?

System: NVIDIA Jetson TX2 (Ubuntu-like L4T)
Reproduction: Following the tutorial.

Thanks!

pablogs9, 2020-06-02:
Hello @karl-schulz , thanks for the issue.

It has been addressed here https://github.com/eProsima/Micro-XRCE-DDS-Agent/pull/167 and here https://github.com/eProsima/Micro-XRCE-DDS-Client/pull/157

I'll close this issue when merged.

karl-schulz, 2020-06-02:
Cool, thanks for the fast reply!

pablogs9, 2020-07-14:
Fixes merged, closing

closed 2020-07-14 by pablogs9