eProsima/Micro-XRCE-DDS-Agent #168: need a way to disable asio, tinyxml2, fastcdr and fastrts and take them from the colcon build instead

razr, 2020-06-01:
Hello @julibert,
I did this, but it does not help.
```
diff --git a/cmake/SuperBuild.cmake b/cmake/SuperBuild.cmake
index 3095517..a47b61c 100644
--- a/cmake/SuperBuild.cmake
+++ b/cmake/SuperBuild.cmake
@@ -49,7 +49,7 @@ endif()

 # Fast CDR.
 unset(fastcdr_DIR CACHE)
-find_package(fastcdr ${_fastcdr_version} EXACT QUIET)
+find_package(fastcdr ${_fastcdr_version} QUIET)
 if(NOT fastcdr_FOUND)
     ExternalProject_Add(fastcdr
         GIT_REPOSITORY
@@ -80,7 +80,7 @@ endif()
 if(UAGENT_FAST_PROFILE)
     # Fast RTPS.
     unset(fastrtps_DIR CACHE)
-    find_package(fastrtps ${_fastrtps_version} EXACT QUIET)
+    find_package(fastrtps ${_fastrtps_version} QUIET)
     if(NOT fastrtps_FOUND)
         ExternalProject_Add(fastrtps
             GIT_REPOSITORY
@@ -242,4 +242,4 @@ ExternalProject_Add(uagent
```
It still tries to build fastrtps.
I need a way to disable asio, tinyxml2, fastcdr and fastrts

```
241.664s] CMake Error at /usr/share/cmake-3.16/Modules/FindPackageHandleStandardArgs.cmake:146 (message):
[241.664s]   Could NOT find Threads (missing: Threads_FOUND)
[241.664s] Call Stack (most recent call first):
[241.664s]   /usr/share/cmake-3.16/Modules/FindPackageHandleStandardArgs.cmake:393 (_FPHSA_FAILURE_MESSAGE)
[241.664s]   cmake/modules/FindThreads.cmake:205 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
[241.664s]   src/cpp/CMakeLists.txt:245 (find_package)
[241.665s]
[241.665s]
[241.665s] -- Configuring incomplete, errors occurred!
[241.665s] See also "/home/akholodn/github/Wind-River/vxworks7-ros2-build/build/ros2/ros2_ws/build/microxrcedds_agent/fastrtps/src/fastrtps-build/CMakeFiles/CMakeOutput.log".
[241.665s] See also "/home/akholodn/github/Wind-River/vxworks7-ros2-build/build/ros2/ros2_ws/build/microxrcedds_agent/fastrtps/src/fastrtps-build/CMakeFiles/CMakeError.log".
```

julibert, 2020-06-02:
Hi @razr,

I have some questions about your environment:

- Does your system support threads?
- Do you have a ROS 2 installation?
- Did you make a `source <ros2-path>/setup.bash` before trying to install the Agent?

It seems that CMake is not able to find `FastRTPS` neither `FastCDR` and that is because they are not in any path that CMake is searching for. Therefore I would try to `source` the ROS 2 installation or add `-DCMAKE_PREFIX_PATH=<fast-path>` flag to Agent build.

razr, 2020-06-02:
@julibert, I have successfully built FastRTPS with my patches applied as a part of the ROS2 source build. So, it finds properly FastRTPS libs e.g. while building rmw etc. Why it should be different here. But let me check.

razr, 2020-06-11:
@pablogs9, even I'm on your dashing release for Micro-XRCE-DDS-Agent
```console
akholodn@windix:~/github/Wind-River/vxworks7-ros2-build/build/ros2/ros2_ws/src/eProsima/Micro-XRCE-DDS-Agent$ git branch
* dashing
```
I see that you are building your own Fast-RTPS version 1.8.3 instead of an official one from the ROS2 dashing repo which is 1.8.2

```console
  eProsima/Fast-RTPS:
    type: git
    url: https://github.com/eProsima/Fast-RTPS.git
    version: v1.8.2
```
I have patched CMakeFiles.txt and it takes 1.8.2 now from my build but fails here:
```c
/home/akholodn/github/Wind-River/vxworks7-ros2-build/build/ros2/ros2_ws/src/eProsima/Micro-XRCE-DDS-Agent/src/cpp/xmlobjects/xmlobjects.cpp:18:10: fatal error: 'fastrtps/attributes/ReplierAttributes.hpp' file not found
#include <fastrtps/attributes/ReplierAttributes.hpp>
```

julibert, 2020-06-12:
Hi @razr, 

We need that Fast RTPS version because it has support for `Requester` and `Replier` entities, which are the entities Micro XRCE-DDS uses to handle services in micro-ROS. Is there any possibility to compile this particular version for the Agent?