eProsima/Micro-XRCE-DDS-Agent #171: About client and agent compatibility information.

OpusK, 2020-06-19:
Hi,

The release notes below currently indicate the most suitable version. (From Agent 1.4.0, it was better to indicate which version of ROS2 is supported!)

- [Release Note](https://micro-xrce-dds.docs.eprosima.com/en/latest/notes.html)

However, it would be better if there is information on which version of `Agent` is compatible with which version of `Client`.  (Backward compatibility)
Likewise, it would be better if compatibility information for ROS2 was provided. In patch units.

FranFin, 2020-07-20:
@OpusK Hi! Thank you for your interest in Micro XRCE-DDS. Notice that we’ll discuss the recent contributions and developments in the next [Embedded WG meeting](https://discourse.ros.org/t/ros-2-embedded-wg-meetings/15460). You’re invited to bring your Micro XRCE-DDS related issue to this platform. We’ll be very happy to have you join us and discuss your questions with the community.

OpusK, 2020-07-27:
Hi, @FranFin 

Thank you for inviting me but I am late to read.
I will close this issue.

closed 2020-07-27 by OpusK