eProsima/Micro-XRCE-DDS-Agent #178: Inquiry about backward compatibility between XRCE Agent and XRCE Client

glchua2, 2020-08-04:
Hi,

Are there any known compatibility issue between Agent 1.4.2 and Client 1.1.1 ?

I notice that when shifting from Agent 1.1.0 to Agent 1.4.2, one of the topic which my client is subscribing to encountered an error 0x80 (STATUS_ERR_DDS_ERROR). This occurred when trying to create entities using uxr_run_session_until_all_status().

pablogs9, 2020-08-04:
There is no known compatibility issue but this is a new Agent version with and quite an old Client version. You can find the compatible version in the releases of the XRCE-DDS repository: https://github.com/eProsima/Micro-XRCE-DDS/releases.

Is it possible for you to upgrade the Client version to v.1.2.5?

glchua2, 2020-08-05:
Looking at https://github.com/eProsima/Micro-XRCE-DDS/releases does it mean that Agent 1.4.0 is only guaranteed to work well with Client 1.2.3? Not even Client 1.2.1?

All the topic that my device is subscribing to have similar datareader_xml, with the exception of topic name and topic dataType. Hence, I am not sure why this is happening just for this particular topic.

It is possible to upgrade the Client version to v.1.2.5, but will upgrading definitely solve this issue?

pablogs9, 2020-08-05:
Could you share the code?

closed 2020-11-16 by pablogs9