eProsima/Micro-XRCE-DDS-Agent #192: Compiling DDS-Agent on Raspberry Pi4

vibnwis, 2020-10-17:
Hi,
I have compiled twice and experienced the similar issue.See below. I used the following steps. except the first line where I downloaded the latest ZIP instead.

# git clone -b v1.4.2 https://github.com/eProsima/Micro-XRCE-DDS-Agent
cd Micro-XRCE-DDS-Agent 
mkdir build && cd build
cmake ..
make

/usr/bin/ld: SharedMemTransport.cpp:(.text._ZN8eprosima7fastdds4rtps16SharedMemManager4Port8try_pushERKSt10shared_ptrINS2_6BufferEE[_ZN8eprosima7fastdds4rtps16SharedMemManager4Port8try_pushERKSt10shared_ptrINS2_6BufferEE]+0x2c8): undefined reference to `__atomic_compare_exchange_8'
collect2: error: ld returned 1 exit status
make[5]: *** [src/cpp/CMakeFiles/fastrtps.dir/build.make:2621: src/cpp/libfastrtps.so.2.0.1] Error 1
make[4]: *** [CMakeFiles/Makefile2:987: src/cpp/CMakeFiles/fastrtps.dir/all] Error 2
make[3]: *** [Makefile:163: all] Error 2
make[2]: *** [CMakeFiles/fastdds.dir/build.make:115: fastdds/src/fastdds-stamp/fastdds-build] Error 2
make[1]: *** [CMakeFiles/Makefile2:227: CMakeFiles/fastdds.dir/all] Error 2
make: *** [Makefile:84: all] Error 2

vibnwis, 2020-10-17:
The same errors apply to v.1.4.2
/usr/bin/ld: SharedMemTransport.cpp:(.text._ZN8eprosima7fastdds4rtps16SharedMemManager4Port8try_pushERKSt10shared_ptrINS2_6BufferEE[_ZN8eprosima7fastdds4rtps16SharedMemManager4Port8try_pushERKSt10shared_ptrINS2_6BufferEE]+0x2c8): undefined reference to `__atomic_compare_exchange_8'
collect2: error: ld returned 1 exit status
make[5]: *** [src/cpp/CMakeFiles/fastrtps.dir/build.make:2621: src/cpp/libfastrtps.so.2.0.1] Error 1
make[4]: *** [CMakeFiles/Makefile2:987: src/cpp/CMakeFiles/fastrtps.dir/all] Error 2
make[3]: *** [Makefile:163: all] Error 2
make[2]: *** [CMakeFiles/fastdds.dir/build.make:115: fastdds/src/fastdds-stamp/fastdds-build] Error 2
make[1]: *** [CMakeFiles/Makefile2:227:

pablogs9, 2020-10-19:
Could you detail where does this issue happen? Where are you compiling? Which OS do you have?

vibnwis, 2020-10-19:
Could you detail where does this issue happen? Where are you compiling?
in Raspberry Pi 4,
Which OS do you have?
The OS is Raspberry OS.

pablogs9, 2020-10-19:
It seems you are not linking against atomic library. Try:

```bash
cmake ..  -DCMAKE_CXX_FLAGS=-latomic
```

vibnwis, 2020-10-19:
Have tried that command, but no avails. Please advise.

pi@raspberrypi:~/Micro-XRCE-DDS-Agent $ rm -rf build/
pi@raspberrypi:~/Micro-XRCE-DDS-Agent $ mkdir build; cd build
pi@raspberrypi:~/Micro-XRCE-DDS-Agent/build $ cmake ..  -DCMAKE_CXX_FLAGS=-latomic
-- The C compiler identification is GNU 8.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- The CXX compiler identification is GNU 8.3.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/Micro-XRCE-DDS-Agent/build

pi@raspberrypi:~/Micro-XRCE-DDS-Agent/build $ make 

/usr/bin/ld: SharedMemTransport.cpp:(.text._ZN8eprosima7fastdds4rtps16SharedMemManager4Port8try_pushERKSt10shared_ptrINS2_6BufferEE[_ZN8eprosima7fastdds4rtps16SharedMemManager4Port8try_pushERKSt10shared_ptrINS2_6BufferEE]+0x2c8): undefined reference to `__atomic_compare_exchange_8'
collect2: error: ld returned 1 exit status
make[5]: *** [src/cpp/CMakeFiles/fastrtps.dir/build.make:2621: src/cpp/libfastrtps.so.2.0.1] Error 1
make[4]: *** [CMakeFiles/Makefile2:987: src/cpp/CMakeFiles/fastrtps.dir/all] Error 2
make[3]: *** [Makefile:163: all] Error 2
make[2]: *** [CMakeFiles/fastdds.dir/build.make:115: fastdds/src/fastdds-stamp/fastdds-build] Error 2
make[1]: *** [CMakeFiles/Makefile2:22

vibnwis, 2020-10-21:
Hi Pablo,
1. Is there a way to test the DDS-Agent such as loop-back ? I wanted to change the Transport, but there is a way to test that would be very useful.
2. Is there anything I help to debug the compiling errors, changing some of the settings. I am happy to help.

Thank you

pablogs9, 2020-10-21:
Hello @vibnwis,

1. What do you mean with loop-back?
2. Sorry about the delay with this, I'm looking for a moment to try it in a Raspberry here. I will reach you with my results.

vibnwis, 2020-10-21:
1. Like sending out a message which will get back a while later? like "ping" command.
2. No worries, I wish I could help, as helping you actually is helping myself to be honest.

pablogs9, 2020-10-21:
1. I'm not sure about what you really want. In order to test a Micro XRCE-DDS Agent the best way is to connect a Micro XRCE-DDS Client and set the flag `-v6` in the Agent.

vibnwis, 2020-10-21:
1. As both parts, the clients and the agent are modified,  I don't know what works and what does not. Hence, having an ability to test individually will help to isolate issues. However, if there is no way  can perform the so-called self-test, it will have to wait when both parts are ready. Many thanks for your assistance, really.

vibnwis, 2020-11-03:
I have decided to switch to Linux PC. Hence, this issue is not valid for time being.

closed 2020-11-03 by vibnwis

ramboerik, 2021-03-01:
Hello,

I ran into a similar error with libatomic  being missed when compiling release v1.3.0 for rpi 4.  The problem seems to be related to GCC version 8.3.0 that's shipped with latest rpi os as it doesn't implicity link with libatomic.

The error:
```
[100%] Linking CXX executable MicroXRCEAgent
/usr/bin/ld: libmicroxrcedds_agent.so.1.3.0: undefined reference to `__atomic_store_8'
/usr/bin/ld: libmicroxrcedds_agent.so.1.3.0: undefined reference to `__atomic_load_8'
collect2: error: ld returned 1 exit status
```

The procedure to get it to work:
```
cd Micro-XRCE-DDS-Agent 
mkdir build && cd build
cmake ..
make
```
Wait for build to fail then do the following modification:

 **build/CMakeCache.txt**
```diff
//Flags used by the linker during the creation of shared libraries
// during all build types.
---CMAKE_SHARED_LINKER_FLAGS:STRING=
+++CMAKE_SHARED_LINKER_FLAGS:STRING=-latomic
```
run **make** again and it should compile.

vibnwis, 2021-03-02:
changed as shown, but still experienced issues. See below. Any advices?
er4Port8try_pushERKSt10shared_ptrINS2_6BufferEE]+0x74): undefined reference
to `__atomic_load_8'
/usr/bin/ld:
SharedMemTransport.cpp:(.text._ZN8eprosima7fastdds4rtps16SharedMemManager4Port8try_pushERKSt10shared_ptrINS2_6BufferEE[_ZN8eprosima7fastdds4rtps16SharedMemManager4Port8try_pushERKSt10shared_ptrINS2_6BufferEE]+0xe8):
undefined reference to `__atomic_compare_exchange_8'
/usr/bin/ld:
SharedMemTransport.cpp:(.text._ZN8eprosima7fastdds4rtps16SharedMemManager4Port8try_pushERKSt10shared_ptrINS2_6BufferEE[_ZN8eprosima7fastdds4rtps16SharedMemManager4Port8try_pushERKSt10shared_ptrINS2_6BufferEE]+0x178):
undefined reference to `__atomic_load_8'
/usr/bin/ld:
SharedMemTransport.cpp:(.text._ZN8eprosima7fastdds4rtps16SharedMemManager4Port8try_pushERKSt10shared_ptrINS2_6BufferEE[_ZN8eprosima7fastdds4rtps16SharedMemManager4Port8try_pushERKSt10shared_ptrINS2_6BufferEE]+0x248):
undefined reference to `__atomic_load_8'
/usr/bin/ld:
SharedMemTransport.cpp:(.text._ZN8eprosima7fastdds4rtps16SharedMemManager4Port8try_pushERKSt10shared_ptrINS2_6BufferEE[_ZN8eprosima7fastdds4rtps16SharedMemManager4Port8try_pushERKSt10shared_ptrINS2_6BufferEE]+0x2c8):
undefined reference to `__atomic_compare_exchange_8'
collect2: error: ld returned 1 exit status

ramboerik, 2021-03-02:
I tried to build v.1.4.2 but got the same linker error in another file so I did the same change as above but in `build/fastdds/src/fastdds-build/CMakeCache.txt` and that compiled.

@vibnwis try change file:  
**build/fastdds/src/fastdds-build/CMakeCache.txt**
with:
```diff
//Flags used by the linker during the creation of shared libraries
// during all build types.
---CMAKE_SHARED_LINKER_FLAGS:STRING=
+++CMAKE_SHARED_LINKER_FLAGS:STRING=-latomic
```

vibnwis, 2021-03-02:
It works finally.  I am wondering why the previous solution worked at your end? did you not use version 1.4.2?

ramboerik, 2021-03-02:
Nope, I'm using v1.3.0 😃 I tried newer  but got some version mismatch with the arduino library I'm using

vibnwis, 2021-03-02:
all good. many thanks for sharing the resolution.