eProsima/Micro-XRCE-DDS-Agent #195: Adding CAN-Bus transport to Micro-XRCE-DDS-Agent

vibnwis, 2020-11-02:
Hi Pablo,
I decided to changed from Raspberry to a Linux PC for Micro-XRCE-DDS-Agent. Having compiled the Micro-XRCE-DDS-Agent successfully, I would like to proceed to implementing CAN-bus transport. As CANSocket is long established in Linux kernel which mimic the Network-Socket, I am thinking, I might able to replicate UDPv4AgentLinux.cpp to adopt CAN-Bus. This is simply a simplistic conclusion out of my research. Would able to share some advice, please? especially the implication of the potential file changes or other issues. Of course, it would be delightful if you have got a template for implementing a transport like in the micro-agent.
I look forward to hear from you.
Many thanks

vibnwis, 2020-11-03:
Hi Pablo,
Would you mind share a bit of the insights of CAN-Bus implementation approach, please? I truly appreciate your advice, It will definitely help,

Thank you.

pablogs9, 2020-11-03:
Hello @vibnwis, we are not as used to create transport for the agent as we are for the client.

I'm adding my colleague @jamoralp to the loop because I think that he can help much more than me with this topic.

vibnwis, 2020-11-03:
Very appreciate that. Mean while I am reading the code and see if there is any pitfalls I need to be aware. Hopefully, Jamoralp have something positive to share. Many thanks.

jamoralp, 2020-11-03:
Good morning @vibnwis,
Regarding the implementation of a new transport for the agent, there are several considerations which need to be taken into account:
* You must implement the `Server` interface. Source code for this interface can be found [here](https://github.com/eProsima/Micro-XRCE-DDS-Agent/blob/master/include/uxr/agent/transport/Server.hpp). Methods to be overriden correspond to the `init` and `fini` functions, for the initialization/destruction phase; as well as `recv_message` and `send_message`, which must contain  the logics used for receiving and sending a message using your protocol.
* You must provide with an implementation of the EndPoint interface, which allows to set the source/destination information of the Input/OutputPacket structures which are handled by the agent to send/receive information.

This said, I would create a `can` folder under `include/uxr/agent/transport` and `src/cpp/transport` to place header/source code files. Final steps would involve adding this to the CLI tools; currently, we support two CLI modes, the "old" one involving the use of an external library we eventually want to get rid of; and another one, which is our built-in argument parser and can be found [here](https://github.com/eProsima/Micro-XRCE-DDS-Agent/blob/master/include/uxr/agent/utils/ArgumentParser.hpp). I suggest integrating the CAN transport just on the second one, so we can move forward deleting the former one.

Hope this information is of any help. Thanks for your interest!

vibnwis, 2020-11-03:
Sounds good. I am reading that part of the code at the moment.  As for the
CLI tools, can you or do you have which I can read to improve myself,
please?

Thank you

jamoralp, 2020-11-03:
> Sounds good. I am reading that part of the code at the moment. As for the CLI tools, can you or do you have which I can read to improve myself, please? Thank you
> […](#)

I developed that CLI part so, thinking about it, it might be easier if I take care of integrating the CAN option within the parser myself. I will do that when the transport is in good shape. Please do not hesitate to contact regarding any doubts and keep us updated. Thanks!

vibnwis, 2020-11-03:
Sounds good. Thanks

vibnwis, 2020-11-04:
Hi jamoralp and team-mates
CAN Bus transport design and Implementation. See the bold texts.

Here are the traces from top ( Applications layers ) to lower transport layer. The main() of microxrce_agent.cpp would remain unchanged. The same for AgenInstance, it will still use the launch_server() of UDPv4Subcommand object. In other words, users would see it as if it is dealing with UDP protocol. However, in actual fact it is dealing with the CAN-protocol.

This is a quick way of proving the concepts of CAN transport implementation. It will be more user as well as system friendly once the CAN transport is functioning.

Very appreciate your comments.

main() [microxrce_agent.cpp]
|---->AgentInstance(...) 	

Class AgentInstance [ AgentInstance.hpp]
	|---->UDPv4Subcommand 

Class UDPv4Subcommand extends  ServerSubcommand [CLI.hpp]
	[---->launch_server()
		|--->UDPv4Agent()
		**|---->CANAgent << New added object for CAN-bus Transport** 

Class UDPv4Agent extends Server<IPv4EndPoint> UDPC4AgentLinux.hpp
	|---->init()
	|---->finit()
	|---->recv_message()
	|---->send_message()
	|---->handl_error()
	|.... etc

**Class CANAgent extends extends Server<IPv4EndPoint> CANLinux.hpp
        |---->init()
	|---->finit()
	|---->recv_message()
	|---->send_message()
	|---->handl_error()
	|.... etc****

vibnwis, 2020-11-04:
Oh yes, I did not mention anything about the server. As from the traces, I did not see the server involves directly and I could be missed out or wrong as well. Appreciate your advice, please.

jamoralp, 2020-11-05:
Good morning @vibnwis,
I do not see which benefits would bring making the CAN the same subcommand as for UDP. Maybe I'm not understanding you correctly, but are you suggesting to use the same subcommand, then underneath decide whether to use CAN or UDP based on which configuration profile is enabled?

vibnwis, 2020-11-05:
If I understand correctly, you are querying the CAN implementation of
replacing UDPsubcommand. If that is the case, I am totally with you. The
reasons why I suggest that is purely intending to reduce/avoid any
implication and complication of work involved when introducing a new
transport protocol. As for now my priority is to get it work being a
proof-of-concept. However, based on your expertise, if you think it is too
complicated, I wouldn't mind going along with it. Would you mind sharing
your thoughts, please?

vibnwis, 2020-11-05:
That was a simplistic idea with intention to get it working first
(hard-coding) and revisiting the configuration later. On second thought, I
decided to start a new CAN-bus Agent instead. I am reading the
IPv4EndPoint.hpp  at the moment. Would you mind explaining the file/folder
structure of the UAgent, such as in the transport folder, it has
"discovery", "endpoint", "udp", and etc. If these are basically the
implementation of the various transport protocols. What are the "discovery"
and "endpoint" for and the UAGENT_DISCOVERY_PROFILE, and anything relating
to profile selection.

Would you mind, if I write up a brief design, a.k.a my implementation for
you to have a look at, please?

jamoralp, 2020-11-05:
Ok, I see what you mean now. Sorry, I misunderstood. If you find it easier to kind of "mock" the UDP agent to get a CAN agent proof of concept up and running quickly, we're ok with that, but the final implementation should be independent.
Yes, you are right. Transport folder is in charge of albegating the different implementations of the supported transports for the agent, as well as the common interface for each of them (that is, the `eprosima::uxr::Server` class. Regarding the discovery, IPv4EndPoint is used, since the discovery mechanism is implemented using UDP transport. the discovery is not a feature that we would always want to include in the final agent's executable, so it is pluggable and can be remove by means of the UAGENT_DISCOVERY_PROFILE parameter.

Most of the other's agent repository folders are related with how the agent schedules the incoming/outcoming messages from XRCE to DDS, and viceversa, as well as defining an abstract interface that then can plug in different implementations of DDS. Initially, Fast-RTPS as well as Fast-DDS implementations are provided, under the `middleware` folder, but it is prepared to use others. Anyway, this is one layer above the transport, so you shouldn't worry about them.

Of course, as soon as you have a draft about your implementation proposal, please write it down here and we'll have a look at it. Thanks and I hope I made myself clear :slightly_smiling_face:

vibnwis, 2020-11-05:
All good! Many thanks

vibnwis, 2020-11-06:
Hi jamoralp,
I was eager to give a go by replicating the UDPv4Agent and UDPv4 related header and source files. I have not changed the logic in those classes' body. I managed to pass the compilation, but failed at linking. Obviously, I missed out something there. Would you mind shade some lights here please. I copied a output of the compilation. See the ending parts for the linking failure. Very appreciate your comments.

Scanning dependencies of target microxrcedds_agent
[  2%] Building CXX object CMakeFiles/microxrcedds_agent.dir/src/cpp/transport/udp/CANAgentLinux.cpp.o
[  4%] Building CXX object CMakeFiles/microxrcedds_agent.dir/src/cpp/transport/udp/UDPv4AgentLinux.cpp.o
[  6%] Building CXX object CMakeFiles/microxrcedds_agent.dir/src/cpp/transport/udp/UDPv6AgentLinux.cpp.o
[  8%] Building CXX object CMakeFiles/microxrcedds_agent.dir/src/cpp/transport/tcp/TCPv4AgentLinux.cpp.o
[ 10%] Building CXX object CMakeFiles/microxrcedds_agent.dir/src/cpp/transport/tcp/TCPv6AgentLinux.cpp.o
[ 13%] Building CXX object CMakeFiles/microxrcedds_agent.dir/src/cpp/transport/serial/SerialAgentLinux.cpp.o
[ 15%] Building CXX object CMakeFiles/microxrcedds_agent.dir/src/cpp/transport/serial/TermiosAgentLinux.cpp.o
[ 17%] Building CXX object CMakeFiles/microxrcedds_agent.dir/src/cpp/transport/serial/PseudoTerminalAgentLinux.cpp.o
[ 19%] Building CXX object CMakeFiles/microxrcedds_agent.dir/src/cpp/transport/discovery/DiscoveryServerLinux.cpp.o
[ 21%] Building CXX object CMakeFiles/microxrcedds_agent.dir/src/cpp/transport/p2p/AgentDiscovererLinux.cpp.o
[ 23%] Building CXX object CMakeFiles/microxrcedds_agent.dir/src/cpp/transport/discovery/DiscoveryServer.cpp.o
[ 26%] Building CXX object CMakeFiles/microxrcedds_agent.dir/src/cpp/types/TopicPubSubType.cpp.o
[ 28%] Building CXX object CMakeFiles/microxrcedds_agent.dir/src/cpp/xmlobjects/xmlobjects.cpp.o
[ 30%] Building CXX object CMakeFiles/microxrcedds_agent.dir/src/cpp/middleware/fast/FastEntities.cpp.o
[ 32%] Building CXX object CMakeFiles/microxrcedds_agent.dir/src/cpp/middleware/fast/FastMiddleware.cpp.o
[ 34%] Building CXX object CMakeFiles/microxrcedds_agent.dir/src/cpp/middleware/fastdds/FastDDSEntities.cpp.o
[ 36%] Building CXX object CMakeFiles/microxrcedds_agent.dir/src/cpp/middleware/fastdds/FastDDSMiddleware.cpp.o
[ 39%] Building CXX object CMakeFiles/microxrcedds_agent.dir/src/cpp/middleware/ced/CedEntities.cpp.o
[ 41%] Building CXX object CMakeFiles/microxrcedds_agent.dir/src/cpp/middleware/ced/CedMiddleware.cpp.o
[ 43%] Building CXX object CMakeFiles/microxrcedds_agent.dir/src/cpp/transport/p2p/AgentDiscoverer.cpp.o
[ 45%] Building CXX object CMakeFiles/microxrcedds_agent.dir/src/cpp/p2p/InternalClientManager.cpp.o
[ 47%] Building CXX object CMakeFiles/microxrcedds_agent.dir/src/cpp/p2p/InternalClient.cpp.o
[ 50%] Linking CXX shared library libmicroxrcedds_agent.so
[ 95%] Built target microxrcedds_agent
Scanning dependencies of target MicroXRCEAgent
[ 97%] Building CXX object CMakeFiles/MicroXRCEAgent.dir/microxrce_agent.cpp.o
[100%] Linking CXX executable MicroXRCEAgent
/usr/bin/ld: libmicroxrcedds_agent.so.1.4.2: undefined reference to `eprosima::uxr::Server<eprosima::uxr::CANEndPoint>::enable_discovery(unsigned short)'
/usr/bin/ld: libmicroxrcedds_agent.so.1.4.2: undefined reference to `eprosima::uxr::Server<eprosima::uxr::CANEndPoint>::Server(eprosima::uxr::Middleware::Kind)'
/usr/bin/ld: libmicroxrcedds_agent.so.1.4.2: undefined reference to `eprosima::uxr::DiscoveryServer<eprosima::uxr::CANEndPoint>::discovery_loop()'
/usr/bin/ld: libmicroxrcedds_agent.so.1.4.2: undefined reference to `eprosima::uxr::Server<eprosima::uxr::CANEndPoint>::~Server()'
/usr/bin/ld: libmicroxrcedds_agent.so.1.4.2: undefined reference to `eprosima::uxr::DiscoveryServer<eprosima::uxr::CANEndPoint>::stop()'
/usr/bin/ld: libmicroxrcedds_agent.so.1.4.2: undefined reference to `eprosima::uxr::Server<eprosima::uxr::CANEndPoint>::enable_p2p(unsigned short)'
/usr/bin/ld: libmicroxrcedds_agent.so.1.4.2: undefined reference to `typeinfo for eprosima::uxr::Server<eprosima::uxr::CANEndPoint>'
/usr/bin/ld: libmicroxrcedds_agent.so.1.4.2: undefined reference to `eprosima::uxr::Server<eprosima::uxr::CANEndPoint>::stop()'
/usr/bin/ld: libmicroxrcedds_agent.so.1.4.2: undefined reference to `eprosima::uxr::Server<eprosima::uxr::CANEndPoint>::start()'
/usr/bin/ld: libmicroxrcedds_agent.so.1.4.2: undefined reference to `eprosima::uxr::DiscoveryServerLinux<eprosima::uxr::CANEndPoint>::DiscoveryServerLinux(eprosima::uxr::Processor<eprosima::uxr::CANEndPoint> const&)'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/MicroXRCEAgent.dir/build.make:87: MicroXRCEAgent] Error 1
make[1]: *** [CMakeFiles/Makefile2:78: CMakeFiles/MicroXRCEAgent.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
"make all" terminated with exit code 2. Build might be incomplete.

vibnwis, 2020-11-07:
I have structurally created CAN transport and compiled without errors. The files I have recreated are:
   - Source - CANAgentLinux.cpp, DiscoveryCANServerLinux.cpp,
  - Header - CANAgentLinux.hpp,DiscoveryCANServerLinux.hpp,CANEndPoint.hpp.

Despite, for time-being, the CANEndPoint class is a replical of IPv4EndPoint, however, I foresee, CANEndPoint should have it own data-structure. Also, in the DiscoveryCANServerLinux.cpp, the recv_message() and the send_message() still use the IPv4EndPoint, as the base-class specifies IPv4EndPoint. See below.
                  virtual bool recv_message(
                              InputPacket<IPv4EndPoint>& input_packet,
                              int timeout) = 0;
I thought I could override it, but no avails. Despite I know C++, but it is not my usual programming language.  Feel free to comment especially the implementation I did.

Meanwhile, I will move on to adding the CAN-Socket replacing the Network-Socket.

Also, I have not run any of those u-clients as well as u-agent, do you have any learning materials, youtube or descriptives.

Feel free to comment.

Many thanks

vibnwis, 2020-11-07:
I have checked-in onto the fork. Feel free to have a look or perform what is needed.

pablogs9, 2020-11-10:
@jamoralp @pablogs9  I'll keep this open in order to check this fork: https://github.com/vibnwis/Micro-XRCE-DDS-Agent
and evaluate the integration in the mainline of the XRCE Agent

jamoralp, 2021-03-02:
Closing since inactive for 4 months.
Just FYI and for anyone else, we have developed a whole new approach for *custom transports* besides the ones that we already provide as built-in within the default `MicroXRCEAgent` executable. You can have a look at the provided example [here](https://github.com/eProsima/Micro-XRCE-DDS-Agent/blob/develop/examples/custom_agent/custom_agent.cpp).

closed 2021-03-02 by jamoralp

vibnwis, 2021-03-02:
Is this backward compatible with the old implementations? I have implemented a CAN-transport, FYI.

jamoralp, 2021-03-02:
What do you mean backwards compatible?
We are preparing a huge update regarding the documentation repository where maybe this is all explained a little bit better. But the thing is, now, you will have the standalone `MicroXRCEAgent` executable, which allows to launch it using `udp4`, `udp6`, `tcp4`, `tcp6`, `serial` and `pseudoterminal` as built-in transports. For custom transports you must create a new executable (as in the example I linked above), then fill the four required methods (CustomAgent::InitFunction, CustomAgent::FiniFunction and CustomAgent::RecvMsgFunction and CustomAgent::SendMsgFunction) and then create an instance of the CustomAgent class.
If, following this approach, you could create a pull request using your CAN implementation, that would surely be great and we will be happy to include it in the examples folder!

vibnwis, 2021-03-02:
"What do you mean backwards compatible?" As I have implemented CAN-bus transport using the old method, will the new repo work with the old method of custom transport implementation? I presume it will not. I think for time being, I will use what I have got.

jamoralp, 2021-03-03:
ok, just if you can share the repository and branch where your implementation is hosted, then we can have a look at it and think about porting it to the new format. Thanks

vibnwis, 2021-03-03:
The link is https://github.com/vibnwis/Micro-XRCE-DDS-Agent. However, I
have no intention to port it to the new platform for time being due to the
tight schedule with my studies. Furthermore, due to the limited CAN-Bus
frame size, it does not meet the reliable communication protocol despite it
is a working CAN-bus transport.