eProsima/Micro-XRCE-DDS-Agent #196: CLI option and CANAgent server arguments

vibnwis, 2020-11-10:
Hi Jose,
1. I am now reading the CLI Class, and wanting to add an option, a CAN interface, eg. "can0", With I understanding (correct me if I am wrong), from the command line, user's selection will be parsed and then passed as argument when calling the corresponding server.  Would you help, where would be best place and how to add in the CLI code, please? My apologies for my request as I am uncertain of the thing I add will work or not. Elaborate please? I would like to conform to the existing CLI style.

2. In CLI.hpp, the lines that calls CANAgent are : 
 void launch_server()
    {
        server_.reset(new eprosima::uxr::CANAgent(port_, common_opts_.middleware_opt_.get_kind()));
        if (server_->start())
        {

To allow the new CAN transport to work with least disruption, what would you suggest to use/replace with  when calling the CANAgent server. FYI, the only argument that CAN transport needs is the interface, can be string, for instance  "can0" or "can1" ? However, I can take numeric value, 0 or 1 from which I can add string "can" to it. My worries are, if I change/add new arguments that the existing application algorithms may be catastrophic.  Another example, when receiving a message, CAN-transport does not have addr, port. However, I can fill the addr and the port respectively with CANID, CAN-message-Length, but will it cause the other algorithms to fail.  
      eg, uint32_t addr = client_addr.sin_addr.s_addr;
            uint16_t port = client_addr.sin_port;
            input_packet.source = CANEndPoint(addr, port);  <---- <CANEndPoint> is a replica of IPv3EndPoint for time being.

Appreciate your suggestion.

Hear from you soon.
Thanks in advance.
vi

jamoralp, 2020-11-11:
Hello,

Sorry for the late reply, but this week the ROS World takes place and we are quite busy.
Regarding this, I saw you already included some code in the `uxr/agent/utils/ArgumentParser.hpp` file. This is our custom CLI implementation and will be the only one that remains in the future. The plan is to delete everything related with CLI11 library.

I checked your code, and saw you are including a port parameter in the CanArgs class. This is not mandatory. Only parameters that are common to every AgentKind and check are the ones present in the CommonArgs class. Therefore, your CanAgent could only have a string parameter, namely the interface. Check for instance `class PseudoTerminalArgs`. Argument class is a template with two specializations, one for numeric values and another for string values, so you can play with that.

Hope this helps a little bit. Please write down any other question and I will answer as soon as I have more free time (Friday morning, probably. ROS World is tomorrow :smiley: )

PS: I haven't digged enough into your implementation of the CAN agent using CANSocket, since I am not familiar with it. But I promise I will, since it seems to be an interesting approach on how to face CAN connections.

jamoralp, 2021-03-02:
Closing since inactive for 4 months.
Just FYI and for anyone else, this should not be necessary from now on. We have developed a whole new approach for *custom transports* besides the ones that we already provide as built-in within the default `MicroXRCEAgent` executable. You can have a look at the provided example [here](https://github.com/eProsima/Micro-XRCE-DDS-Agent/blob/develop/examples/custom_agent/custom_agent.cpp).

closed 2021-03-02 by jamoralp