eProsima/Micro-XRCE-DDS-Agent #197: recv_message and how to write back into input_packet and read from out_packet

vibnwis, 2020-11-13:
Basically I am stuck with how to passing the CAN frames/contents up for upper layers for processing. If I am not wrong, the following recv_message()  will call the CAN recv_mesg. Assuming that I have read a CAN frame, how do I convert it to InputPacket?

void Server<EndPoint>::receiver_loop()
{
    InputPacket<EndPoint> input_packet{};
    while (running_cond_)
    {
        TransportRc transport_rc = TransportRc::ok;
        if (recv_message(input_packet, RECEIVE_TIMEOUT, transport_rc))
        {
            input_scheduler_.push(std::move(input_packet), 0);
        }
        ....

bool CAN2Agent::recv_message(
        InputPacket<CAN2EndPoint>& input_packet,
        int timeout,
        TransportRc& transport_rc)
{
    bool rv = false;
    SocketCanStatus can_status = STATUS_CAN_OP_ERROR;
    struct CanFrame can_msg;
    int i = 0;

    if (socket_can.read(can_msg) != STATUS_OK)
    {
    	printf("*** CAN - read error *** \n");
    	transport_rc = TransportRc::server_error;
    }

I manage to read the content of the CAN interface, however, I have no idea how to pass the content back to the input_packet, despite long research. I thought it would be easier to replicate the <serialEndPoint> structure which I have done it later, namely CAN2EndPoint.  Unfortunately, I still not able to write back to the input_packet. Would you shed some light here, please?

vibnwis, 2020-11-13:
Since bool CAN2Agent::send_message(
        OutputPacket<CAN2EndPoint> output_packet,
        TransportRc& transport_rc)
and 
bool CAN2Agent::recv_message(
        InputPacket<CAN2EndPoint>& input_packet,
        int timeout,
        TransportRc& transport_rc)

and the <CAN2EndPoint> is extended from <EndPoint>
I use the "source" and "destination" members of respective input_packet, and output_packet structures for writing to and reading from the values,which are different from using "OutputMessagePtr" and  "InputMessagePtr". Let me know if that work for the rest of the system, please?

jamoralp, 2020-11-17:
Hello and sorry for the late reply. Last week we were very busy with ROS World 2020 and I could not address this issue. Also please try to keep all the information regarding this CAN transport into a single issue, because that way is easier to track it. There are currently three different issues open in the Micro XRCE-DDS Agent repository adressing this port.

Regarding your question, it should be very similar to the logics involved in other transport using Linux sockets. InputPacket must be filled with a source member, which is the endpoint information; and a message member, which is an InputMessage class instance, initializated with the raw `uint8_t *` buffer read from the socket, and the number of bytes read. Please have a look at `TCPv$AgentLinux.cpp:412 (read_message)` method implementation.

vibnwis, 2020-12-03:
Hello, 
I am wanting to use gdb for debugging the code. Is it possible to generate gdb debugging code with the existing Makefile. I have search Makefile, but could not identify anything, compilation options which I use or add. Any advice?

jamoralp, 2020-12-03:
Good morning @vibnwis, you can indeed build the Agent using the `-DCMAKE_BUILD_TYPE=DEBUG` flag to generate symbols that can later be used by `gdb` to debug your code. I do not really understand your question

vibnwis, 2020-12-03:
Thank you for the answer. That is exactly what I wanted to know.

vibnwis, 2020-12-03:
I think, I asked this question before, is the UAGENT_DISCOVERY_PROFILE a
must? I wakely remember, it is not necessary. Please advise.

jamoralp, 2020-12-03:
> I think, I asked this question before, is the UAGENT_DISCOVERY_PROFILE a must? I wakely remember, it is not necessary. Please advise.
> […](#)

Well, not really. The Discovery profile enables a discovery server that allow clients to automatically retrieve information about which agent(s) are present in their network, by means of sending a GET_INFO petition to which the discovery server answers with pairs of IP addresses and ports to which the client tries to establish a session to. If you know your agent's IP and port prior to connecting, which is the case most of the times, you can disable this feature.

vibnwis, 2020-12-03:
CAN-bus has not address or any forms of identification. Connection is
established at the physical layer through arbitration.
I presume the option is -D UAGENT_DISCOVERY_PROFILE = OFF

jamoralp, 2020-12-03:
Yes, that is right :slightly_smiling_face:

vibnwis, 2020-12-03:
1. Also, I switched Off the UAGENT_P2P_PROFILE. I presume it is
consequential.

2. I was not sure what options for running uagent. I use MicroXRCEAgent
can2 dds. However, I am still not able to run to completion. Feel free to
advise.

Thank you in advance.

vibnwis, 2020-12-04:
I am trying to get uagent to run. Obviously, I don't if mine will work. which in facts, I tried many times, it failed. Hence, I tried with udp instead with the following commands.
> ./MicroXRCEAgent udp, ./MicroXRCEAgent udp -p=8888,./MicroXRCEAgent udp -p 8888  
None of those work. It keeps displaying messages "A subcommand is required
Run with --help for more information."
At "Enter command", I have tried, udp, tcp, can2, can, 8888, also none of those work. 
I am sure what it is expecting. Can you advise, please?

vibnwis, 2020-12-04:
Okie, I managed to advance one step. However, now it shows an error.

Enter command: can2 -d 222
Press CTRL+C to exit
terminate called after throwing an instance of 'std::logic_error'
  what():  basic_string::_M_construct null not valid

I am trying to resolve it. Appreciate your advice, please?

vibnwis, 2020-12-04:
To be honest, C++ is not something I use often. This is my time dealing with such highly professional C++ program. It is a bit overwhelm for me. Also, the editor is not able to jump to the declarations or references, make it very hard to grab to good understanding.  
I try to look for where the error, "'std::logic_error'
what(): basic_string::_M_construct null not valid" was generated. Still searching  since the last message. Hence, very appreciate your advice, please?

jamoralp, 2020-12-04:
Hello, how could I test all these in order to get to the point you are stuck with? Which branch do I have to use? I assume I should refer to your fork of the Micro XRCE-DDS Agent.
Also, regarding the error, it seems like you are using the "old" CLI. It can get a little bit messy when entering not valid commands. Did you add the `can` option in the `include/uxr/agent/utils/CLI.hpp`? If not, please do so. If you want to switch to the new CLI (which is lighter, and easier in my opinion) you should compile the agent with the `-DUAGENT_CLI_PROFILE=OFF`, then check that the can option is correctly added in `uxr/agent/utils/argumentParser.hpp`. We are planning to eventually remove the former version, anyways.

vibnwis, 2020-12-04:
Yes, I have added the CAN (old) and CAN2 (newer) options in the
include/uxr/agent/utils/CLI.hpp and uxr/agent/utils/argumentParser.hpp. I
will use the CLI by adding -DUAGENT_CLI_PROFILE=OFF and see how it does.
As for testing at your, you can get a copy from the fork of the Micro
XRCE-DDS Agent. However, probably not a good idea until next monday, as I
have not uploaded the latest changes, but I will do it over the weekend.
Hence, next Monday is better.

jamoralp, 2020-12-04:
ok, do not worry. We are on holiday in Spain Monday and Tuesday, so I will test your changes on Wednesday. Feel free to add any update in this issue.

Have a nice weekend!

vibnwis, 2020-12-09:
hi,
I have some parts of uagent working. It could receive a complete from micro-client. However, the Read() SocketCAN failed with "segmentation fault". I am looking into the problem now. 
Back to micro-client. At the moment, I have removed the very first transmitting frame. I added it before allowing uagent to test its receiving functions. 
Here are my questions.
When running uagent, It goes to CAN2Agent::init(), and then CAN2Agent::recv_message(). From what I observe, both the uclient and uagent are at the listening state. What messaging would take place? why no messaging take place? what have I missed out?

Very appreciate your comments.

vibnwis, 2020-12-09:
Further test, I experienced the following issues. Perhaps it is easier to show the output of the CAN-Transport communication.
In general, uagent succeeded initializing SocketCAN, and moved onto recv_message() from SocketCAN successfully. However, it came back to recv_message(), and caused "segmentation fault" at read(). However, I have re-run the read() on it own, which posed no issues. However, the integrated with uagent experienced "segmentation faults". From gdb, it stopped at session_id(), and the system exit(). I wonder why causes the error. See the output messages for more information.

SocketCAN  bind() succeeded
SocketCAN  getsockopt() succeeded
SocketCAN  error none
SocketCAN for interface can0 
 succeeded
CAN2Agent::recv_message() entered 
SocketCAN::read  getsockopt() succeeded len = 4
SocketCAN:: Read() - About to read from CAN interface can0
Read() returns values = 16 and data length of 8
SocketCAN: Read() Classis frame.flags = 291 
SocketCAN: Read() Non-Error frame.flags = 291 
SocketCAN: Read() frame.can_dlc= 8, frame.can_id = 291
data[17]data[34]data[51]data[68]data[85]data[102]data[119]data[136]
SocketCAN: Read() ended
len 8 byte, id: 291 data[0]: 11 data[1]: 22 data[2]: 33 data[3]: 44 data[4]: 55 data[5]: 66 data[6]: 77 data[7]: 88 
write frame_id to source of input_packet 
write frame_len to source of input_packet 
write frame_data to source of input_packet 
set_data()- len 8
set_data()- ended
CAN2Agent::recv_message() exited 
CAN2Agent::recv_message() entered 
SocketCAN::read  getsockopt() succeeded len = 4
SocketCAN:: Read() - About to read from CAN interface can0

vibnwis, 2020-12-09:
I suspect re-entering of re-entering to CAN2Agent::recv_message() may have modified the data-structure of SocketCAN socket_can_. At here, I have confirmed that it works by itself. If possible would you mind verify the variable definitions in CAN2AgentLinux and the scheduler that manages the recv_message(). I wonder if that makes sense.

vibnwis, 2020-12-10:
Here are the traces with GDB, I have verified and confirmed that the SocketCAN handler and  the memory space for struct can_frame are intact, not corrupted. Please see the second segment for the messages. The s, CAN socket handler and the cframe are not destroyed. Appreciate your comments please?
-------------------------gdb traces -----------------------------
Temporary breakpoint 1, main (argc=21845, argv=0x0) at /home/wiki-ros/ms-bot-evn/Micro-XRCE-DDS-Agent/microxrce_agent.cpp:18
18      {
(gdb) [New Thread 0x7ffff5e00700 (LWP 84523)]                                                
[New Thread 0x7ffff55ff700 (LWP 84524)]
[New Thread 0x7ffff4dfe700 (LWP 84525)]
[New Thread 0x7fffeffff700 (LWP 84526)]
[New Thread 0x7fffef7fe700 (LWP 84527)]
[Switching to Thread 0x7ffff4dfe700 (LWP 84525)]

Thread 4 "MicroXRCEAgent" hit Breakpoint 4, eprosima::uxr::SocketCan::read (this=0x7ffff00112e8, msg=...) at /home/wiki-ros/ms-bot-evn/Micro-XRCE-DDS-Agent/src/cpp/transport/can/socketcan_cpp.cpp:345
345             printf("SocketCAN:: Read() - A Frame memory space is ready and about to read from CAN interface can0\n");
[Switching to thread 6 (Thread 0x7fffef7fe700 (LWP 84527))](running)
[New Thread 0x7fffeeffd700 (LWP 84533)]
[Switching to thread 6 (Thread 0x7fffef7fe700 (LWP 84527))](running)
[Switching to thread 1 (Thread 0x7ffff5e0c280 (LWP 84512))]
#0  __pthread_clockjoin_ex (threadid=140737318487808, thread_return=0x0, clockid=<optimized out>, abstime=<optimized out>, block=<optimized out>) at pthread_join_common.c:145
145     pthread_join_common.c: No such file or directory.
[Switching to thread 7 (Thread 0x7fffeeffd700 (LWP 84533))](running)
[Switching to thread 7 (Thread 0x7fffeeffd700 (LWP 84533))](running)
[Switching to thread 7 (Thread 0x7fffeeffd700 (LWP 84533))](running)

Thread 4 "MicroXRCEAgent" hit Breakpoint 4, eprosima::uxr::SocketCan::read (this=0x7ffff00112e8, msg=...) at /home/wiki-ros/ms-bot-evn/Micro-XRCE-DDS-Agent/src/cpp/transport/can/socketcan_cpp.cpp:345
345             printf("SocketCAN:: Read() - A Frame memory space is ready and about to read from CAN interface can0\n");

Thread 6 "MicroXRCEAgent" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffef7fe700 (LWP 84527)]
0x00007ffff7d00394 in dds::xrce::MessageHeader::session_id (this=0x10) at /home/wiki-ros/ms-bot-evn/Micro-XRCE-DDS-Agent/include/uxr/agent/types/MessageHeader.hpp:90
90              return m_session_id;
[Switching to thread 6 (Thread 0x7fffef7fe700 (LWP 84527))]
#0  0x00007ffff7d00394 in dds::xrce::MessageHeader::session_id (this=0x10) at /home/wiki-ros/ms-bot-evn/Micro-XRCE-DDS-Agent/include/uxr/agent/types/MessageHeader.hpp:90
90              return m_session_id;
[Thread 0x7fffef7fe700 (LWP 84527) exited]
[Thread 0x7fffeffff700 (LWP 84526) exited]
[Thread 0x7ffff4dfe700 (LWP 84525) exited]
[Thread 0x7ffff55ff700 (LWP 84524) exited]
[Thread 0x7ffff5e00700 (LWP 84523) exited]
[Thread 0x7ffff5e0c280 (LWP 84512) exited]

Program terminated with signal SIGSEGV, Segmentation fault.
The program no longer exists.

---------------------------------------------------- printf --------------------------------------
an_frame is 16 bytesSocketCAN  bind() succeeded
SocketCAN  getsockopt() succeeded
SocketCAN  error none
SocketCAN for interface can0 
 succeeded
CAN2Agent::recv_message() entered 
SocketCAN:: handler 3 
SocketCAN::read  getsockopt() succeeded len = 4
SocketCAN:: Read() - A Frame memory space is ready and about to read from CAN interface can0
Read() returns values = 16 and data length of 8
SocketCAN: Read() Classis frame.flags = 291 
SocketCAN: Read() Non-Error frame.flags = 291 
SocketCAN: Read() frame.can_dlc= 8, frame.can_id = 291
data[17]data[34]data[51]data[68]data[85]data[102]data[119]data[136]
SocketCAN: Read() ended
len 8 byte, id: 291 data[0]: 11 data[1]: 22 data[2]: 33 data[3]: 44 data[4]: 55 data[5]: 66 data[6]: 77 data[7]: 88 
write frame_id to source of input_packet 
write frame_len to source of input_packet 
write frame_data to source of input_packet 
set_data()- len 8
set_data()- ended
CAN2Agent::recv_message() exited 
CAN2Agent::recv_message() entered 
SocketCAN:: handler 3 
SocketCAN::read  getsockopt() succeeded len = 4
SocketCAN:: Read() - A Frame memory space is ready and about to read from CAN interface can0
Segmentation fault (core dumped)

vibnwis, 2020-12-10:
Hi,
I single stepped through using gdb. I found out that after CAN2AgentLinux::recv_message(), the system control passed back to the scheduler of void Server<EndPoint>::receiver_loop(), and then segmentation fault took place. See the gdb messages below. Kindly advise please.

------------------------
Temporary breakpoint 1, main (argc=21845, argv=0x0) at /home/wiki-ros/ms-bot-evn/Micro-XRCE-DDS-Agent/microxrce_agent.cpp:18
18      {
[New Thread 0x7ffff5e00700 (LWP 98532)]
[New Thread 0x7ffff55ff700 (LWP 98533)]
[New Thread 0x7ffff4dfe700 (LWP 98534)]
[New Thread 0x7fffeffff700 (LWP 98535)]
[New Thread 0x7fffef7fe700 (LWP 98536)]
[Switching to Thread 0x7ffff4dfe700 (LWP 98534)]

Thread 4 "MicroXRCEAgent" hit Breakpoint 3, eprosima::uxr::SocketCan::read (this=0x7ffff00112e8, msg=...) at /home/wiki-ros/ms-bot-evn/Micro-XRCE-DDS-Agent/src/cpp/transport/can/socketcan_cpp.cpp:345
345             printf("SocketCAN:: Read() - A Frame memory space is ready and about to read from CAN interface can0\n");
[Switching to thread 6 (Thread 0x7fffef7fe700 (LWP 98536))](running)
[New Thread 0x7fffeeffd700 (LWP 98564)]
[Switching to thread 6 (Thread 0x7fffef7fe700 (LWP 98536))](running)
[Switching to thread 1 (Thread 0x7ffff5e0c280 (LWP 98520))]
#0  __pthread_clockjoin_ex (threadid=140737318487808, thread_return=0x0, clockid=<optimized out>, abstime=<optimized out>, block=<optimized out>) at pthread_join_common.c:145
145     pthread_join_common.c: No such file or directory.
[Switching to thread 7 (Thread 0x7fffeeffd700 (LWP 98564))](running)
[Switching to thread 7 (Thread 0x7fffeeffd700 (LWP 98564))](running)
[Switching to thread 7 (Thread 0x7fffeeffd700 (LWP 98564))](running)
[Switching to thread 7 (Thread 0x7fffeeffd700 (LWP 98564))](running)
[Switching to thread 7 (Thread 0x7fffeeffd700 (LWP 98564))](running)

Thread 1 "MicroXRCEAgent" received signal SIGINT, Interrupt.
[Switching to Thread 0x7ffff5e0c280 (LWP 98520)]
__pthread_clockjoin_ex (threadid=140737318487808, thread_return=0x0, clockid=<optimized out>, abstime=<optimized out>, block=<optimized out>) at pthread_join_common.c:145
145     in pthread_join_common.c
[Switching to thread 7 (Thread 0x7fffeeffd700 (LWP 98564))](running)
[Switching to Thread 0x7ffff4dfe700 (LWP 98534)]

Thread 4 "MicroXRCEAgent" hit Breakpoint 9, eprosima::uxr::SocketCan::read (this=0x7ffff00112e8, msg=...) at /home/wiki-ros/ms-bot-evn/Micro-XRCE-DDS-Agent/src/cpp/transport/can/socketcan_cpp.cpp:356
356             if (num_bytes == 0)
[Switching to thread 7 (Thread 0x7fffeeffd700 (LWP 98564))](running)

Thread 4 "MicroXRCEAgent" hit Breakpoint 6, eprosima::uxr::SocketCan::read (this=0x7ffff00112e8, msg=...) at /home/wiki-ros/ms-bot-evn/Micro-XRCE-DDS-Agent/src/cpp/transport/can/socketcan_cpp.cpp:424
424             printf("\nSocketCAN: Read() ended\n");
[Switching to thread 7 (Thread 0x7fffeeffd700 (LWP 98564))](running)
[Switching to thread 7 (Thread 0x7fffeeffd700 (LWP 98564))](running)

Thread 4 "MicroXRCEAgent" hit Breakpoint 7, eprosima::uxr::SocketCan::read (this=0x7ffff00112e8, msg=...) at /home/wiki-ros/ms-bot-evn/Micro-XRCE-DDS-Agent/src/cpp/transport/can/socketcan_cpp.cpp:428
428             return STATUS_OK;
[Switching to thread 7 (Thread 0x7fffeeffd700 (LWP 98564))](running)
[Switching to thread 7 (Thread 0x7fffeeffd700 (LWP 98564))](running)
[Switching to thread 7 (Thread 0x7fffeeffd700 (LWP 98564))](running)
[Switching to thread 7 (Thread 0x7fffeeffd700 (LWP 98564))](running)
[Switching to thread 7 (Thread 0x7fffeeffd700 (LWP 98564))](running)
[Switching to thread 7 (Thread 0x7fffeeffd700 (LWP 98564))](running)
[Switching to thread 7 (Thread 0x7fffeeffd700 (LWP 98564))](running)
[Switching to thread 7 (Thread 0x7fffeeffd700 (LWP 98564))](running)
[Switching to thread 7 (Thread 0x7fffeeffd700 (LWP 98564))]
#0  0x00007ffff75833bf in __GI___clock_nanosleep (clock_id=clock_id@entry=0, flags=flags@entry=0, req=0x7fffeeffcd60, rem=0x7fffeeffcd60) at ../sysdeps/unix/sysv/linux/clock_nanosleep.c:78
78      ../sysdeps/unix/sysv/linux/clock_nanosleep.c: No such file or directory.
[Switching to thread 7 (Thread 0x7fffeeffd700 (LWP 98564))](running)
[Switching to thread 7 (Thread 0x7fffeeffd700 (LWP 98564))](running)
[Switching to thread 7 (Thread 0x7fffeeffd700 (LWP 98564))](running)
[Switching to thread 7 (Thread 0x7fffeeffd700 (LWP 98564))]
#0  0x00007ffff75833bf in __GI___clock_nanosleep (clock_id=clock_id@entry=0, flags=flags@entry=0, req=0x7fffeeffcd60, rem=0x7fffeeffcd60) at ../sysdeps/unix/sysv/linux/clock_nanosleep.c:78
78      in ../sysdeps/unix/sysv/linux/clock_nanosleep.c
[Switching to thread 7 (Thread 0x7fffeeffd700 (LWP 98564))](running)
[Switching to thread 7 (Thread 0x7fffeeffd700 (LWP 98564))](running)
[Switching to thread 7 (Thread 0x7fffeeffd700 (LWP 98564))](running)
[Switching to thread 7 (Thread 0x7fffeeffd700 (LWP 98564))](running)
[Switching to thread 7 (Thread 0x7fffeeffd700 (LWP 98564))](running)
[Switching to thread 7 (Thread 0x7fffeeffd700 (LWP 98564))](running)
[Switching to thread 7 (Thread 0x7fffeeffd700 (LWP 98564))]
#0  0x00007ffff75833bf in __GI___clock_nanosleep (clock_id=clock_id@entry=0, flags=flags@entry=0, req=0x7fffeeffcd60, rem=0x7fffeeffcd60) at ../sysdeps/unix/sysv/linux/clock_nanosleep.c:78
[... 29 lines cut ...]
[Switching to thread 7 (Thread 0x7fffeeffd700 (LWP 98564))](running)
[Switching to thread 7 (Thread 0x7fffeeffd700 (LWP 98564))](running)
[Switching to thread 7 (Thread 0x7fffeeffd700 (LWP 98564))](running)
[... cut ...]
0x00007ffff7d00394 in dds::xrce::MessageHeader::session_id (this=0x10) at /home/wiki-ros/ms-bot-evn/Micro-XRCE-DDS-Agent/include/uxr/agent/types/MessageHeader.hpp:90
90              return m_session_id;
[Switching to thread 7 (Thread 0x7fffeeffd700 (LWP 98564))](running)
[Thread 0x7fffeeffd700 (LWP 98564) exited]
[Thread 0x7fffef7fe700 (LWP 98536) exited]
[Thread 0x7fffeffff700 (LWP 98535) exited]
[Thread 0x7ffff4dfe700 (LWP 98534) exited]
[Thread 0x7ffff55ff700 (LWP 98533) exited]
[Thread 0x7ffff5e00700 (LWP 98532) exited]

Program terminated with signal SIGSEGV, Segmentation fault.
The program no longer exists.

vibnwis, 2020-12-11:
Hi, 
Further probes, I found the cause of the error, ie segmentation fault. By adding these two lines, it resolved the error. 
        input_packet.message.reset(new InputMessage(buffer_, CAN_MTU));
    	input_packet.source = CAN2EndPoint(can_msg.id, can_msg.len);
However, something I am not sure are:
1.  How do I test the CAN2Agent::send_message() routine. The reason is when running in gdb with a breakpoint in the send_message(), it did not pause ie it does not send out any messages. 
bool CAN2Agent::send_message(
        OutputPacket<CAN2EndPoint> output_packet,
        TransportRc& transport_rc)
{
    bool rv = false;
    SocketCanStatus can_status = STATUS_CAN_OP_ERROR;
    struct CanFrame can_msg;
    uint8_t retry = 0;

    printf("CAN2Agent::send_message() entered \n");

    can_msg.id = output_packet.destination.get_id(); <--------------1
    can_msg.len = output_packet.destination.get_len();  <----------2
    
     data_size = output_packet.destination.get_data(can_msg.data);  <-------------3

2. From the code with number 1, 2, 3 above. However, I am not sure where those values are set?

3. The Micro-Client, does not send out any messages as well. It is in listening mode as far I am concerned. What should I expect?

Appreciate your comments, please?

Thank you

vibnwis, 2020-12-13:
I have deleted some old messages because I think it would cause confusion whoever read it, and also it has been resolved. It would be cleaner as well for you to focus on my questions.
Basically, I have completed adding the CAN-transport. Partially I have tested it working, as init(), recv_message(), finit(), except send_message(). However, the main issues I have are: 
1. What and how to get a complete UAgent and U-Client to establish a client-server connection. Currently, I don't any nodes are transmitting any message. Results of my current testing.
2. When transmitting message from a node, it needs to specify the an frame ID for bus arbitration purpose. Being at transport layer protocol, I don't hard-code or add that. At application layer, how does it set it, and the frame size? FYI it can not more than 8 bytes.
I look forward hearing from you soon.
Thank you
Vi

vibnwis, 2020-12-15:
Hi jamoralp 

I sent out this massage to pablo too. 

Okie, further trace, I found that uxr_create_session(&session) of u-client indeed calls send_message() routine below, and the send_msg() also correctly calls the send_can_msg() of CAN-Transport implementation. However, here it comes the issue, from the gdb, I could see that the length of the buffer (ie request message) is 20. However, CAN bus could only take 8 bytes. Would you able to help advising what would be the best strategy for resolving the solution. I can think of reducing it to 8, but I do not know about the implication of it may involve for other parts of the system. Obviously, its counterpart, uagent will need to change accordingly. Would you help to discuss it with jamoralp too. I am happy to make those changes here. However, your advice and assistance are very much needed, please. I look forward to hearing from you soon.

pablogs9, 2020-12-18:
Hello @vibnwis  I have found that there are a couple of persons interested also in a micro-ROS CAN transport, I have created a slack channel to talk about this: https://micro-ros.slack.com/archives/C01H4A538CV

Please join if interested.

vibnwis, 2020-12-18:
Okie. I will

vibnwis, 2020-12-18:
Hi Pablo,
It does not allow me to join except those with domain of

   - de.bosch.com
   - eprosima.com
   - fiware.org
   - piap.pl

pablogs9, 2020-12-18:
Here you have a place to put your mail and we can accept you in the group: https://micro-ros.github.io/ In the bottom right

vibnwis, 2020-12-20:
Hi Pablo and Jose,
I have implemented the CAN-framing protocol that breaking down long messages to 8-bytes CAN frames and at the other end rejoin the CAN-frames to the original message. Would you please advise what is happening, and the issues. From the messages after rejoined, they are identical, however, it still not able to create a session. From the client output message, you can see that the "appMain: Error-Create session fail". Of course, when it hits this, the system simply halts. Please advise.

How to read to files.
[1]. For u-client, 
    "CAN payload: data[0] = 0x3", represents it is sending out frame #4 of a long message. A long message is broken down to 7 bytes, and the very first byte is the frame number. The last frame is "data[0] = 0x0", of which I use to determine to re-join the previously received frames.
  "< CAN2 concatenated frames" is showing the complete message after rejoining together.
  "CAN ID" is CAN identification 
  "CAN DLC" is the CAN frame data length.
  "HAL_CAN_AddTxMessage()"  represents success in sending a CAN frame. The numbers of the message are same as the number CAN breakdown frames of a long messages, ie 24 bytes of a long messages is broken down to 4 CAN frames of 7, 7, 7, 3 bytes length,
[2]. This applies to u-Agent.

[U-agent-session-messaging-comms.txt](https://github.com/eProsima/Micro-XRCE-DDS-Agent/files/5720859/U-agent-session-messaging-comms.txt)
[u-client-CAN-framing session comms.txt](https://github.com/eProsima/Micro-XRCE-DDS-Agent/files/5720865/u-client-CAN-framing.session.comms.txt)

vibnwis, 2020-12-20:
I suspect that the received message not being accepted or passed properly back to the caller of receiving routine.
 This is how the client, recv_can_msg() looks like. The contents of arr[] has copied to transport->buffer, and update its pointer to *buf as well as the the len value. However, I don't see anything abnormal here. 

static bool recv_can_msg(void* instance, uint8_t** buf, size_t* len, int timeout)
{
    bool rv = false;

    uxrUDPTransport* transport = (uxrUDPTransport*)instance;

    uint8_t arr[XRCE_CAN2_MTU];

    uint32_t id, length, msg_len = 0;
    uint8_t fr_num = 0;
    int i;

    /*
     * CAN-bus frame can be read directly the circular buffer.
     *
     * Hence, discard the uxr_read_udp_data_platform(), instead read the CAN frame directly
     * from the circular buffer.
     *
     * However, the platform->fd can be added for handling Txheader (sending out CAN frame) and reading CAN
     * frame from the circular buffer (cb).
     * For time being, it accesses to the low-level data structure directly
     */

    size_t bytes_received = cb_read_CAN_frame(arr, &id, &length);

    printf("\n\r recv_can_msg() bytes_received is %d \n\r",bytes_received );

    if (0 < bytes_received)
    {

    	fr_num = arr[0];
    	printf ("\n\r inFrame id=0x%x, fr_num=%d  length=%d\n\r",id,fr_num, length);
    	msg_len = processInFrame(id, arr, fr_num, length);

    	printf ("\n\r msg_len %d \n\r", msg_len);
    	if (msg_len) {
        	printf ("\n\r< CAN2 concatenated frames \n\r");
        	for (i=0; i<msg_len; i++) {
        		if (!(i%8))
        		    printf ("\n\r");
        		printf("[%d]=0x%x ", i, arr[i]);
        	}
        	printf ("\n\rCAN2 concatenated frames > \n\r");

        	resetIDMessage(id);
        	memcpy(transport->buffer, arr, msg_len);
        	*buf = transport->buffer;
        	*len = msg_len;
        	rv = true;
        }
    }

    return rv;
}

vibnwis, 2020-12-21:
Hi Pablo and Jose,

Further traces, I found that for a session to be created, besides those stream of data, it also needs info.last_requested_status to updated to non-0xFF. However, from the traces, see below, it shows despite having the right data stream ( I almost confident it is so), the info.last_requested_status remains 0xFF. Appreciate your advice, please.

 msg_len 24 

< CAN2 concatenated frames 

[0]=0x80 [1]=0x0 [2]=0x0 [3]=0x0 [4]=0x0 [5]=0x1 [6]=0x10 [7]=0x0 
[8]=0x58 [9]=0x52 [10]=0x43 [11]=0x45 [12]=0x1 [13]=0x0 [14]=0x1 [15]=0xf 
[16]=0xba [17]=0x5e [18]=0xba [19]=0x11 [20]=0x81 [21]=0x0 [22]=0xfc [23]=0xf 
CAN2 concatenated frames > 

 listen_message() must_be_read is 1
 listen_message() data-length is 24 

[0]=0x80 [1]=0x0 [2]=0x0 [3]=0x0 [4]=0x0 [5]=0x1 [6]=0x10 [7]=0x0 
[8]=0x58 [9]=0x52 [10]=0x43 [11]=0x45 [12]=0x1 [13]=0x0 [14]=0x1 [15]=0xf 
[16]=0xba [17]=0x5e [18]=0xba [19]=0x11 [20]=0x81 [21]=0x0 [22]=0xfc [23]=0xf 
session->info.last_requested_status 255 

 listen_message() ended

jamoralp, 2020-12-21:
Hello, thank you for the effort being put into this.

For me, it is quite difficult to understand what is happening if there is no example I can execute to debug the session (I have never really used CAN sockets). Therefore, if possible, could you provide with the exact steps to reproduce the situation you are describing above? Do you have any micro XRCE-DDS Client code that replicates a real CAN buffer device, in order to establish a "fake" connection and being able to debug the code? Thank you.

vibnwis, 2020-12-21:
To be honest, for the following is about u-client as I am solving one at a time. At the moment, I see issues take place in the u-client.

It is understandable. I think, at your end you may able to run or debug without CAN devices. Those output messages were intended to give some information and scenario from which you may relate to the code that may cause the issues, being hopeful. 
However, I think I could give some more information. 
From u-client, uxr_create_session_retries() routine where tries to create a session. By that, it sends out  "[0]=0x80 [1]=0x0 [2]=0x0 [3]=0x0 [4]=0x0 [5]=0x1 [6]=0x10 [7]=0x0
[8]=0x58 [9]=0x52 [10]=0x43 [11]=0x45 [12]=0x1 [13]=0x0 [14]=0x1 [15]=0xf
[16]=0xba [17]=0x5e [18]=0xba [19]=0x11 [20]=0x81 [21]=0x0 [22]=0xfc [23]=0xf", 
, and u-Agent responses to it by sending the similar message.  I could be wrong, as this infers from the output traces. 

It then execute wait_session_status() {
bool wait_session_status(uxrSession* session, uint8_t* buffer, size_t length, size_t attempts)
{
    session->info.last_requested_status = UXR_STATUS_NONE;

    int poll_ms = UXR_CONFIG_MIN_SESSION_CONNECTION_INTERVAL;
    for(size_t i = 0; i < attempts && session->info.last_requested_status == UXR_STATUS_NONE; ++i)
    {
        send_message(session, buffer, length);
        poll_ms = listen_message(session, poll_ms) ? UXR_CONFIG_MIN_SESSION_CONNECTION_INTERVAL : poll_ms * 2;
    }

    return session->info.last_requested_status != UXR_STATUS_NONE;
}
where you can the for-loop conditions, and one of it is session->info.last_requested_status = = UXR_STATUS_NONE. For breaking the loop, session->info.last_requested_status should not be UXR_STATUS_NONE.

In the routine, it calls listen_message() 

bool listen_message(uxrSession* session, int poll_ms)
{
    uint8_t* data; size_t length;
    int rv = 0, i;
    bool must_be_read = recv_message(session, &data, &length, poll_ms);

    if (must_be_read == true)
    	rv = 1;

    printf("\n\r listen_message() must_be_read is %d", rv);
    printf("\n\r listen_message() data-length is %d \n\r", length);

    for (i=0; i<length; i++) {
    	if (!(i%8))
    	    printf ("\n\r");
    	printf("[%d]=0x%x ", i, (*(data + i)));

    }

    if(must_be_read)
    {
        ucdrBuffer ub;
        ucdr_init_buffer(&ub, data, (uint32_t)length);
        read_message(session, &ub);
        printf("\r\nsession->info.last_requested_status %d \r\n", session->info.last_requested_status);
    }

    printf("\n\r listen_message() ended \n\r ", rv);
    return must_be_read;
}

recv_message() call recv_can_mesg() which is a CAN-transport implementation.

Having received all the frames and put them together as a complete message, it set the must_be_read to true, and execute the following code
    if(must_be_read)
    {
        ucdrBuffer ub;
        ucdr_init_buffer(&ub, data, (uint32_t)length);
        read_message(session, &ub);
        printf("\r\nsession->info.last_requested_status %d \r\n", session->info.last_requested_status);
    }

read_message(session, &ub); is where info.last_requested_status  is updated. However, it stays 0xFF for some reasons. I suspect the new CAN-Transport does not full some of those conditions. I don't know what conditions of those. Hence your advice are appreciated.

vibnwis, 2020-12-21:
Answering your question on "Do you have any micro XRCE-DDS Client code that replicates a real CAN buffer device, in order to establish a "fake" connection and being able to debug the code? Thank you."
Unfortunately, no. I did not expect this to happen. Hence, it never crossed my mind wanting to setup that.

jamoralp, 2020-12-21:
Well, you must be testing your code somewhere. Which board are you using? Where is the application responsible of sending the requests to the agent? That is the code I want to get access to, in order to better understand it.

vibnwis, 2020-12-21:
On Olimex STM32-F407 board with a PC. 

uClient sending request and listening to the response
   int poll_ms = UXR_CONFIG_MIN_SESSION_CONNECTION_INTERVAL;
for(size_t i = 0; i < attempts && session->info.last_requested_status == UXR_STATUS_NONE; ++i)
{
    send_message(session, buffer, length);
    poll_ms = listen_message(session, poll_ms) ? UXR_CONFIG_MIN_SESSION_CONNECTION_INTERVAL : poll_ms * 2;
}

return session->info.last_requested_status != UXR_STATUS_NONE;

vibnwis, 2020-12-21:
It is in session.c, and attached is interface of the CAN driver, if this helps.
[udp_transport_external_c.txt](https://github.com/eProsima/Micro-XRCE-DDS-Agent/files/5724182/udp_transport_external_c.txt)

CAN implementation

vibnwis, 2021-01-10:
Hi Pablo and Jose,
Happy new year. I think, the CAN-Bus transport implementation for uClient and uAgent is very close to the end. I think, I have missed out some conditions which cause the session not able to establish. Appreciate your comments, please?

vibnwis, 2021-01-14:
Hi Pablo and Jose,
I am thinking of how to lighten your support. I was wondering if possible sharing the requirements for creating a session i.e., the communication protocol. With that, I could debug and correct it myself at my end.

pablogs9, 2021-01-14:
Hello @vibnwis, you can find details about the XRCE-DDS standard here: https://www.omg.org/spec/DDS-XRCE/1.0/PDF

 You can find documentation about our implementation here: https://micro-xrce-dds.docs.eprosima.com/en/latest/

vibnwis, 2021-01-15:
Hi Pablo,
Many thanks for the document. I am reading the document now.
I was wondering if this is a mistake, in 8.2.1 Message, the sentence "the first byte has a 16-byte alignment", otherwise, what does it that mean?

pablogs9, 2021-01-15:
It just that the first byte of an XRCE message has a [16 B alignment](https://en.wikipedia.org/wiki/Data_structure_alignment). Where do you see the mistake?

vibnwis, 2021-01-15:
I don't understand "the first byte",  "has a",  "16-Byte alignment". It
does not make sense.

vibnwis, 2021-01-15:
Does it mean the first segment of an XRCE message has a 16-Byte alignment?

vibnwis, 2021-01-16:
Hi Pablo and Jose,

XRCE protocols say uAgent responses to submessageID CREATE_CLIENT(0x0) with submessageID STATUS_AGENT (0x4).

However, from the messages captured, showing below, it shows that the uAGent received  CREATE_CLIENT(0x0) submessages, but responded to it with also a CREATE_CLIENT submessage.

Any comments, please?

------ uAgent receiving a message, made out of 4 frames due to limitation of CAN-Bus ------
< CAN2 concatenated frames 

[0]=0x80 [1]=0x0 [2]=0x0 [3]=0x0 [4]=0x0 [5]=0x1 [6]=0x10 [7]=0x0 

[8]=0x58 [9]=0x52 [10]=0x43 [11]=0x45 [12]=0x1 [13]=0x0 [14]=0x1 [15]=0xf 

[16]=0xba [17]=0x5e [18]=0xba [19]=0x11 [20]=0x81 [21]=0x0 [22]=0xfc [23]=0xf 

CAN2 concatenated frames > 

set_data()- len 24
set_data()- ended
len 24 byte, id: 0x123 
data[0]: 0x80 data[1]: 0x0 data[2]: 0x0 data[3]: 0x0 data[4]: 0x0 data[5]: 0x1 data[6]: 0x10 data[7]: 0x0 
data[8]: 0x58 data[9]: 0x52 data[10]: 0x43 data[11]: 0x45 data[12]: 0x1 data[13]: 0x0 data[14]: 0x1 data[15]: 0xf 
data[16]: 0xba data[17]: 0x5e data[18]: 0xba data[19]: 0x11 data[20]: 0x81 data[21]: 0x0 data[22]: 0xfc data[23]: 0xf 

write frame_id 0x123 to source of input_packet 
write message_len 24 to source of input_packet 
CAN2Agent::recv_message() exited 
CAN2Agent::recv_message() entered 
SocketCAN:: handler 3 
SocketCAN::read  getsockopt() succeeded len = 4
SocketCAN:: Read() - A Frame memory space is ready and about to read from CAN interface can0
[35m[1608462735.786223][m info     | [34mSessionManager.hpp[m | [37mestablish_session       [m | [32msession re-established[m | client_key: 0x3126770193, address: 291

------ uAgent sending out a message, then breaking into 4 frames due to limitation of CAN-Bus ------
CAN2Agent::send_message() entered 
CAN2Agent::send_message() copied id 291CAN2Agent::send_message() copied len 24SocketCAN: write() frame.can_dlc= 8, frame.can_id = 291
data[0x3] data[0x80] data[0x0] data[0x0] data[0x0] **data[0x0]** data[0x1] data[0x10] Frame 3 was written to the CANSocket.
SocketCAN: write() frame.can_dlc= 8, frame.can_id = 291
data[0x2] data[0x0] data[0x58] data[0x52] data[0x43] data[0x45] data[0x1] data[0x0] Frame 2 was written to the CANSocket.
SocketCAN: write() frame.can_dlc= 8, frame.can_id = 291
data[0x1] data[0x1] data[0xf] data[0xba] data[0x5e] data[0xba] data[0x11] data[0x81] Frame 1 was written to the CANSocket.
SocketCAN: write() frame.can_dlc= 4, frame.can_id = 291
data[0x0] data[0x0] data[0xfc] data[0xf] Frame 0 was written to the CANSocket.
CAN2Agent::send_message() exited

vibnwis, 2021-01-17:
I revisited the CAN2Agent::send_message() and the bool CAN2Agent::recv_message(), making sure they do not use a common message buffers. I can confirm that they don't use a common message buffer.
Knowing that you guys are busy. As I am lack of the upper layer protocols implementation knowledge, however, I am happy to help.   Can you please point out to the possible causes of the issues, I will then try it at this end.  I look forward to hearing from you soon. Thank you

vibnwis, 2021-01-23:
Hi Pablo, Jose and team,
 I have been tracing the code line by line. I hope you guys can give me some clues. This project is a Master Degree project, which I initiated since August 2020, it is through Massey University Auckland, New Zealand. Should you require further confirmation, I am happy to introduce my supervisors. The reasons are I do not step on anyone/organization toes. I will fully disclose the code as required by the open-source requirements. 

The following are the traces, output messages when I run the uAgent and the Client. In simplicity, the uclient sent out 0x80 session_id and 0x00 submesssage_id, in return, it received 0x80 session_id, and 0x00 submessage_id as well. For me to understand it further, I created the traces as well as fake the messages of session_id 0x81 and 0x80 together with submessage_id 0x0 and 0x4. Below are the code and the condition. Nevertheless, they still failed in terms of creating a session. Hence, I hope you guys can provide some insight or advice. Also, If it is not causing any problems, to make the work simpler for you guys, can I request complete working messages for creating a session between uClient and uAgent. With that at least, I can compare and make necessary changes to code.

------ A message response sending from uAgent ----------
CAN2Agent::send_message() entered 
CAN2Agent::send_message() copied id 291CAN2Agent::send_message() copied len 24SocketCAN: write() frame.can_dlc= 8, frame.can_id = 291
data[0x3] data[0x80] data[0x0] data[0x0] data[0x0] data[0x0] data[0x1] data[0x10] Frame 3 was written to the CANSocket.
SocketCAN: write() frame.can_dlc= 8, frame.can_id = 291
data[0x2] data[0x0] data[0x58] data[0x52] data[0x43] data[0x45] data[0x1] data[0x0] Frame 2 was written to the CANSocket.
SocketCAN: write() frame.can_dlc= 8, frame.can_id = 291
data[0x1] data[0x1] data[0xf] data[0xba] data[0x5e] data[0xba] data[0x11] data[0x81] Frame 1 was written to the CANSocket.
SocketCAN: write() frame.can_dlc= 4, frame.can_id = 291
data[0x0] data[0x0] data[0xfc] data[0xf] Frame 0 was written to the CANSocket.
CAN2Agent::send_message() exited 

-------------------------------------------
At uClient, session creation still failed
-------------------------------------------

It begins by initialization through function, void uxr_init_session(uxrSession* session, uxrCommunication* comm, uint32_t key)
{
--
    //uxr_init_session_info(&session->info, 0x81, key);
;;
}
Which set the &session->info = 0x81, key = BA5EBA!!

When session_id received is 0x80, in function bool uxr_read_session_header(const uxrSessionInfo* info, ucdrBuffer* ub, uint8_t* stream_id_raw, uxrSeqNum* seq_num) at line must_be_read = session_id == info->id;, due to the session_id is 0x80, expected value info->id = 0x81, it returned a must_be_read=false value which stop reading the submessage contents. Subsequently the session->info.last_requested_status remains unchanged at 255. This causes function wait_session_status(uxrSession* session, uint8_t* buffer, size_t length, size_t attempts) to retry till the last attempt by the followign for-loop for(size_t i = 0; i < attempts && session->info.last_requested_status == UXR_STATUS_NONE; ++i). However, for a session to be created, it must receive submessage_id=0x04. As such I fake a message with submessage_id=0x04, and the session_id=0x81. However, from the traces, it returned session->info.last_requested_status = 88 which still does not meet UXR_STATUS_OK. See the code below.

bool created = received && UXR_STATUS_OK == session->info.last_requested_status;

vibnwis, 2021-02-03:
Resolved.  Should use output_packet.message instead for sending messages.

closed 2021-02-03 by vibnwis