eProsima/Micro-XRCE-DDS-Agent #206: Publisher (or Topic) is not Registered to DDS/ROS2 Network

platinum78, 2021-01-21:
I'm working on porting _ros2arduino_ implementation from ROBOTIS (https://github.com/ROBOTIS-GIT/ros2arduino), which is originally intended for use on Arduino, onto XMC4800 platform from Infineon.

Somehow I made <participant> to be registered to the agent, and this shows up in the ROS2 network using `ros2 node list` command. However, <topic>s do not attach to the ROS2 network, and using `ros2 topic list` does not show any topics related to XRCE DDS, and I'm stuck there.

It seems that stream reciprocates twice during topic/publisher registration, first register the topic and then create the publisher. Registering topic runs fine, and then some error occurs while creating the publisher.

In the first screenshot, after the long transmission (which is the XML configuration for entity creation), response goes fine; the trailing two bytes, which corresponds to `status` and `implementation_status` each, are both null bytes. However, in the second screenshot, the `status` byte becomes 0x80. It indicates something is wrong in the agent-side, and simultaneously causes the client-side to be incompletely initialized.

[image]

[image]

I tried virtually every possible method in my knowledge, but I couldn't resolve this problem. I would be appreciated if anyone would cooperate on this issue and help me get out from the stuck.

For your convenience, I wrote my environment and copy-pasted the terminal outputs below.
* OS: Ubuntu 18.04 (WSL on Windows 10)
* Communication Method: Serial @ 38,400 baud (tried 9600, 19200, 38400, 57600, 115200)
* MTU Setting: 2048
* Embedded Environment: XMC4800-2048 (Infineon)

[Terminal Outputs]

[1611220055.796359] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1611220055.797941] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6
[1611220721.797807] info     | Root.cpp           | create_client            | create                 | client_key: 0xAABBCCDD, session_id: 0x81
[1611220721.798654] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x2864434397, address: 0
[1611220721.799896] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0xAABBCCDD, len: 19, data:
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1611220731.412895] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0xAABBCCDD, len: 96, data:
0000: 81 80 00 00 01 05 56 00 00 0A 00 11 01 02 00 00 48 00 00 00 3C 64 64 73 3E 3C 70 61 72 74 69 63
0020: 69 70 61 6E 74 3E 3C 72 74 70 73 3E 3C 6E 61 6D 65 3E 73 63 5F 6E 6F 64 65 3C 2F 6E 61 6D 65 3E
0040: 3C 2F 72 74 70 73 3E 3C 2F 70 61 72 74 69 63 69 70 61 6E 74 3E 3C 2F 64 64 73 3E 00 00 00 00 00
[1611220731.466333] debug    | ProxyClient.cpp    | create_participant       | participant created    | client_key: 0xAABBCCDD, participant_id: 0x001(1)
[1611220731.467316] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0xAABBCCDD, len: 14, data:
0000: 81 80 00 00 05 01 06 00 00 0A 00 11 00 00
[1611220731.468552] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0xAABBCCDD, len: 13, data:
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1611220731.476855] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0xAABBCCDD, len: 13, data:
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1611220731.477674] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0xAABBCCDD, len: 14, data:
0000: 81 80 00 00 05 01 06 00 00 0A 00 11 00 00
[1611220731.477922] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0xAABBCCDD, len: 13, data:
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1611220731.479035] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0xAABBCCDD, len: 13, data:
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1611220731.493329] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0xAABBCCDD, len: 13, data:
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1611220731.494255] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0xAABBCCDD, len: 13, data:
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1611220731.495997] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0xAABBCCDD, len: 13, data:
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1611220733.732759] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0xAABBCCDD, len: 13, data:
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1611220733.748690] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0xAABBCCDD, len: 24, data:
0000: 81 80 01 00 01 05 0F 00 00 0B 00 13 03 02 00 00 01 00 00 00 00 00 11 00
[1611220733.755965] debug    | ProxyClient.cpp    | create_publisher         | publisher created      | client_key: 0xAABBCCDD, publisher_id: 0x001(3), participant_id: 0x001(1)
[1611220733.757427] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0xAABBCCDD, len: 14, data:
0000: 81 80 01 00 05 01 06 00 00 0B 00 13 00 00
[1611220733.759561] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0xAABBCCDD, len: 13, data:
0000: 81 00 00 00 0A 01 05 00 02 00 00 00 80
[1611220733.764739] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0xAABBCCDD, len: 14, data:
0000: 81 80 01 00 05 01 06 00 00 0B 00 13 00 00
[1611220733.765600] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0xAABBCCDD, len: 13, data:
0000: 81 00 00 00 0A 01 05 00 02 00 00 00 80
[1611220733.781040] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0xAABBCCDD, len: 13, data:
0000: 81 00 00 00 0A 01 05 00 02 00 00 00 80
[1611220733.783516] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0xAABBCCDD, len: 13, data:
0000: 81 00 00 00 0A 01 05 00 02 00 00 00 80
[1611220734.948752] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0xAABBCCDD, len: 13, data:
0000: 81 00 00 00 0A 01 05 00 02 00 00 00 80
[... 48 lines cut ...]
0000: 81 80 04 00 01 05 A1 00 00 0E 00 15 05 02 00 00 93 00 00 00 3C 64 64 73 3E 3C 64 61 74 61 5F 77
0020: 72 69 74 65 72 3E 3C 74 6F 70 69 63 3E 3C 6B 69 6E 64 3E 4E 4F 5F 4B 45 59 3C 2F 6B 69 6E 64 3E
0040: 3C 6E 61 6D 65 3E 72 74 2F 73 63 5F 69 6E 74 65 67 65 72 3C 2F 6E 61 6D 65 3E 3C 64 61 74 61 54
0060: 79 70 65 3E 73 74 64 5F 6D 73 67 73 3A 3A 6D 73 67 3A 3A 64 64 73 5F 3A 3A 49 6E 74 33 32 5F 3C
0080: 2F 64 61 74 61 54 79 70 65 3E 3C 2F 74 6F 70 69 63 3E 3C 2F 64 61 74 61 5F 77 72 69 74 65 72 3E
00A0: 3C 2F 64 64 73 3E 00 00 13 00 00 00
[1611220890.020470] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0xAABBCCDD, len: 13, data:
0000: 81 00 00 00 0B 01 05 00 03 00 04 00 80
[*** LOG ERROR #0001 ***] [2021-01-21 18:21:30] [] {argument index out of range}
[1611220890.025020] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0xAABBCCDD, len: 14, data:
0000: 81 80 04 00 05 01 06 00 00 0E 00 15 80 00
[1611220890.025506] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0xAABBCCDD, len: 13, data:
0000: 81 00 00 00 0A 01 05 00 05 00 00 00 80
[1611220890.026051] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0xAABBCCDD, len: 13, data:
0000: 81 00 00 00 0A 01 05 00 05 00 00 00 80
[1611220890.035257] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0xAABBCCDD, len: 14, data:
0000: 81 80 04 00 05 01 06 00 00 0E 00 15 80 00
[1611220890.036325] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0xAABBCCDD, len: 13, data:
0000: 81 00 00 00 0A 01 05 00 05 00 00 00 80
[1611220890.051163] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0xAABBCCDD, len: 13, data:
0000: 81 00 00 00 0A 01 05 00 05 00 00 00 80
[1611220890.052997] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0xAABBCCDD, len: 13, data:
0000: 81 00 00 00 0A 01 05 00 05 00 00 00 80
[1611220890.054660] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0xAABBCCDD, len: 13, data:
0000: 81 00 00 00 0A 01 05 00 05 00 00 00 80

jamoralp, 2021-01-21:
Good morning. Regarding the `ros2arduino` library, I'm afraid we are far from being in a good position to attend this issue. Sorry. Have you tried to contact ROBOTIS in the source code repo? We are unaware of which `Micro XRCE-DDS Client` library version they are using for the implementation, but that certainly could be a problem. Currently we are working on unifying the associated versions of the Micro XRCE-DDS Client and Agent libraries, to ensure full compatibility among them. In the meantime, I suggest you try using the latest available image of the agent, which is something very easy to achieve by means of its dedicated [snap package](https://snapcraft.io/micro-xrce-dds-agent).

As part of the micro-ROS team, what I can do is encourage you to keep an eye on the [Micro-ROS Arduino](https://github.com/micro-ROS/micro_ros_arduino) tool, which provides micro-ROS as a precompiled library that can be perfectly integrated within the Arduino IDE environment. We are pretty sure that the HW platform you are using can be perfectly integrated within this tool, and it would be great to make a conjoint effort to support it. What do you think?

platinum78, 2021-01-22:
Sounds sad 😢, but I understand eProsima is not in charge of _ros2arduino_. Thanks for your recommendation about `micro-ROS` package; I would take a deep look if I got a chance to work with Arduino environment.
Besides, I wish I could get some hints about two questions below, which is actually Micro-XRCE DDS agent/client stuff.

* In which case does the agent transmit 0x80 for `status` byte? I found from the Micro-XRCE documentation that 0x80 is something like a trailing byte for already-established streams, but there were some comments in closed issues that 0x80 is an error code.
* What does `[*** LOG ERROR #0001 ***] [2021-01-21 18:21:30] [] {argument index out of range}` error mean? This error occurs right after receiving the XML snippet (from the client via serial) describing the data writer, and is followed by `status == 0x80`. Perhaps the XML transmission is malfunctioning? May the LOG ERROR affect the `status` byte become `0x80`?

XML data to be sent is `<dds><topic><name>Int32</name><dataType>std_msgs::msg::dds_::Int32_</dataType></topic></dds>`, and the corresponding byte array is shown below; is there any violation to internal transmission protocol?
`
0000: 81 80 04 00 01 05 A1 00 00 0E 00 15 05 02 00 00 93 00 00 00 3C 64 64 73 3E 3C 64 61 74 61 5F 77
0020: 72 69 74 65 72 3E 3C 74 6F 70 69 63 3E 3C 6B 69 6E 64 3E 4E 4F 5F 4B 45 59 3C 2F 6B 69 6E 64 3E
0040: 3C 6E 61 6D 65 3E 72 74 2F 73 63 5F 69 6E 74 65 67 65 72 3C 2F 6E 61 6D 65 3E 3C 64 61 74 61 54
0060: 79 70 65 3E 73 74 64 5F 6D 73 67 73 3A 3A 6D 73 67 3A 3A 64 64 73 5F 3A 3A 49 6E 74 33 32 5F 3C
0080: 2F 64 61 74 61 54 79 70 65 3E 3C 2F 74 6F 70 69 63 3E 3C 2F 64 61 74 61 5F 77 72 69 74 65 72 3E
00A0: 3C 2F 64 64 73 3E 00 00 13 00 00 00
`

platinum78, 2021-01-22:
I found the solution. It happened from using the wrong version of Micro-XRCE DDS Agent. I used v1.5.0, while the _ros2arduino_ repo says to use v1.3.0. Actually, I did try to use 1.3.0 at first, but then encountered a termios-related error, which misled me to use 1.5.0. Anyway, I think the latest version of Agent does have problem in terms of backward compatibility, so those who are facing similar issues with my case should try to match the corresponding version of agent and client. Matching versions are shown in https://github.com/eProsima/Micro-XRCE-DDS .

Those encountering issues from `termios` using v1.3.0 should fix the source code and build again.
* Find and open `TermiosAgentLinux.cpp` (from Micro-XRCE DDS Agent project)
* Change `new_attrs.c_ispeed = termios_attrs_.c_ispeed` to `cfsetispeed(&new_attrs, termios_attrs_.c_ispeed)`
* Change `new_attrs.c_ospeed = termios_attrs_.c_ospeed` to `cfsetospeed(&new_attrs, termios_attrs_.c_ospeed)`
* Build the project again. Things worked fine, at least for my case.

closed 2021-01-22 by platinum78