eProsima/Micro-XRCE-DDS-Agent #212: CAN-bus Transport failed on bool uxr_run_session_until_all_status()

vibnwis, 2021-02-02:
Hi,
I am new to XRCE-Protocol. Hence, I am not able to pick up many issues of the existing XRCE-messaging protocols as quickly as I wanted. I have got the CAN-Bus transport implemented. Both able to send and receive XRCE-messages. It has created a session successfully and succeeding in running line up to uxr_run_session_until_all_status() where it failed.

My questions are:
1. Since the CAN transports is already working. As far as being a transport layer protocol what are the conditions and requirements needed by the micro-ROS? Due to CAN-bus limiting up to 8 bytes, it will need to breaking down into 8-byte frames and reconstructing a complete XRCE-message from frames. As a result, it will take a longer time for long messages. Will this cause an issue? what timing requirements are if they do exist? and any other conditions or requirements?
2. As I see from the communication between uClient and uAgent.  uAgent sendis out Heartbeat messages regularly. Since these messages may fill up the buffer, and when recv_msg() is called, usually after a sending message, at that point, recv_msg() may process(or read) messages in the buffer which can be old HeartBeat messages. I presume the upper layer protocols have no issues with that scenario. However, if it does, what are the expectation or resolutions?

Feel free to comment. Your response will be very appreciated.

pablogs9, 2021-02-03:
1. XRCE messaging should not desist communicating despite any timeout. Usually, the agent or client stops sending heartbeats if no answer found but once the other entity starts communication again the status of the XRCE connection is the same. This protocol is designed to support architectures where the client goes to deep sleep modes for a long time, so there are no timing requirements between complete XRCE messages.
2. You can configure the heartbeat period [here](https://github.com/eProsima/Micro-XRCE-DDS-Agent/blob/35fa39a83e447af0ea5d40469f0105544e3e6529/CMakeLists.txt#L54) and [here](https://github.com/eProsima/Micro-XRCE-DDS-Client/blob/3286ebe78030ae5c160acb3abc3455e5ac6ee150/CMakeLists.txt#L55). Tune it to not fill your buffer.

By the way, in this evening we are going to talk about CAN transports in the ROS Embedded Working Group. Check the agenda [here](https://docs.google.com/document/d/1hGxwpuHDPCyEYHBCd5IIgvoFGJgkC5WKI12ghEfqwXg).

vibnwis, 2021-02-03:
Many thanks for the explanations. 
As for the ROS Embedded Working Group, seeing the CAN agenda is about FDCAN or CAN2.0, I would like to suggest that FDCAN which takes payload up to 64 bytes whereas CAN2.0 can take only up to 8 bytes. The overhead of communication is too much. I may implement FDCAN through SPI bus on the Olimex board provided my supervisor agree. Hope this helps.

pablogs9, 2021-02-03:
I'm not sure about which CAN option is the group targeting. Regarding the implementation of the custom transport, we are going to present soon a new API that will let the user write runtime transports instead of dealing with integrating them in the middleware.

vibnwis, 2021-02-03:
Understandable.

vibnwis, 2021-02-04:
As this involves editing CMakeLists.txt which I think I am a bit hazy. Do you need to rebuild them using the commands below?

[1]  <--- Not so sure if this is valid
cd Micro-XRCE-DDS-Client   
mkdir build && cd build
cmake .. -DUCLIENT_BUILD_EXAMPLES=ON  
make

and under FreeRTOS folder, type commands
[2]
make libmicroxrcedds
make

and the same applies to uAgent.
[3]
rm -rf build/
mkdir build
cd build/
cmake .. -DCMAKE_BUILD_TYPE=DEBUG -DUAGENT_DISCOVERY_PROFILE=OFF -DUAGENT_P2P_PROFILE=OFF -DUAGENT_CLI_PROFILE=OFF
make

vibnwis, 2021-02-08:
Hi,
I have captured the status of the request_list. See below.
status_list[0]=0, request_list[0]=10, status_confirmed = 1 
 status_list[1]=0, request_list[1]=11, status_confirmed = 1 
 status_list[2]=0, request_list[2]=12, status_confirmed = 1 
 status_list[3]=0, request_list[3]=13, status_confirmed = 1 
 status_list[4]=128, request_list[4]=14, status_confirmed = 1 
 status_list[5]=128, request_list[5]=15, status_confirmed = 1 
However, it still failed eventually.
From the description of the bool uxr_run_session_until_one_status(), it says "If there is any reliable stream, it will perform the associated reliable behaviour to ensure communication."  I was wondering what conditions constitute to the reliable behaviour. Does timeliness to the requests be one of them? 
I am trying to figure out what goes wrong with my CAN transport implementation that fails to meet the conditions. Of course, due to the CAN 2.0 8-bytes payload, the flushed Create requests, due to framing, take as long as 30 seconds, and the following flushed Create requests also take 15 seconds. Any thoughts? Appreciate your comments please.

vibnwis, 2021-02-08:
FYI, I have tuned up the Heartbeat values. It does help to avoid the buffer from being crushed. However, I think it does not help on meeting the reliable communication requirements.

vibnwis, 2021-02-10:
After many attempts, yet it still fails. I have decided to stop working on the issue and moving on.

closed 2021-02-10 by vibnwis

pablogs9, 2021-02-10:
have you checked the new custom transport approach? https://github.com/eProsima/Micro-XRCE-DDS-Client/pull/179

vibnwis, 2021-02-10:
No.Thanks for letting me know. However, due to the pressing of time, I can't afford to spend more time, as I am required to develop some an application for proof of concepts - the studies requirements. Instead of CAN, we decided to use Ethernet.