eProsima/Micro-XRCE-DDS-Agent #216: can't connect with Client when Agent is launched in a docker container

chaos44, 2021-02-07:
[image]
Although the Client and the Agent is in the same network, They can't connect with each other when using a docker container.
I can ping from container to host IP or gateway IP but I can't ping container's IP from my host machine (window 10).
docker official's site also say it https://docs.docker.com/docker-for-windows/networking/#i-cannot-ping-my-containers

I want to know is there any way to connect Micro-XRCE-DDS-Agent with Micro-XRCE-DDS-Client even though Micro-XRCE-DDS-Agent is installed in Ubuntu which inside a docker container created by docker desktop.

jamoralp, 2021-02-08:
Good morning. Which flags are you using to launch the Docker container? Have you tried using the `--net=host` option, or is there some reason preventing you from doing that?

chaos44, 2021-02-09:
I tried to use `--net=host` but still can't to my micro computer.
[image]
[image]

jamoralp, 2021-02-09:
This should be working, but I have not faced the situation of running the Micro XRCE-DDS Agent within a Docker that is running within a virtualized Ubuntu environment. I remember having issues with virtualization systems in the past though, not being able to access to the virtualized environment from my host system. This could be related with the network settings of the virtualization tool that you are using (I'm assuming it is virtualBox). Have you checked your network settings? Could you paste them here?

pablogs9, 2021-02-09:
Have you tried to use a configured IP and port in the client? I'm not sure about the multicast networking (used by Agent discovery) inside a virtual machine.

chaos44, 2021-02-09:
To jamoralp:
I use docker desktop and the host、the client use the same wireless network.

ubuntu's (in docker container) network settings :
[image]

host's (windows 10) network settings :
[image]

To pablogs9 :
the client's IP is allocated by micro computer's script and I don't need to do anything about the client. (I think so)

pablogs9, 2021-12-17:
Closing due to inactivity. Reopen if needed.

closed 2021-12-17 by pablogs9