eProsima/Micro-XRCE-DDS-Agent #237: What is the GuidPrefix_t of XRCE-DDS Agent's discovery server?

liyue75, 2021-04-01:
Hi, I want to use XRCE-DDS Agent 's discovery server:
`MicroXRCEAgent udp6 -p 2021 -d 7400`
but  on client side setup with xml, I don't know how to fill the value of `RemoteServer prefix`
`<RemoteServer prefix=" ">`

pablogs9, 2021-04-05:
Hello, @liyue75 in which platform are you using the Micro XRCE-DDS Client?

In general, when you enable the Agent's discovery server, the client just uses broadcast to discover Agent's interfaces, so no configuration is required. Can you share some more details about your use case so we can help you?

liyue75, 2021-04-05:
@pablogs9 , I'm using  XRCE-DDS to get sensor state, those sensors are running on stm32.
The data is sent from STM32 boards to the agent on the computer. On the computer, I need to run a FastDDS  program to get the data from the agent and send they to influxdb.  
On the Client side, I use default transport, `<useBuiltinTransports>true<useBuiltinTransports>` when create participant from xml.
On computer, I use  FASTDDSv2.3.0 and write a fastdds sub program to get the data. 

In my SUB program, If I use default transport, `<useBuiltinTransports>true<useBuiltinTransports>`, I find the sub can not recv data. The DDS on the same computer should use SHM to transfer data, but my sub program can not get data from Agent,  I think it may be caused by the lower DDS version of the Agent.
So I use udp transport,  
```
<useBuiltinTransports>false<useBuiltinTransports>
<userTransports><transport_id>udp</transport_id></userTransports>
```
this time, my sub program can get data from the Agent.

When I using udp transport, I find there are lots of multicast packets on the computer, I know they are sent by the discovery process, every interface on the computer sends multicast packets for every entity created by Client. 
I dont want these multicast packets on the network, so I want to use a discovery server, each entity sends they discovery message to the discovery server.
I find the Agent can run as a discovery server, but I don't know its guid.  I need the id in my fastdds sub program when I create the sub participant with XML file.

pablogs9, 2021-04-05:
Hello, @liyue75 as explained in the [XRCE-DDS Documentation](https://micro-xrce-dds.docs.eprosima.com/en/latest/client.html?highlight=discovery#discovery-profile) the discovery profile is used by the Client to discover the Agent IP and start the XRCE communication. So, it is not related to DDS or RTPS.

liyue75, 2021-04-05:
@pablogs9, thanks for your repley.

closed 2021-04-05 by liyue75