eProsima/Micro-XRCE-DDS-Agent #240: Delay of data transfer between Micro-XRCE-DDS-Agent and Fast-DDS

liyue75, 2021-04-08:
## Issue template
I'm running an IoT system that monitors sensors using Micro-XRCE-DDS, Fast-DDS, InfluxDB and Granafa.

The diagram below shows the data flow:
[image]

Sensor send data to DDS-Agent, and I run a Fast-DDS subscriber to receive data from the Agent. In my Fast-DDS sub 
program, I caculate the interval between the  data and send it to Influxdb, finally I use Grafana to display  the interval.
An example of Grafana's output is shown below:
[image]

Sensor data is transmitted at an interal of about 17ms. However, Fast-DDS subscriber occasionally receives data at 
interval greater than 20ms.
Is this problem related to Micro-XRCE-DDS-Agent or network protocol stack? How to solve it?

- Hardware description: ThinkPad-T470
- OS: Ubuntu18.04, XRCE-DDS Agent is running in docker
- Version or commit hash: XRCE-DDS-Agent:v2.0.0, Fast-DDS:v2.3.0

#### Additional information

pablogs9, 2021-04-08:
Hello @liyue75:
- Regarding shm: Ubuntu 18.04 is not a Tier 1 platform for Fast 2.3.0. Check [here](https://github.com/eProsima/Fast-DDS/blob/master/PLATFORM_SUPPORT.md)
- Regarding `<address>169.254.107.203</address>`, forcing a network interface may introduce jitter, can you set this parameter to `<address>localhost</address>`?
- I'm not sure about which part is introducing this jitter: the XRCE part or the DDS part. Could you "mock" the XRCE part by creating a fake DDS publisher with your desired rate and make the same measurement. My colleague at FastDDS @EduPonz told me that jitter of 3ms is not acceptable in our DDS implementation. So if we discard the DDS part we can focus on the embedded subsystem and check if everything is working as expected in XRCE.

Information required about the sensor:
- Are you using an RTOS?
- Which transport is using XRCE? Serial, UDP?
- How are you creating the XRCE entities? Could you share your sensor code or at least provide details?

liyue75, 2021-04-12:
@pablogs9 Thank you very much for your help.
As for your question:
- I'm using RIOT-OS on stm32. I have ported micro-XRCE-DDS-Client to RIOT.
- The transport is UDPv6

After I reading your reply, I tested the jitter between  2 DDS nodes on computer and it were indeed less than 3ms.
So you are right, the problem comes from the embed side.
And then I found the cause of the problem, in publish thread in my embed system, I used an API provided by RIOT 
called `xtimer_usleep()` to switch threads. I thought this API would wake up the pub thread on time, but I was wrong.
So there's no problem with DDS, it's my program.
I'm going to fix my program now. Thanks.

closed 2021-04-12 by liyue75

pablogs9, 2021-04-12:
Thanks a lot for the feedback @liyue75, have you consider contributing your RIOT-OS port to https://github.com/eProsima/Micro-XRCE-DDS-Apps?

liyue75, 2021-04-12:
@pablogs9 Of course, how can I do that? I can contribute my port when I'm free.

pablogs9, 2021-04-12:
Well we can start by seeing what changes have you done and writing some instructions to build it for RIOT-OS

liyue75, 2021-04-30:
Well, I am free in the next few days, I will write about my port here.

liyue75, 2021-05-19:
@pablogs9 Hi, I'm sorry for taking so long to update, because everytime I wanted to update before, I didn't konw how to explain it.
So I just created a repository and wrote an example: [https://github.com/liyue75/DDS-Client-in-Riot/](url).
For simple use, I have packaged microdds api.