eProsima/Micro-XRCE-DDS-Agent #242: Agent crash sometimes

liyue75, 2021-04-16:
## Issue template
I run 3 DDS-participant , a XRCE-DDS-Agent on nanopi and 3 XRCE-Clients on 3 stm32 boards. The nanopi and stm32 boards are connected by LAN. Client and Agent communicate to each other by UDPv6, Agent and DDS-participant communicate by UDPv4.
Data flow: 3 clients send data to these 3 DDS-participant, and a client would receive from a DDS-participant.
But after running for an indeterminate period of time yesterday and the day before, the Agent crashed, the Client and the DDS-participant was still running. 
The program may run for hours and it looks working fine, but it would quit accidentally.

- Hardware description: nanopi
- OS: ubuntu18.04
- Installation type: Agent:v2.0.0, client:v1.2.5, FastDDS:v2.3.0

I'm not sure what caused this issue. Do you hanve any suggestions for finding out why?

## More imformation
When start up the Agent, I use default verbose 4`./MicroXRCEAgent udp6 -p 2021 -v 4`, and when it quit, I can not see any information, would verbose 6 help? Or is there any way to display more detailed information about Agent exit?
On the Client side, I have disabled STM32 to receive broadcast and multicast packages, because these packets will cause program interrupt and affect the real-time performance(DDS nodes would send multicast packages). Will this affect the Cient and Agent?

pablogs9, 2021-04-16:
Some points:
- Please try to use `-v6` to check if something is happening on the agent side when it closes
- `client_key` of DDS participants is correct
- Client-agent interaction should not use broadcast and multicast when discovery is not used
- Why are you using client `v1.2.5` instead of `v2.0.0`?

liyue75, 2021-04-16:
@pablogs9 
Thanks, I will test with v6.
I disable multicast because the DDS-participants will always send multicast packages and stm32 can receive them.
I'm using client v1.2.5 because I had previously ported v1.2.5 to RIOT-OS and I haven't updated the client version yet.

pablogs9, 2021-04-16:
If you have v1.2.5 ported, version 2 should not be so difficult.

liyue75, 2021-04-16:
@pablogs9 
Ok, I will try to port version 2.0.0.
I find that sometimes the Agent would run an error when `send_message` In **UDPv6AgentLinux.cpp**, and then reboot in `bool UDPv6Agent::handle_error()`.
In **Server.cpp**, in `error_handler_loop`, the error_handle_thread will sleep_for 500ms after handle the error.
During the reboot and sleep time, other threads are still running, will this cause any error?

pablogs9, 2021-04-16:
Have you tested it in UDPv4? Is IPv6 a requirement for your use case?

> In Server.cpp, in error_handler_loop, the error_handle_thread will sleep_for 500ms after handle the error.
During the reboot and sleep time, other threads are still running, will this cause any error?

Let me check that

liyue75, 2021-04-16:
@pablogs9 
I use UDPv6 because RIOT's default network stack GNRC only support IPv6.

liyue75, 2021-04-17:
@pablogs9 
When I look at `/var/log/apport.log`, I find the relevant information:
```ERROR: apport (pid 3783) Wed Apr 14 03:27:15 2021: called for pid 1356, signal 11, core limit 0, dump mode 1
ERROR: apport (pid 3783) Wed Apr 14 03:27:15 2021: executable: /home/pi/ddsagent/MicroXRCEAgent (command line "./MicroXRCEAgent udp6 -p 2021")
ERROR: apport (pid 3783) Wed Apr 14 03:27:15 2021: executable does not belong to a package, ignoring
ERROR: apport (pid 28283) Wed Apr 14 09:54:58 2021: called for pid 4141, signal 11, core limit 0, dump mode 1
ERROR: apport (pid 28283) Wed Apr 14 09:54:58 2021: executable: /home/pi/ddsagent/MicroXRCEAgent (command line "./MicroXRCEAgent udp6 -p 2021")
ERROR: apport (pid 28283) Wed Apr 14 09:54:58 2021: executable does not belong to a package, ignoring
ERROR: apport (pid 5847) Thu Apr 15 11:00:04 2021: called for pid 1369, signal 11, core limit 0, dump mode 1
ERROR: apport (pid 5847) Thu Apr 15 11:00:04 2021: executable: /home/pi/ddsagent/MicroXRCEAgent (command line "./MicroXRCEAgent udp6 -p 2021")
ERROR: apport (pid 5847) Thu Apr 15 11:00:04 2021: executable does not belong to a package, ignoring
```
But I didn't have CoreDump enabled before, so I don't have a core file to debug.
These errors occur on Wed and Thu. On Fri, I charged devices of my system, restart, and these errors have not happened again until today(Sat).
So I wonder if the reason for the error is that:When the battery is low, there are some errors on the network of my system, which leads to `handle_error()` in `error_handle_loop` in **Server.cpp**. But other threads are still running and may require some resources that have already been released in `UDPv6Agent::fini()` in `handle_error()`, like `poll_fd`.

pablogs9, 2021-04-19:
Is possible for you to test the same setup but running the agent on a general-purpose PC with Linux? I would like to check if this is related to the nanopi or with the UPDv6 transport.

liyue75, 2021-04-19:
Ok, I will test the agent on my computer.

liyue75, 2021-04-19:
On my computer, I have two docker images:
`ddsagent:v2.0-arm64`, `ddsagent:v2.0-x86`
NanoPi's architecture is arm64, should I run the Agent in arm64 docker enviroment on my PC or run in x86 docker enviroment.

pablogs9, 2021-04-19:
Well, it would be interesting if you use a pure XRCE-DDS Agent in order to avoid having more points of error such as docker, so:

```bash
git clone https://github.com/eProsima/Micro-XRCE-DDS-Agent
cd Micro-XRCE-DDS-Agent
mkdir build; cd build;
cmake ..
make -j$(nproc)
```

You will have inside the build folder an executable that will launch your agent. Please try that.

liyue75, 2021-04-19:
@pablogs9 Alright.
But I didn't use docker on nanopi.

liyue75, 2021-04-30:
@pablogs9, Maybe the problem is probably caused by the mismatched version of client and agent.
I updated the client version in embedded side to 2.0 last week, and the problem has not appeared since then.

closed 2021-04-30 by liyue75