eProsima/Micro-XRCE-DDS-Agent #247: Agent Won't Work with Number of Topics More Than Specific Number

platinum78, 2021-05-14:
- Hardware description: Infineon XMC4800, 144MHz, 2MB storage
- OS: Ubuntu 18.04
- Installation type: ROS2 Dashing + MicroXRCE Agent 1.3.0
- Version or commit hash: v1.3.0

#### Steps to reproduce the issue
When there exist up to three data writers (ROS2 publisher in my case), things work properly.
When there exist four or more data writers, bad things happen!

#### Expected behavior
After registering each participant and data writer, the application (on embedded side) should be able to transceive data for each topic.

#### Actual behavior
After registering each participant and data writer, the application tries to transceive data, but no reply is received.

#### Additional information
Below is the footage of what happens.

**[Case with Three Publishers - Works Properly]**
`
=== Phase 1 - UDPv6AgentLinux.cpp
[1620969165.942595] info | UDPv6AgentLinux.cpp | init | running... | port: 8888

=== Phase 2 - Root.cpp
[1620969165.943522] info | Root.cpp | set_verbose_level | logger setup | verbose_level: 6

=== Phase 3 - UDPv6AgentLinux.cpp
[1620969198.196597] debug | UDPv6AgentLinux.cpp | recv_message | [==>> UDP <<==] | client_key: 0x00000000, len: 48, data: 

=== Phase 4 - Root.cpp
[1620969198.196807] info | Root.cpp | create_client | create | client_key: 0x12345678, session_id: 0x81

=== Phase 5 - SessionManager.hpp
[1620969198.196878] info | SessionManager.hpp | establish_session | session established | client_key: 0x305419896, address: fd00:0000:0000:0000:1111:1111:1111:0065:b822

=== Phase 6 - UDPv6AgentLinux.cpp
[1620969198.197036] debug | UDPv6AgentLinux.cpp | send_message | [** <<UDP>> **] | client_key: 0x12345678, len: 19, data: 
[1620969198.201894] debug | UDPv6AgentLinux.cpp | recv_message | [==>> UDP <<==] | client_key: 0x12345678, len: 24, data: 

=== Phase 7 - SessionManager.hpp
[1620969198.201955] info | SessionManager.hpp | establish_session | session re-established | client_key: 0x305419896, address: fd00:0000:0000:0000:1111:1111:1111:0065:b822

=== Phase 8 - UDPv6AgentLinux.cpp
[1620969198.202089] debug | UDPv6AgentLinux.cpp | send_message | [** <<UDP>> **] | client_key: 0x12345678, len: 19, data: 
[1620969198.207056] debug | UDPv6AgentLinux.cpp | recv_message | [==>> UDP <<==] | client_key: 0x12345678, len: 96, data: 

=== Phase 9 - ProxyClient.cpp
[1620969198.212361] debug | ProxyClient.cpp | create_participant | participant created | client_key: 0x12345678, participant_id: 0x001(1)

=== Phase 10 - UDPv6AgentLinux.cpp
[1620969198.212491] debug | UDPv6AgentLinux.cpp | send_message | [** <<UDP>> **] | client_key: 0x12345678, len: 14, data: 
[1620969198.212682] debug | UDPv6AgentLinux.cpp | send_message | [** <<UDP>> **] | client_key: 0x12345678, len: 13, data: 

=== Phase 11 - FastEntities.cpp
[1620969198.213839] trace | FastEntities.cpp | onParticipantDiscovery | matched | entity_id: 0.0.1.c1, guid_prefix: 1.f.88.b5.f5.76.0.0.1.0.0.0
[1620969198.213883] trace | FastEntities.cpp | onParticipantDiscovery | matched | entity_id: 0.0.1.c1, guid_prefix: 1.f.88.b5.72.76.0.0.1.0.0.0
[1620969198.215095] trace | FastEntities.cpp | onParticipantDiscovery | matched | entity_id: 0.0.1.c1, guid_prefix: 1.f.88.b5.f7.1.0.0.1.0.0.0
[1620969198.215385] trace | FastEntities.cpp | onParticipantDiscovery | matched | entity_id: 0.0.1.c1, guid_prefix: 1.f.88.b5.5e.76.0.0.1.0.0.0

=== Phase 12 - UDPv6AgentLinux.cpp
[1620969198.221895] debug | UDPv6AgentLinux.cpp | recv_message | [==>> UDP <<==] | client_key: 0x12345678, len: 37, data: 
[1620969198.256907] debug | UDPv6AgentLinux.cpp | recv_message | [==>> UDP <<==] | client_key: 0x12345678, len: 13, data: 
[1620969198.257183] debug | UDPv6AgentLinux.cpp | send_message | [** <<UDP>> **] | client_key: 0x12345678, len: 13, data: 
[1620969198.266849] debug | UDPv6AgentLinux.cpp | recv_message | [==>> UDP <<==] | client_key: 0x12345678, len: 24, data: 

=== Phase 13 - ProxyClient.cpp
[1620969198.266914] debug | ProxyClient.cpp | create_publisher | publisher created | client_key: 0x12345678, publisher_id: 0x001(3), participant_id: 0x001(1)

=== Phase 14 - UDPv6AgentLinux.cpp
[1620969198.267058] debug | UDPv6AgentLinux.cpp | send_message | [** <<UDP>> **] | client_key: 0x12345678, len: 14, data: 
[... 306 lines cut ...]
=== Phase 32 - UDPv6AgentLinux.cpp
[1620970428.179302] debug | UDPv6AgentLinux.cpp | send_message | [** <<UDP>> **] | client_key: 0x12345678, len: 14, data: 
[1620970428.179378] debug | UDPv6AgentLinux.cpp | send_message | [** <<UDP>> **] | client_key: 0x12345678, len: 13, data: 
[1620970428.193587] debug | UDPv6AgentLinux.cpp | recv_message | [==>> UDP <<==] | client_key: 0x12345678, len: 113, data: 
[1620970428.198686] debug | UDPv6AgentLinux.cpp | recv_message | [==>> UDP <<==] | client_key: 0x12345678, len: 324, data: 
[1620970428.198809] debug | UDPv6AgentLinux.cpp | send_message | [** <<UDP>> **] | client_key: 0x12345678, len: 13, data: 
[1620970428.204968] debug | UDPv6AgentLinux.cpp | recv_message | [==>> UDP <<==] | client_key: 0x12345678, len: 716, data: 
[1620970428.205211] debug | UDPv6AgentLinux.cpp | send_message | [** <<UDP>> **] | client_key: 0x12345678, len: 13, data: 
[1620970428.209889] debug | UDPv6AgentLinux.cpp | recv_message | [==>> UDP <<==] | client_key: 0x12345678, len: 124, data: 
[1620970428.210015] debug | UDPv6AgentLinux.cpp | send_message | [** <<UDP>> **] | client_key: 0x12345678, len: 13, data: 
[1620970428.350671] debug | UDPv6AgentLinux.cpp | recv_message | [==>> UDP <<==] | client_key: 0x12345678, len: 100, data: 
[1620970428.350893] debug | UDPv6AgentLinux.cpp | send_message | [** <<UDP>> **] | client_key: 0x12345678, len: 13, data: 
[1620970428.356774] debug | UDPv6AgentLinux.cpp | recv_message | [==>> UDP <<==] | client_key: 0x12345678, len: 324, data: 
[1620970428.357006] debug | UDPv6AgentLinux.cpp | send_message | [** <<UDP>> **] | client_key: 0x12345678, len: 13, data: 
[1620970428.361938] debug | UDPv6AgentLinux.cpp | recv_message | [==>> UDP <<==] | client_key: 0x12345678, len: 1452, data: 
[1620970428.361983] debug | UDPv6AgentLinux.cpp | recv_message | [==>> UDP <<==] | client_key: 0x12345678, len: 404, data: 
[1620970428.362082] debug | UDPv6AgentLinux.cpp | send_message | [** <<UDP>> **] | client_key: 0x12345678, len: 13, data: 
[1620970428.366921] debug | UDPv6AgentLinux.cpp | recv_message | [==>> UDP <<==] | client_key: 0x12345678, len: 1388, data: 
[1620970428.367062] debug | UDPv6AgentLinux.cpp | send_message | [** <<UDP>> **] | client_key: 0x12345678, len: 13, data: 
[1620970428.371937] debug | UDPv6AgentLinux.cpp | recv_message | [==>> UDP <<==] | client_key: 0x12345678, len: 1264, data: 
[1620970428.372060] debug | UDPv6AgentLinux.cpp | send_message | [** <<UDP>> **] | client_key: 0x12345678, len: 13, data: 
[1620970428.376958] debug | UDPv6AgentLinux.cpp | recv_message | [==>> UDP <<==] | client_key: 0x12345678, len: 1264, data: 
[1620970428.377083] debug | UDPv6AgentLinux.cpp | send_message | [** <<UDP>> **] | client_key: 0x12345678, len: 13, data: 
[1620970428.381938] debug | UDPv6AgentLinux.cpp | recv_message | [==>> UDP <<==] | client_key: 0x12345678, len: 1264, data: 
`

pablogs9, 2021-05-14:
Please share your client-side code so we can replicate this.

platinum78, 2021-05-14:
Thanks for the fast reply. I would preliminarily apologize that the code that I can reveal might be somewhat discrete, since this is a part of proprietary project in my company. Sorry for the inconvenience.

While registering ROS2 topics to DDS network using MicroXRCE, the following codes are executed.

First, `registerTopic()` function is called.

```
bool xrcedds::registerTopic(xrcedds::Participant_t* participant, const char* topic_name, const char* topic_type, uint8_t topic_id)
{
    if (participant == NULL || participant->is_created == false || topic_name == NULL || topic_type == NULL)
    {
        return false;
    }

    bool ret = false;
    uint8_t status;
    uint16_t topic_req;
    
    uxrObjectId part_object_id = uxr_object_id(participant->id, UXR_PARTICIPANT_ID);
    uxrObjectId topic_object_id = uxr_object_id(topic_id, UXR_TOPIC_ID);

#if (UXR_CREATE_ENTITIES_USING_REF)
    topic_req = uxr_buffer_create_topic_ref(&g_uxr_session.session, getOutputStreamInfo(), topic_object_id, part_object_id, topic_name, UXR_REUSE);
#else
    char topic_profile[256];
    sprintf(topic_profile, UXR_TOPIC_XML, topic_name, topic_type);
    topic_req = uxr_buffer_create_topic_xml(&g_uxr_session.session, getOutputStreamInfo(), topic_object_id, part_object_id, topic_profile, UXR_REUSE);
#endif

    ret = uxr_run_session_until_all_status(&g_uxr_session.session, 1000, &topic_req, &status, 1);
    if (ret == false && status == UXR_STATUS_ERR_ALREADY_EXISTS)
    {
        ret = true;
    }

    return ret;
}
```

Next, `createDataWriter()` function is executed.

```
bool xrcedds::createDataWriter(xrcedds::Publisher_t* publisher, xrcedds::DataWriter_t* data_writer, char* writer_name, const char* topic_type)
{
    if (publisher == NULL || data_writer == NULL || writer_name == NULL || topic_type == NULL)
    {
        return false;
    }

    uint8_t status;
    uint16_t datawriter_req;
    static uint8_t writer_id = 0x01;

    data_writer->publisher = publisher;
    data_writer->id        = writer_id++;

    uxrObjectId pub_object_id  = uxr_object_id(publisher->id, UXR_PUBLISHER_ID);
    uxrObjectId writer_object_id = uxr_object_id(data_writer->id, UXR_DATAWRITER_ID);

#if (UXR_CREATE_ENTITIES_USING_REF)
    datawriter_req = uxr_buffer_create_datawriter_ref(&g_uxr_session.session, getOutputStreamInfo(), writer_object_id, pub_object_id, writer_name, UXR_REPLACE);
#else
    char writer_profile[256];
    sprintf(writer_profile, UXR_WRITER_XML, writer_name, topic_type);
    datawriter_req = uxr_buffer_create_datawriter_xml(&g_uxr_session.session, getOutputStreamInfo(), writer_object_id, pub_object_id, writer_profile, UXR_REPLACE);  
#endif

    data_writer->is_created = uxr_run_session_until_all_status(&g_uxr_session.session, 1000, &datawriter_req, &status, 1);

    return data_writer->is_created;
}
```

The two functions above are called once each per one topic. The topic I'm currently using is implemented as in the link.
* https://github.com/ROBOTIS-GIT/ros2arduino/blob/master/src/ros2/sensor_msgs/LaserScan.hpp
* https://github.com/ROBOTIS-GIT/ros2arduino/blob/master/src/ros2/sensor_msgs/Imu.hpp
* https://github.com/ROBOTIS-GIT/ros2arduino/blob/master/src/ros2/sensor_msgs/MagneticField.hpp
* https://github.com/ROBOTIS-GIT/ros2arduino/blob/master/src/ros2/nav_msgs/Odometry.hpp

Well.. again I'm so sorry that the code above is the most that I can reveal without permission. I don't expect the code above would be sufficient for you to replicate my problem, so I think I would be satisfied if I can get just some hints about what goes on inside the XRCE client side.

* I set MTU to 10,000, and no other modification.
* In accordance to what I've observed, communication seems to happen in the following way.
    * In Phase 32 of four-publisher case in above loggings, agent sends UDP messages twice each with 14-bytes and 13-bytes length.
    * The two messages triggers initial topic transmissions from the client.
    * Initial transmission is translated by the agent, and the agent does what it should
    * The agent then replies the same content except twelve header bytes. <-- This happens in three-publisher case, but not in four-publisher case. In four-publisher case, only 13-byte-long heartbeat(?) signal is received from agent.
    * The client accepts the response (original transmission - header), and confirms communication is all right. <-- This doesn't happen in four-publisher case, since the response is not correct.

pablogs9, 2021-05-14:
Well, with that code I'm not sure I can replicate your use case.

Which versions of the Micro XRCE-DDS Client are you using? 

I have test data writer creating and it works with an it works with many entities. Please find attached below my test.

I'm running the agent using:

```bash
docker run -it --rm -v /dev:/dev --privileged --net=host microros/micro-ros-agent:foxy udp4 --port 8888 -v6
```

My testing code:

```c
#include <uxr/client/client.h>
#include <ucdr/microcdr.h>

#define STREAM_HISTORY  8
#define BUFFER_SIZE     UXR_CONFIG_UDP_TRANSPORT_MTU* STREAM_HISTORY

#define NUMBER_OF_PUBLISHER 50

int main()
{
    // Transport
    uxrUDPTransport transport;
    uxr_init_udp_transport(&transport, UXR_IPv4, "localhost", "8888");

    // Session
    uxrSession session;
    uxr_init_session(&session, &transport.comm, 0xAAAABBBB);
    uxr_create_session(&session);

    // Streams
    uint8_t output_reliable_stream_buffer[BUFFER_SIZE];
    uxrStreamId reliable_out = uxr_create_output_reliable_stream(&session, output_reliable_stream_buffer, BUFFER_SIZE, STREAM_HISTORY);

    uint8_t input_reliable_stream_buffer[BUFFER_SIZE];
    uxr_create_input_reliable_stream(&session, input_reliable_stream_buffer, BUFFER_SIZE, STREAM_HISTORY);

    // Create entities
    uxrObjectId participant_id = uxr_object_id(0x01, UXR_PARTICIPANT_ID);
    const char* participant_xml = "<dds>"
            "<participant>"
            "<rtps>"
            "<name>default_xrce_participant</name>"
            "</rtps>"
            "</participant>"
            "</dds>";
    uint16_t participant_req = uxr_buffer_create_participant_xml(&session, reliable_out, participant_id, 0,
                    participant_xml, UXR_REPLACE);

    uxrObjectId topic_id = uxr_object_id(0x01, UXR_TOPIC_ID);
    const char* topic_xml = "<dds>"
            "<topic>"
            "<name>TopicName</name>"
            "<dataType>TypeName</dataType>"
            "</topic>"
            "</dds>";
    uint16_t topic_req = uxr_buffer_create_topic_xml(&session, reliable_out, topic_id, participant_id, topic_xml,
                    UXR_REPLACE);

    uxrObjectId publisher_id = uxr_object_id(0x01, UXR_PUBLISHER_ID);
    const char* publisher_xml = "";
    uint16_t publisher_req = uxr_buffer_create_publisher_xml(&session, reliable_out, publisher_id, participant_id, publisher_xml, UXR_REPLACE);

    uint16_t requests[] = {participant_req, topic_req, publisher_req};
    uint8_t status[sizeof(requests)/2];
    uxr_run_session_until_all_status(&session, 1000, requests, status, sizeof(status));

    for (size_t i = 0; i < NUMBER_OF_PUBLISHER; i++)
    {
        uxrObjectId datawriter_id = uxr_object_id(i, UXR_DATAWRITER_ID);
        const char* datawriter_xml = "<dds>"
                "<data_writer>"
                "<topic>"
                "<kind>NO_KEY</kind>"
                "<name>TopicName</name>"
                "<dataType>TypeName</dataType>"
                "</topic>"
                "</data_writer>"
                "</dds>";
        uint16_t datawriter_req = uxr_buffer_create_datawriter_xml(&session, reliable_out, datawriter_id, publisher_id, datawriter_xml, UXR_REPLACE);
        
        uint16_t requests[] = {datawriter_req};
        uint8_t status[sizeof(requests)/2];
        uxr_run_session_until_all_status(&session, 1000, requests, status, sizeof(status));
    }

    // Write topics
    uint32_t data = 0;
    while (true)
    {
        ucdrBuffer ub;
        uint32_t topic_size = sizeof(uint32_t);
        for (size_t i = 0; i < NUMBER_OF_PUBLISHER; i++){
            uxrObjectId datawriter_id = uxr_object_id(i, UXR_DATAWRITER_ID);
            uxr_prepare_output_stream(&session, reliable_out, datawriter_id, &ub, topic_size);
            ucdr_serialize_uint32_t(&ub, data);
        }
        data++;

        uxr_run_session_time(&session, 1000);
    }

    // Delete resources
    uxr_delete_session(&session);
    uxr_close_udp_transport(&transport);

    return 0;
}
```

pablogs9, 2021-05-20:
Any update on that?

platinum78, 2021-05-20:
First, thank you for your information.

I am planning to try microROS (instead of the ported version of ros2arduino which I'm working on), but I've got no enough time to do it since I'm currently working on the entire firmware of the project. So I think I would need pretty much time until I report any progress on this issue.

If keeping this issue open bothers you, you are welcomed to close it; otherwise, pls just forget about this issue until I report something about this issue again :)

pablogs9, 2021-05-20:
@platinum78 thanks a lot for the update! I'm closing to keep our issue tracker clean. But do not hesitate to reopen an issue in any of our repos.

Looking forward to hearing about your progress!

closed 2021-05-20 by pablogs9