eProsima/Micro-XRCE-DDS-Agent #286: How Agent thread are scheduled by linux?

crivi30, 2021-11-17:
Hello everyone! 

How the following thread are scheduled?
It depend on the OS or do you configure it in some way to have specific scheduling algorithm? 

```
    error_handler_thread_ = std::thread(&Server::error_handler_loop, this);
    receiver_thread_ = std::thread(&Server::receiver_loop, this);
    sender_thread_ = std::thread(&Server::sender_loop, this);
    processing_thread_ = std::thread(&Server::processing_loop, this);
    heartbeat_thread_ = std::thread(&Server::heartbeat_loop, this);
```

There is a documentation on how this work in the XRCE-DDS Library? 
Thank you in advance

Acuadros95, 2021-11-17:
The server threads don't follow an specific schedule, their behavior will depend on the OS.

The only schedule event happens when an transport error is detected on the `receiver_loop` or `sender_loop` threads, which will halt until the error is handled on `error_handler_loop` or the server is stopped.

Acuadros95, 2022-01-28:
Closing due to inactivity, reopen if needed.

closed 2022-01-28 by Acuadros95