eProsima/Micro-XRCE-DDS-Agent #300: When payload size is 1017, the agent does not publish data.

zhangyaothu, 2022-01-20:
## Issue template

- Hardware description: Client on STM32, Agent on PC
- OS: FreeRTOS for client, Ubuntu 20.04 LTS for agent
- Installation type: v2.1.0 -DUAGENT_BUILD_USAGE_EXAMPLES=ON
- Version or commit hash: 070c438

#### Steps to reproduce the issue

I use an IDL like below to test the throughput between the client publisher on STM32 nucleo board and the subscriber on Ubuntu, also wih the agent on the same Ubuntu machine.
```
struct Throughput
{
  unsigned long count;
  sequence<octet, 8192> payload;
};
```

#### Expected behavior
When the payload size is 1016, the pub and sub works well. And I expect when payload size is 1017, it will also work.

#### Actual behavior
When the payload size is 1017, the publisher client and agent works ok. But no message received in the subscriber.

#### Additional information
I attach the packets captured by wireshark [pcapng.zip](https://github.com/eProsima/Micro-XRCE-DDS-Agent/files/7904051/pcapng.zip)

Acuadros95, 2022-01-20:
Hi @zhangyaothu, I need more details on the client build, which repository or tutorial did you use to build micro-ROS?

zhangyaothu, 2022-01-20:
Thanks @Acuadros95 , the code is basically generated by microxrcegen, and changed a little to run as a freertos task.
```
static void main_throughput_publisher(void const *argument)
{
  (void)argument;
  const char* participant_xml = "<dds>"
                                    "<participant>"
                                        "<rtps>"
                                            "<name>default_xrce_participant</name>"
                                        "</rtps>"
                                    "</participant>"
                                "</dds>";
  const char* topic_xml = "<dds>"
                              "<topic>"
                                  "<name>ThroughputTopic</name>"
                                  "<dataType>Throughput</dataType>"
                              "</topic>"
                          "</dds>";
  const char* publisher_xml = "";
  const char* datawriter_xml = "<dds>"
                                   "<data_writer>"
                                       "<topic>"
                                           "<kind>NO_KEY</kind>"
                                           "<name>ThroughputTopic</name>"
                                           "<dataType>Throughput</dataType>"
                                       "</topic>"
                                   "</data_writer>"
                               "</dds>";
    // Transport
    uxrUDPTransport transport;
    if(!uxr_init_udp_transport(&transport, UXR_IPv4, "192.168.1.1", "2018"))
    {
        printf("Error at create transport.\n");
        return;
    }

    // Session
    uxrSession session;
    uxr_init_session(&session, &transport.comm, 0xAAAABBBB);
    if(!uxr_create_session(&session))
    {
        printf("Error at create session.\n");
        return;
    }

    // Streams
    uint8_t output_reliable_stream_buffer[BUFFER_SIZE];
    uxrStreamId reliable_out = uxr_create_output_reliable_stream(&session, output_reliable_stream_buffer, BUFFER_SIZE, STREAM_HISTORY);

    uint8_t input_reliable_stream_buffer[BUFFER_SIZE];
    uxr_create_input_reliable_stream(&session, input_reliable_stream_buffer, BUFFER_SIZE, STREAM_HISTORY);

    // Create entities
    uxrObjectId participant_id = uxr_object_id(0x01, UXR_PARTICIPANT_ID);
    uint16_t participant_req = uxr_buffer_create_participant_xml(&session, reliable_out, participant_id, 0, participant_xml, UXR_REPLACE);

    uxrObjectId topic_id = uxr_object_id(0x01, UXR_TOPIC_ID);
    uint16_t topic_req = uxr_buffer_create_topic_xml(&session, reliable_out, topic_id, participant_id, topic_xml, UXR_REPLACE);

    uxrObjectId publisher_id = uxr_object_id(0x01, UXR_PUBLISHER_ID);
    uint16_t publisher_req = uxr_buffer_create_publisher_xml(&session, reliable_out, publisher_id, participant_id, publisher_xml, UXR_REPLACE);

    uxrObjectId datawriter_id = uxr_object_id(0x01, UXR_DATAWRITER_ID);
    uint16_t datawriter_req = uxr_buffer_create_datawriter_xml(&session, reliable_out, datawriter_id, publisher_id, datawriter_xml, UXR_REPLACE);

    // Send create entities message and wait its status
    uint8_t status[4];
    uint16_t requests[4] = {participant_req, topic_req, publisher_req, datawriter_req};
    if(!uxr_run_session_until_all_status(&session, 1000, requests, status, 4))
    {
        printf("Error at create entities: participant: %i topic: %i publisher: %i darawriter: %i\n", status[0], status[1], status[2], status[3]);
        return;
    }

    // Write topics
    Throughput topic = {0};
    topic.count = 0;
    topic.payload_size = 1017;
    memset(topic.payload, 'A', topic.payload_size);

    while(true)
    {

        ucdrBuffer mb;
        uint32_t topic_size = Throughput_size_of_topic(&topic, 0);
        uxr_prepare_output_stream(&session, reliable_out, datawriter_id, &mb, topic_size);
        Throughput_serialize_topic(&mb, &topic);

        //send and wait one second
        uxr_run_session_time(&session, 1000);
        topic.count ++;
    }

    // Delete resources
    uxr_delete_session(&session);
}
```

zhangyaothu, 2022-01-21:
@Acuadros95 , add more information.

When payload size is 1017, [write_data.data().serialized_data().size()](https://github.com/eProsima/Micro-XRCE-DDS-Agent/blob/master/src/cpp/datawriter/DataWriter.cpp#L123) is 1025, and the write is not ok.
After I enlarge the [m_typeSize](https://github.com/eProsima/Micro-XRCE-DDS-Agent/blob/master/src/cpp/types/TopicPubSubType.cpp#L23) to 2048, the write is ok.

I guess m_typeSize is failure reason. So Why set m_typeSize to 1024? Could it be larger or other solutions?

Acuadros95, 2022-01-21:
You need to change your datawriter memory management to `PREALLOCATED_WITH_REALLOC` ([Fast-DDS docs](https://fast-dds.docs.eprosima.com/en/latest/fastdds/api_reference/rtps/resources/MemoryManagementPolicy.html)) so the writer on the Agent side increases its memory upon bigger payloads.

You can set this option like this:
```
  const char* datawriter_xml = "<dds>"
                                   "<data_writer>"
                                       "<historyMemoryPolicy>PREALLOCATED_WITH_REALLOC</historyMemoryPolicy>"
                                       "<topic>"
                                           "<kind>NO_KEY</kind>"
                                           "<name>ThroughputTopic</name>"
                                           "<dataType>Throughput</dataType>"
                                       "</topic>"
                                   "</data_writer>"
                               "</dds>";
```

You can also use the binary creation mode where this is activated by default: [API documentation](https://micro-xrce-dds.docs.eprosima.com/en/latest/client_api.html?highlight=alloc#create-entities-by-binary)  and [example](https://github.com/eProsima/Micro-XRCE-DDS-Client/blob/develop/examples/BinaryEntityCreation/main.c)

zhangyaothu, 2022-01-27:
Thanks very much, it is ok just as you said.

zhangyaothu, 2022-04-24:
@Acuadros95， sorry to bother you again.
The xml creation mode is ok, when I add ` "<historyMemoryPolicy>PREALLOCATED_WITH_REALLOC</historyMemoryPolicy>"`. But the binary creation mode does not activate by default.
Could you give some advices?

pablogs9, 2022-04-25:
In binary entity creation mode `PREALLOCATED_WITH_REALLOC` is enabled by default both in data writers and data readers , check [here](https://github.com/eProsima/Micro-XRCE-DDS-Agent/blob/22587ee829d29b257f3fc6c01c2e2a21712e140b/src/cpp/middleware/fastdds/FastDDSEntities.cpp#L178)

closed 2022-06-07 by Acuadros95