eProsima/Micro-XRCE-DDS-Agent #307: Readers/Writers match, but samples not delivered when using ConnextDDS

spiderkeys, 2022-03-15:
When subscribing to topics published by the XRCE Agent from RTI Connext DDS 6.1.0 based applications, discovery appears to complete, and readers/writers match according to both QOS and the topic's datatype, but samples are not delivered to the Connext DataReader over any available transport.

Samples are successfully delivered when subscribing from FastDDS or CycloneDDS.

- Hardware description: 
  - Client running on x64 Host talking over local network or embedded NuttX target talking over serial
  - Agent running on x64 Host
  - DDS applications running on x64 Host 
- OS: 
  - Ubuntu 20.04
- Installation type: 
  - Built from source on master
- Version or commit hash: 
  - 22587ee829d29b257f3fc6c01c2e2a21712e140b

#### Steps to reproduce the issue
1. Build the XRCE Client PublishHelloWorld example: https://github.com/eProsima/Micro-XRCE-DDS-Client/tree/master/examples/PublishHelloWorld
2. Build the XRCE Agent
3. Run the agent in UDP mode:
```sh
./MicroXRCEAgent udp4 --port 2048
```
4. Run the client in UDP mode:
```sh
./PublishHelloWorldClient "127.0.0.1" 2048
```
5. Confirm that the agent discovers the client and the topic and datawriter entities are created on the client's behalf
```
spiderkeys@spiderdesk:~/z/workspace/Micro-XRCE-DDS-Agent/build$ ./MicroXRCEAgent udp4 --port 2048
[1647303594.428805] info     | UDPv4AgentLinux.cpp | init                     | running...             | port: 2048
[1647303594.428952] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 4
[1647303595.980317] info     | Root.cpp           | create_client            | create                 | client_key: 0xAAAABBBB, session_id: 0x81
[1647303595.980355] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0xAAAABBBB, address: 127.0.0.1:48034
[1647303595.982796] info     | ProxyClient.cpp    | create_participant       | participant created    | client_key: 0xAAAABBBB, participant_id: 0x001(1)
[1647303595.982986] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0xAAAABBBB, topic_id: 0x001(2), participant_id: 0x001(1)
[1647303595.983066] info     | ProxyClient.cpp    | create_publisher         | publisher created      | client_key: 0xAAAABBBB, publisher_id: 0x001(3), participant_id: 0x001(1)
[1647303595.984776] info     | ProxyClient.cpp    | create_datawriter        | datawriter created     | client_key: 0xAAAABBBB, datawriter_id: 0x001(5), publisher_id: 0x001(3)
```
6. Use `rtiddsgen` to generate typesupport source files from the same HelloWorld.idl file in the XRCE Client project
7. Create a simple application using Connext DDS that subscribes to the HelloWorld topic published by the XRCE client:
```c++
    // Create datawriter
    auto dr_qos = ::dds::core::QosProvider::Default()->datareader_qos( nadir::dds::qos::BEST_EFFORT );
    _dr_sensor = _node->create_reader<px4::msg::SensorCombined>( "rt/fmu/out/SensorCombined", dr_qos );
    _rc_sensor = { _dr_sensor, ::dds::sub::status::DataState::any(), [&]()
    {
        auto samples = _dr_sensor.take();
        for(auto& sample : samples) {
            spdlog::info( "Got sample!" );
            if (sample.info().valid()){
                spdlog::info( "Got valid sample!" );
            }
        }
    }};

    // Wait for samples
    _waitset += _rc_sensor;
    while( true ) {
        _waitset.dispatch( dds::core::Duration::from_millisecs( 5 ) );
    } 
```
8. Confirm in RTI AdminConsole that the datawriter and datareader have matched on both type consistency and QOS
[image]

#### Additional information
Both applications seem to agree on XCDR vs XCDR2.

The only difference I can see right now is that ConnextDDS 6.1.0 is using RTPS 2.3 whereas XRCE Agent (FastDDS) is using v2.2 and Cyclone is using v2.1.

pablogs9, 2022-03-15:
Hello @spiderkeys 

I guess that it is an error, but `PublishHelloWorldClient` won't match with the code you have shared since it does not have the same topic name, type name, or reliability QoS.

I assume that you have prepared a Connext DDS application that results in the output that you share in the screenshot of the RTI AdminConsole. That is that matches with `PublishHelloWorldClient`.

Regarding the RTPS version, I have been talking with @MiguelCompany, (responsible in eProsima of Fast DDS) and there should be no problem here.

We have been analyzing the situation and it seems to be a serialization issue. Could you share the type of support generated by Micro XRCE-DDS Gen and the one generated by RTI Connext in order to ensure that the serialization is compatible?

pablogs9, 2022-03-15:
Could you also add the Micro XRCE-DDS type support?

pablogs9, 2022-03-16:
Because of https://github.com/PX4/PX4-Autopilot/pull/19326 I thought that you were generating another type of support for the XRCE-DDS side.

Can you check that the RTI Connect side is ensuring XCDR and not XCDR2 in order to guarantee that we are not having serialization and alignment issues?

spiderkeys, 2022-03-16:
I can confirm that the Connext side is using XCDR. The default QOS value in 6.x for DataRepresentationQosPolicy is AUTO_DATA_REPRESENTATION, which translates to XCDR by default (unless some other specific opt-in criteria are met, which are not being used). This is also reflected in the screenshot from Admin Console above, where requested and offered DataRepresentation both reflect XCDR.

It is worth noting that in wireshark, only metatraffic is shown ever having been sent to Connext applications - no samples are ever transmitted from the XRCE agent.

Upon subscribing to the HelloWorld topic, I can see that XRCE agent sends a few HEARTBEATs, to which Connext replies with ACKNACKs. Upon unsubscribing, Connext sends DATA submessages marking the given key Unregistered/Disposed.

I can share a pcap of the RTPS traffic, if it helps.

pablogs9, 2022-03-16:
Yes please, share a PCAP of the discovery process.

spiderkeys, 2022-03-16:
Attached as zip, due to github file restrictions.

The sequence of events are:
1. Before capture, XRCE Agent is running with HelloWorld client already connected and publishing to it. Admin Console is closed.
2. Capture is started.
3. Admin Console is opened, discovery of participants and resources occurs.
4. Admin Console subscribes to the HelloWorldTopic topic. Its view of the type information is:
[image]
5. A few seconds later, I unsubscribe from the topic (Unregister/Dispose seen in wireshark)
6. A few seconds later, I subscribe again (Heartbeats from Agent, AckNack from Admin Console)
7. A few seconds later, I unsubscribe again (Unregister/Dispose again)
8. Admin console is closed
9. Capture is ended
[xrce_connext_discovery.zip](https://github.com/eProsima/Micro-XRCE-DDS-Agent/files/8260079/xrce_connext_discovery.zip)

spiderkeys, 2022-03-16:
Just to make absolutely sure about the XCDR vs XCDR2 situation, I explicitly set the DataRepresentation QOS to XCDR_DATA_REPRESENTATION, as opposed to leaving it default. No change.

pablogs9, 2022-03-16:
We found something. Could you please try to create the subscription in RTI AdminConsole removing the partitions?

I have seen [here](https://community.rti.com/forum-topic/partition-settings-admin-console) that there is a text field with the text `{empty},*`, try the same scenario but with this field empty.

spiderkeys, 2022-03-16:
I can confirm that removing the { "", "*" } entries results in receiving samples in Admin Console. I'll test the fix on the Fast-DDS side later today. Thanks!