eProsima/Micro-XRCE-DDS-Agent #309: Multiple agents

lxc777, 2022-04-11:
Hi，When there are multiple agents and each agent has only different IP, how to distinguish the connection in the network of ros2. What parameters or settings should the agent add or change the settings of ros2?

pablogs9, 2022-04-11:
Could you explain better the issue?

lxc777, 2022-04-11:
Sorry, maybe I didn't make it clear.
Problems encountered: I have multiple micro xrce DDS agents in the same network segment. They communicate with ros2 and encounter problems when sending and listening to ros2. When ros2 publishes a message, micro xrce DDS starts subscription, and multiple agents will receive messages at this time. Is there any way to specify a specific proxy subscription?

pablogs9, 2022-04-11:
Well, DDS uses the publication/subscription paradigm where the data produced by one publisher will reach all the subscribers that match the publisher.

I'm not sure if I have understood well your question. Could you expose a use case and a demo where we can reproduce your problem?

lxc777, 2022-04-12:
Yes, DDS network, sharing publishers and subscribers. Now I need to limit this sharing. I want to use ros2 for specific users

Micro xrce DDS agent node, publish and subscribe, and other nodes are not affected.

lxc777, 2022-04-12:
What I learned can be achieved through ROS_ DOMAIN_ ID is restricted. How can I find the registered ROS_ DOMAIN_ ID to avoid duplication. 
PC1  open micro xrce DDS agent：
[image]

PC1   open ros2-android：
[image]

PC2  open  micro xrce DDS  PublishHelloWorld：
[image]

PC2 Vm ubuntu20.04  open publication and subscription ：
[image]

pablogs9, 2022-04-12:
On the ROS 2 side, as you can see in the [documentation](https://docs.ros.org/en/foxy/Concepts/About-Domain-ID.html), the domain id can be configured by setting the environment variable `DOMAIN_ID` to de desired value. In XRCE-DDS, as stated in the [documentation](https://micro-xrce-dds.docs.eprosima.com/en/stable/client_api.html#create-entities-by-xml), each participant creatin API has a `domain` argumento for setting this value.

I'm closing, please reopen if you have further questions.

closed 2022-04-12 by pablogs9

lxc777, 2022-04-14:
Thank you. It's very useful to me