eProsima/Micro-XRCE-DDS-Agent #317: delete client

stmoon, 2022-10-12:
## Issue 

I found that agent delete a client suddenly.  Could you tell me why the client is delete?

I used v2.2.1 client and agent in PX4 and Jetson nano each other. 

'''
[1661766386.857641] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0xAAAACCCD, subscriber_id: 0x005(4), participant_id: 0x001(1)
[1661766386.858022] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0xAAAACCCD, topic_id: 0x3EC(2), participant_id: 0x001(1)
[1661766386.947710] info     | ProxyClient.cpp    | create_datareader        | datareader created     | client_key: 0xAAAACCCD, datareader_id: 0x005(6), subscriber_id: 0x005(4)

**[1661768432.735306] info     | Root.cpp           | delete_client            | delete                 | client_key: 0xAAAACCCD
[1661768432.735474] info     | SessionManager.hpp | destroy_session          | session closed         | client_key: 0xAAAACCCD, address: 192.168.10.3:10516**
'''

Acuadros95, 2022-10-13:
Hi @stmoon,

Could you provide more details of your implementation?
- Are you destroying entities on your micro-ROS app?
- Are you using the [HARD_LIVELINESS_CHECK](https://discourse.ros.org/t/hard-liveliness-check-in-micro-ros/24891) feature on your micro-ROS client?
- Could you share the debug Agent output with `-v6` flag?

Acuadros95, 2022-10-20:
Any update on this?

closed 2023-07-24 by Acuadros95