eProsima/Micro-XRCE-DDS-Agent #319: Cmake could not find microcdr  cross-compile (bitbake recipe)

dirksavage88, 2022-11-10:
Cross compiling for arm using the yocto build tools (bit bake and dev tool) I’m getting this error due to cmake not finding the microcdr package

this error is very similar to those experiencing this dependency issue for the [Client](https://github.com/eProsima/Micro-XRCE-DDS-Client/issues/217#issue-842213998) code which I see no resolution for.

Target platform is yocto zeus 5.4.24-2.1.0

I followed the exact steps in the standalone agent installation manual as well as parallel installing and building on my amd64 (this found microcdr and built successfully)

I did try to create a recipe for microcdr and build this as a dependency to the agent but ram into the same error. Where is microcdr called out in the cmakelist (find package?). I wish to only install the agent if possible and can disable a lot of options if this is solely a client based dependency…

also why is my host amd64 platform able to find the package without installing a standalone microcdr build?

As [PX4 autopilot ](https://github.com/PX4/PX4-Autopilot/pull/20227#issue-1375237581) moved to xrcedds and micro ROS, I anticipate more and more users will experience this issue when cross compiling the agent for off board computing so it would be awesome if I can help find a resolution to save future headache on the user side.

pablogs9, 2022-11-10:
Micro XRCE-DDS Agent uses a CMake superbuild approach: https://github.com/eProsima/Micro-XRCE-DDS-Agent/blob/f984380d092f593cde35592be2f7571fb2bfa365/CMakeLists.txt#L123

This functionality shall be able to download and build the dependencies.

Could you provide steps for replicating this issue?

dirksavage88, 2022-11-10:
Perhaps the superbuild approach may not be conducive to the yocto build tools?

Attached is a screen snip of the error.

Steps to reproduce:

1) In a yocto build environment, with known working core recipes, use the devtool to add the micro-XRCE-DDS Agent on the host platform:
$ devtool add https://github.com/eProsima/Micro-XRCE-DDS-Agent.git
2) Edit the recipe to make sure 'inherit cmake' is present in the *.bb recipe: this is all bitbake needs in order to cross-compile and does not require any 'do_configure' sections. In a nutshell, inherit cmake does the equivalent of entering into the src repo and running cmake, make install, install, etc. 
3) The commit hash in my bitbake recipe is: f984380d092f593cde35592be2f7571fb2bfa365
4) Run this command to tell the yocto build tools to do all the work of configuring, building, installing, and packaging on the target platform (ARM):
$ devtool build uagent-superbuild

That last command is when I received the error.

Additionally, it seems other users with different build tools are also finding issues with the superbuild modality: https://madahindev.com/post/micro_xrce_dds_cmake_integration

[image]

pablogs9, 2022-11-11:
Hi @dirksavage88, sorry but I have little experience with Yocto. My only recommendation can be to avoid using superbuild (disabling the CMake flag) and try to add the dependencies manually.

In any case, it would be great if you generate some documentation when you achieve this in order to have a procedure for integrating into those kinds of platforms. Thanks a lot.

dirksavage88, 2022-11-11:
@pablogs9 that sounds like it could work, however I’m worried those changes made would only exist on a fringe fork of this repo and not be of help to others using build systems facing similar issues. 

I see that you also maintain the micro_ros_setup repo as well…which wraps this implementation in a ROS 2 node. This is also a desirable package since it is easily configurable on a native or on-target capable build system with ROS 2, but also suffers from its own build tool integration complexities as with this implementation.

 Either implementation requires tweaking and/or facilitation of post install config and build steps (micro ROS setup for example, uses a ‘configure_firmware.sh’ script as well as a build script on top of the meta build tools in ROS 2…)

The gist of the issue is that both implementations of these packages aren’t ideal for cross compiling to a embedded target (lacking on-target build tools). Also, installing microROS via snap package brings its own intricacies, but given that snaps add additional  package manager overhead in the build recipes on top of apt,  I’m not convinced it’s viable for yocto build tools either.

dirksavage88, 2022-11-12:
For anyone looking for a buildable xrce-dds client (it's quick and dirty, definitely not an elegant solution) see here: https://github.com/dirksavage88/meta-dds.git

closed 2022-11-12 by dirksavage88

pablogs9, 2022-11-14:
Also I have found some interesting article (and Github repo) here: https://antmicro.com/blog/2022/10/kenning-tvm-and-micro-ros-in-the-meta-antmicro-yocto-layer/ in the `Introduction of micro-ROS to meta-antmicro` section

agwegsol, 2025-10-17:
Some update. I have managed to make it work using scartgath and the compiled versions from meta-ros. It is almost complete but I have some hickups. This is my recipe.

```
# Recipe created by recipetool
# This is the basis of a recipe and may need further editing in order to be fully functional.
# (Feel free to remove these comments when editing.)

# WARNING: the following LICENSE and LIC_FILES_CHKSUM values are best guesses - it is
# your responsibility to verify that the values are complete and correct.
LICENSE = "Apache-2.0"
LIC_FILES_CHKSUM = "file://LICENSE;md5=3b83ef96387f14655fc854ddc3c6bd57"

SRC_URI = "git://github.com/eProsima/Micro-XRCE-DDS-Agent.git;protocol=https;branch=master \
           "

# Modify these as desired
PV = "1.0+git"
SRCREV = "40954c2379dfcb2bfa9f6ba22146c57c6742b5b7"

S = "${WORKDIR}/git"
DEPENDS = "fastcdr fastrtps"

inherit cmake

do_install() {
    install -d ${D}${libdir}
    install -m 0644 ${WORKDIR}/build/libmicroxrcedds_agent.so.2.4 ${D}${libdir}/

    install -d ${D}${base_prefix}/opt/microxrce/
    cp -r ${WORKDIR}/build/MicroXRCEAgent ${D}${base_prefix}/opt/microxrce/
}

FILES:${PN} = "${libdir}/* ${base_prefix}/opt/*"

# Specify any options you want to pass to cmake using EXTRA_OECMAKE:
EXTRA_OECMAKE += "\
    -DUAGENT_SUPERBUILD=OFF \
    -DUAGENT_P2P_PROFILE=OFF \
    -DUAGENT_LOGGER_PROFILE=OFF \
    -DUAGENT_USE_SYSTEM_FASTCDR=TRUE \
    -DUAGENT_USE_SYSTEM_FASTDDS=TRUE \
    -Dfastcdr_DIR=${WORKDIR}/recipe-sysroot/opt/ros/humble/lib/cmake/fastcdr \
    -Dfastrtps_DIR=${WORKDIR}/recipe-sysroot/opt/ros/humble/share/fastrtps/cmake \
    "

```

It does cross compile to respberryPi5 and I can run the MicroXRCEAgent.
I have not made any patch for the Micro-XRCE-DDS-Agent.git repo, but instead the recipe forces the build to use the system libs, UAGENT_USE_SYSTEM_**= TRUE.
DUAGENT_SUPERBUILD=OFF is set since git clone is not allowed from the build process when running scartgath.
So it works.

But the -Dfastcdr_DIR/fastrps_DIR is quite ugly, it should be set by yocto, but it seems that the problem is that no Cmake module files is present.
So to complete this setup, it would be nice is someone could tell how to get rid of the -D**_DIR=<path>

dirksavage88, 2025-10-20:
@agwegsol would you consider making a PR [here](https://github.com/dirksavage88/meta-dds/tree/main/recipes-xrce-agent/xrce-agent)?

I can also make a branch for scarthgap. I am working on kirkstone on the jetson tx2 nx (meta-tegrademo)

agwegsol, 2025-10-21:
I consider doing that but I need it to work 100%.
Problem: When i build using my ubuntu host and connect I get a some nice messages.
When I build using the above recipe it is silent but some messages is forwarded. I need to figure out why and what.