eProsima/Micro-XRCE-DDS-Agent #320: Latest docker Agent Image not working!

kalaajiahmad, 2022-11-24:
## Issue template

- Hardware description: Server PC
- OS: Ubuntu 20.04 Running ROS2 Foxy
- Installation type: Original version
- Version or commit hash: Latest Image from microros/micro-ros-agent with DIGEST 
b77237904993
03c786974076

#### Steps to reproduce the issue

Just pull the latest docker image and run using:
`sudo docker run -it --rm -v /dev:/dev --privileged --net=host microros/micro-ros-agent:foxy udp4 -p 28000`

#### Expected behavior
Agent opens and all is good!!

#### Actual behavior
[image]

#### Additional information

pablogs9, 2022-11-25:
Hello, @kalaajiahmad thanks for alerting.

We backported this one https://github.com/micro-ROS/docker/pull/90 and seems that in foxy and galactic is breaks the agent.

I have reverted the changes on those two branches and I'm regenerating the images: https://github.com/micro-ROS/docker/actions/runs/3545842999

As soon as they are uploaded you can try again and if it is solved, please close.

Thanks one more time.

pablogs9, 2022-11-25:
I have just tried them, now they are working. Please reopen if you have any problem.

closed 2022-11-25 by pablogs9

kalaajiahmad, 2022-11-25:
Nice! thanks @pablogs9  for the quick reaction :)
Yes I can confirm , looks good and stable now
I also just kept a copy locally just to not pull the latest every time
Any idea on safely syncing the Agent and Client versions?
Since, i experience sometime sudden updated from either side (Agent/client) and that causes some version mismatch and thus misbehavior and bad connections.
Would love your opinion on this :)

pablogs9, 2022-11-25:
It should be safe to keep micro-ROS client and agent in the same ROS 2 distro. Regarding XRCE-DDS it shall be safe to use always the same major and minor version of the client and agent.

If you found any problem with this, please report it.