eProsima/Micro-XRCE-DDS-Agent #325: Error while starting IPvX agent!

mohamed1915, 2022-12-22:
I'm running a multiple part simulation with gazebo ros2 and and micro ros agent. 

I launch the agent with : cmd=['micro-ros-agent', 'udp4', '--port', '8888' ]

Is the error for the DDS or from another component? do you have an idea of how I can I get more info about the error?

pablogs9, 2022-12-22:
Which is the error and how can we reproduce it?
Also, please open issues in micro-ROS Agent repo if you are using micro-ROS Agent, not in XRCE-DDS Agent.

mohamed1915, 2022-12-22:
The error is in the title, it shows "Error while starting IPvX agent!". Concerning your second point I found that this error message exits in the function launch_agent() in the file ArgumentParser.hpp in the XRCE-DDS Agent. Maybe I'm wrong, but does that not mean the issue is more with XRCE? what can cause this error?

pablogs9, 2022-12-22:
Do you have this error if you launch `ros2 run micro_ros_agent micro_ros_agent udp4 --port 8888`?

mohamed1915, 2022-12-22:
I launch it with a code cmd=['micro-ros-agent', 'udp4', '--port', '8888' ] which I think is equivalent.

pablogs9, 2022-12-22:
Could you test if there is an error if you launch `ros2 run micro_ros_agent micro_ros_agent udp4 --port 8888`?

mohamed1915, 2022-12-22:
I installed the micro ros agent with snap so I don't have a ros package to run from the ws, is this wrong, am I missing sth?

pablogs9, 2022-12-22:
snap is kind of outdated and unsupported for micro-ROS Agent due to problems with accessing serial ports and shared memory volumes.

Consider using dockers or binary distributions: https://docs.vulcanexus.org/en/latest/rst/installation/linux_binary_installation.html

mohamed1915, 2022-12-22:
I will remove the snap installed version and install using binaries and retest and see and let you know, thank you

achyutsun, 2023-02-16:
Hi @pablogs9 , I am also not able to start IPvX agent from micro-ROS, please help. I followed **Step 8** on https://medium.com/@SameerT009/connect-esp32-to-ros2-foxy-5f06e0cc64df . Thank you.
achyut@achyutu:~/microros_ws$ ros2 run micro_ros_agent micro_ros_agent udp4 --port 8888
[1676560888.406331] error    | UDPv4AgentLinux.cpp | init                     | bind error             | port: 8888, errno: 98
Error while starting IPvX agent!
[1676560888.406597] info     | UDPv4AgentLinux.cpp | fini                     | server stopped         | port: 8888
achyut@achyutu:~/microros_ws$

achyutsun, 2023-02-16:
I was able to progress from the 'Error while starting IPvX agent!' , by killing/closing the process using that socket as suggested in https://stackoverflow.com/questions/4465959/python-errno-98-address-already-in-use
   But I am not able to continue, as I only see the bellow message in terminal and it doesnt' progress. Please suggest if I need to create another issue or if I can follow up in this same issue.
   Also please note my error was 'errno: 98 Error while starting IPvX agent!' which got resolved because of Address/Socket already in use. Thank you.

`achyut@achyutu:~/microros_ws$ ros2 run micro_ros_agent micro_ros_agent udp4 --port 8888
[1676565117.056567] info     | UDPv4AgentLinux.cpp | init                     | running...             | port: 8888
[1676565117.057040] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 4

`

achyutsun, 2023-02-16:
My above remaining issue of not establishing the session with Microcontroller(ESP32 in my case) is resolved. I just had to press the 'Reset' button in ESP32 and it started establishing session. I subscribed to 'freertos_int32_publisher' in another terminal and I started receiving message/data' Thanks
`achyut@achyutu:~/microros_ws$ ros2 run micro_ros_agent micro_ros_agent udp4 --port 8888    
[1676565117.056567] info     | UDPv4AgentLinux.cpp | init                     | running...             | port: 88                                                                                          
[1676565117.057040] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 4                                                                            
[1676566660.628553] info     | Root.cpp           | create_client            | create                 | client_key: 0x2366888D, session_id: 0x81                                                    
[1676566660.628806] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x2366888D, address: 192.168.0.126:52712                                        
[1676566660.663359] info     | ProxyClient.cpp    | create_participant       | participant created    | client_key: 0x2366888D, participant_id: 0x000(1)                                            
[1676566660.676908] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x2366888D, topic_id: 0x000(2), participant_id: 0x000(1)                        
[1676566660.685885] info     | ProxyClient.cpp    | create_publisher         | publisher created      | client_key: 0x2366888D, publisher_id: 0x000(3), participant_id: 0x000(1)                    
[1676566660.695091] info     | ProxyClient.cpp    | create_datawriter        | datawriter created     | client_key: 0x2366888D, datawriter_id: 0x000(5), publisher_id: 0x000(3)`

Acuadros95, 2023-07-31:
Closing as issue looks resolved, reopen if needed

closed 2023-07-31 by Acuadros95