eProsima/Micro-XRCE-DDS-Agent #327: MicroXRCEAgent: command not found

mengchaoheng, 2023-01-31:
## Issue template

- Hardware description:vm ubuntu on macos
- OS: ubuntu20
- Installation type: [Installing the Agent standalone](https://micro-xrce-dds.docs.eprosima.com/en/latest/installation.html#install-agent)
- Version or commit hash: master

#### Steps to reproduce the issue
```Console
$ cd /usr/local/bin && MicroXRCEAgent udp4 -p 2019
MicroXRCEAgent: command not found
```
#### Expected behavior

#### Actual behavior

#### Additional information

mengchaoheng, 2023-01-31:
I fix it by:

```Console
parallels@ubuntu-linux-20-04-desktop:~/Micro-XRCE-DDS-Agent/build$ ./MicroXRCEAgent udp4 -p 2019
[1675157951.684301] info     | UDPv4AgentLinux.cpp | init                     | running...             | port: 2019
[1675157951.684440] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 4

```

closed 2023-01-31 by pablogs9

mengchaoheng, 2023-01-31:
Why not modify the documentation? @pablogs9

pablogs9, 2023-01-31:
Because normally `/usr/local/bin` in PATH and `MicroXRCEAgent` is an executable that shall be resolved by searching in PATH. But somehow in your system `/usr/local/bin` in not in PATH.

mengchaoheng, 2023-01-31:
in fact it is
```Console

parallels@ubuntu-linux-20-04-desktop:~/Micro-XRCE-DDS-Agent/build$ MicroXRCEAgent udp4 -p 2019
MicroXRCEAgent: command not found
parallels@ubuntu-linux-20-04-desktop:~/Micro-XRCE-DDS-Agent/build$ echo $PATH
/usr/local/lib/:/usr/local/bin/:/home/parallels/Micro-XRCE-DDS-Gen/scripts/:/home/parallels/Fast-DDS/src/fastddsgen/scripts/:/opt/ros/noetic/bin:/usr/local/lib/:/usr/local/bin/:/home/parallels/Micro-XRCE-DDS-Gen/scripts/:/home/parallels/Fast-DDS/src/fastddsgen/scripts/:/usr/local/bin/:/home/parallels/Micro-XRCE-DDS-Gen/scripts/:/home/parallels/Fast-DDS/src/fastddsgen/scripts/:/usr/local/lib/:/home/parallels/Micro-XRCE-DDS-Gen/scripts/:/home/parallels/Fast-DDS/src/fastddsgen/scripts/:/home/parallels/Micro-XRCE-DDS-Gen/scripts/:/home/parallels/Fast-DDS/src/fastddsgen/scripts/:/home/parallels/Fast-DDS/install/fastrtps/bin:/home/parallels/Fast-DDS/install/foonathan_memory_vendor/bin:/opt/gcc-arm-none-eabi-7-2017-q4-major/bin:/home/parallels/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
parallels@ubuntu-linux-20-04-desktop:~/Micro-XRCE-DDS-Agent/build$ 
```

pablogs9, 2023-01-31:
In that case, you are not running `sudo make install`

mengchaoheng, 2023-02-02:
In fact, it not the `/usr/local/bin`, it is `/usr/local/lib/`.