eProsima/Micro-XRCE-DDS-Agent #328: Cannot use cmake to install in local prefix

Ryanf55, 2023-02-20:
## Issue template

- Hardware description: N/A
- OS: Ubuntu 22.04
- Installation type: local prefix install
- Version or commit hash: `v2.3.0`

#### Steps to reproduce the issue
After cloning
```
git clone <repo>
cd Micro-XRCE-DDS-Agent
cmake -S . -B build 
 cmake --build build
 cmake --install build --prefix install
 ```

#### Expected behavior

Everything gets installed in the local directly `install` rather than globally in my system. I would like to set my own [install prefix](https://cmake.org/cmake/help/latest/manual/cmake.1.html#cmdoption-cmake--install-0) using standard CMake options. 

#### Actual behavior

It tries to write to a system directory and fails. 
```
-- Up-to-date: /home/ryan/Documents/Micro-XRCE-DDS-Agent/install/share/microxrcedds_agent/cmake/microxrcedds_agentConfig.cmake
-- Up-to-date: /home/ryan/Documents/Micro-XRCE-DDS-Agent/install/share/microxrcedds_agent/cmake/microxrcedds_agentConfigVersion.cmake
-- Up-to-date: /usr/local
CMake Error at build/cmake_install.cmake:165 (file):
  file INSTALL cannot set permissions on "/usr/local": Operation not
  permitted.
  ```
  
  And, on that line in `build/cmake_install.cmake`: 
    `file(INSTALL DESTINATION "/usr/local" TYPE DIRECTORY FILES "/home/ryan/Documents/Micro-XRCE-DDS-Agent/build/temp_install/fastcdr-1.0.26/" USE_SOURCE_PERMISSIONS)`

#### Additional information

Why is this desired? Well, I want to install the agent on my computer, but not write to my system headers. I like to reserve writing to root access files with package managers like debian only. Why don't I use debian? Well, I want to contribute here, so I need to compile and run locally :)

pablogs9, 2023-02-20:
This is not desired. If you have a patch please feel free to contribute.

In the meanwhile I have found no problem with:

```
mkdir build
cd build
cmake .. -DCMAKE_INSTALL_PREFIX=../install
make
make install
```

Ryanf55, 2023-02-20:
Thanks for that workaround, it worked! 

I did try debugging, and also turned on `-v` flag for install, but there was no extra information. 
I am not sure how I can help any further; at least the team knows about the issue. 

Once I installed according to your instructions, the microcrcedds_agent object file cannot be found. 
```
cd ../install/bin
 ./MicroXRCEAgent udp4 --help
./MicroXRCEAgent: error while loading shared libraries: libmicroxrcedds_agent.so.2.3: cannot open shared object file: No such file or directory
```

pablogs9, 2023-02-20:
AFAIK you need to add this folder to LD_LIBRARY_PATH

Ryanf55, 2023-02-20:
Ah, like so. 
```
cd Micro-XRCE-DDS-Agent/install$ LD_LIBRARY_PATH=$LD_LIBRARY_PATH:$PWD/lib/ ./bin/MicroXRCEAgent udp4
Warning: '--port <value>' is required
Usage: 'MicroXRCEAgent <udp4|udp6|tcp4|tpc6|canfd|serial|multiserial|pseudoterminal> <<args>>'
For a more detailed description about all the available arguments, please execute the agent with '-h/--help' option.

pablogs9, 2023-02-20:
that's now it is working

Ryanf55, 2023-02-20:
Thanks, I'll run with this. Do you mind leaving the issue open since the workflow I posted should theoretically be the same result as calling `make` directly?