eProsima/Micro-XRCE-DDS-Agent #329: Receiving UXR_STATUS_ERR_DDS_ERROR on 2nd publisher creation

Ryanf55, 2023-03-14:
## Issue template

- Hardware description: Ubuntu 22.04, or Pixhawk 6X
- OS: Ubuntu 22.04 or ChibiOS
- Installation type: latest develop, compiled from source
- Version or commit hash: 
- Micro ROS Agent: `humble 3.0.4`

#### Steps to reproduce the issue

* Compile and run the Ardupilot simulator, this time on a custom branch that tries to set up two publishers. Here is the [commit](https://github.com/Ryanf55/ardupilot/commit/25af822eda6430b6c52ddc2d7fe7bc772ccc1550) which switches to two publishers. 
* Run the micro ROS agent
* Run the ardupilot simulator

#### Expected behavior

* The requests to set up the 2nd publisher succeed in the client

#### Actual behavior

* The client receives a `UXR_STATUS_ERR_DDS_ERROR` for the 2nd publisher creation request

#### Additional information

Micro ROS Logs:
```bash
[micro_ros_agent-1] [1678524873.791848] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0xAAAABBBB, topic_id: 0x002(2), participant_id: 0x001(1)
[micro_ros_agent-1] [1678524873.791858] info     | ProxyClient.cpp    | create_publisher         | publisher created      | client_key: 0xAAAABBBB, publisher_id: 0x002(3), participant_id: 0x001(1)
[micro_ros_agent-1] [1678524873.791875] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0xAAAABBBB, len: 14
[micro_ros_agent-1] [1678524873.791881] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0xAAAABBBB, len: 14
[micro_ros_agent-1] [1678524873.791898] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0xAAAABBBB, len: 14
[micro_ros_agent-1] [1678524873.791902] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0xAAAABBBB, len: 13
[micro_ros_agent-1] [*** LOG ERROR #0001 ***] [2023-03-11 01:54:33] [] {argument not found}
[micro_ros_agent-1] [1678524873.797640] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0xAAAABBBB, len: 13
[micro_ros_agent-1] [1678524873.797646] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0xAAAABBBB, len: 13
[micro_ros_agent-1] [1678524873.797648] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0xAAAABBBB, len: 13
```

Acuadros95, 2023-03-14:
The second topic is not being created, check this comment: https://github.com/Ryanf55/ardupilot/commit/25af822eda6430b6c52ddc2d7fe7bc772ccc1550#r104355412

Also, I would advise to use the [BinaryEntityCreation](https://github.com/eProsima/Micro-XRCE-DDS-Client/tree/master/examples/BinaryEntityCreation)/main.c) mode, which does not use extra memory and will avoid the need to keep a `xml` on the agent side.

Ryanf55, 2023-03-14:
Thanks. Is there any way to improve the logging error message in the Micro ROS Agent to help others troubleshoot this?

pablogs9, 2023-03-16:
Can this be closed?

Acuadros95, 2023-03-16:
> Is there any way to improve the logging error message in the Micro ROS Agent to help others troubleshoot this?

Not at this moment, the logging messages are oriented to raw data.

closed 2023-03-16 by Acuadros95